epics-base/epics-base #1: Syntax Issue: Using epics base with legacy mongo-cxx driver

arkilic, 2015-08-07:
I am developing a service with mongo backend and v4. Naturally, I need to include certain features from base. As legacy mongo driver and epics base both have the concept of sockets, there is a small issue issue  defining invalid sockets. We do it using the macros (sock.h, #define INVALID_SOCKET (-1)) and mongo driver uses const. Per @shengb suggested, sharing this here so someone can help us find a peaceful solution that does not require me to ship my own version of the base or the driver.

anjohnson, 2015-08-07:
Hi Arman, please don't use github for issues related to EPICS Base. Either email them to the tech-talk mailing list or post them to our bug-tracker on Launchpad at https://bugs.launchpad.net/epics-base .

closed 2015-08-07 by anjohnson

arkilic, 2015-08-07:
Noted, thx Andrew!