epics-base/epics-base #28: Possible divide by zero in libcom/src/osi/os/posix/osdThread.c -> epicsThreadSleepQuantum ()

hjunkes, 2019-06-25:


hjunkes, 2019-06-25:
Sent too early ;-(

epicsShareFunc double epicsShareAPI epicsThreadSleepQuantum ()
{
    double hz;
    hz = sysconf ( _SC_CLK_TCK );
    if(hz<0)
        return 0.0;
    return 1.0 / hz;
}

Should be ?

epicsShareFunc double epicsShareAPI epicsThreadSleepQuantum ()
{
    double hz;
    hz = sysconf ( _SC_CLK_TCK );
    if(hz<=0.0)
        return 0.0;
    return 1.0 / hz;
}

hjunkes, 2019-06-26:
I know in IEEE 754 everything is well defined, but I find it "unattractive" programmed.

ralphlange, 2019-06-26:
Looks dangerous, yes.

But in this case, since `sysconf` returns a long that is either `-1` or the number of clock ticks per second, `0` seems a rather unlikely value.

anjohnson, 2019-06-26:
I agree with both of you. I don't see any negative consequences to @hjunkes code change, but after reading both the Linux manpage and [OpenGroup specification](http://pubs.opengroup.org/onlinepubs/9699919799/functions/sysconf.html) I agree with @ralphlange that in theory at least `sysconf()` should never be returning zero. The change would make this code fractionally more robust against weird system errors, so I don't object to our making it. The simplest way to do it might be for Heinz to add it to his RTEMS-5 work.

hjunkes, 2019-06-26:
Ok, will do.

I was just a little worried because it's a configuration value at least for RTEMS. I.e. you can simply (accidentally) set it to 0. But it will be checked in the configuration check:
...
/** The configures the number of microseconds per clock tick. */
#ifndef CONFIGURE_MICROSECONDS_PER_TICK
  #define CONFIGURE_MICROSECONDS_PER_TICK \
          RTEMS_MILLISECONDS_TO_MICROSECONDS(10)
#endif

#if 1000000 % CONFIGURE_MICROSECONDS_PER_TICK != 0
  #warning "The clock ticks per second is not an integer"
#endif

#if CONFIGURE_MICROSECONDS_PER_TICK <= 0
  #error "The CONFIGURE_MICROSECONDS_PER_TICK must be positive"
#endif

closed 2019-06-26 by hjunkes

mdavidsaver, 2019-06-26:
Maybe quicker to fix than dismiss?  c1ab30142abbfe3d354efdb08c07ef28fa3036e8