epics-base/epics-base #29: libCom/calc optimization and alignment issues

anjohnson, 2019-07-03:
I recently noticed [this change](https://github.com/hir12111/epics-base/commit/121aa19025601391c44bdab74e5dc766ed28e5f0) in Diamond's RTEMS-4.11 work, which might explain some of the optimization issues we've been seeing with both calcPerform.c and postfix.c.

When the calls to `memcpy()` that copy double literals into/out of the byte-code stream are being optimized, maybe in some cases the optimizer doesn't realize that the byte-code end of the copy isn't guaranteed to be 8-byte aligned, although the other end is. A later commit removed this change to postfix.c with a comment that the `-mstrict-align` flag was sufficient to resolve their problem, so we definitely should add that to our RTEMS build flags — @hjunkes didn't you disable optimization for one or both of the libCom/calc files? We have also seen similar issues with the Microsoft optimizer though, which doesn't have `-mstrict-align`.

Diamond's solution in the above commit was to wrap the call to `memcpy()` which forced the optimizer to not make assumptions about the alignment. Is there a better, cross-platform way to do this?

mdavidsaver, 2019-07-03:
More targeted tests would be `-fno-builtin-memcpy` and/or explicitly casting the source to `char*`.

https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html

@hjunkes @hir12111

hir12111, 2019-07-04:
I remember trying `-fno-builtin-memcpy` and not getting the expected result.
` -mstrict-align` was also preferred over it, because it doesn't have such performance impact.
I also attempted casting to (char *), but it didn't work as the factor that the compiler considered is the constant copying size, having a volatile variable for passing the size would work, because the compiler couldn't make any assumption about the it, but that solution is a bit ugly.

hjunkes, 2019-07-04:
I disabled optimization just for postfix.c
...
#include "postfixPvt.h"
#include "shareLib.h"

#ifdef RTEMS_HAS_ALTIVEC
#pragma GCC push_options
#pragma GCC optimize ("O0")
#endif

/* declarations for postfix */

/* element types */
typedef enum {
        OPERAND,
...

mdavidsaver, 2019-07-04:
> I disabled optimization just for postfix.c

I agree that this is the appropriate scope for a workaround.

> I remember trying `-fno-builtin-memcpy` and not getting the expected result.

@hir12111 I wonder if this can be ignored, and if your "noinline" change work by itself?  Is `-mstrict-align` required as well?

Of interest, have you trapped the faulting instruction?  I don't believe @hjunkes has been able to.  I would be curious to know for certain that it is a store, and which kind.

Also, it might be worth trying to build/run the GCC regression tests for the memcpy() builtin.  Though looking through `test2_sub()` I don't see a test for our exact case of: dest has lesser alignment than source, with a small constant size.

https://github.com/gcc-mirror/gcc/blob/master/gcc/testsuite/gcc.c-torture/execute/builtins/memcpy-chk.c

hir12111, 2019-07-05:
> @hir12111 I wonder if this can be ignored, and if your "noinline" change work by itself? Is -mstrict-align required as well?

That's correct, it works by itself, but I considered that `-mstrict-align` will prevent the same problem in other files (which is desirable).
The specific faulting instruction is `stfd`, this instruction is storing one floating point register into memory, and because in this case destination memory was not 8 bytes aligned (approx 87.5% chances), we got the exception.
You can reproduce that code-generation pattern easily by cross-compiling the following code (in my case with flags `-g -O2`):
```c
#include <string.h>
void fun(void *dest, double num)
{
    memcpy(dest, &num, sizeof(num));
}
```

And disassembling that in gdb will show that `memcpy` call was replaced by one single instruction:
```
(gdb) disass fun
Dump of assembler code for function fun:
   0x01800490 <+0>:     stfd    f1,0(r3)
   0x01800494 <+4>:     blr
```

mdavidsaver, 2019-07-05:
> You can reproduce that code-generation pattern easily by cross-compiling

Ah, just that easy.  The result is interesting...  Testing with gcc 4.4.7 (circa RTEMS 4.10) and 6.3.0 (circa debian 9) shows some (I think) inconsequential differences.

With `-mpowerpc` or `-mcpu=7400` I see what you do (with some unnecessary stack activity from 4.4.7).

Adding `-msoft-float` replaces the `stfd` with a pair of `stw`, which is no improvement.

I also check with `-mcpu=8540` (a la mvme3100) and see something quite different, and much closer to what I expect (stores with `stb`).  Again, adding `-msoft-float` replaces the first `stfd` with 2x `stw`.

```c
void test_double(char *dest, double src)
{
   0:   94 21 ff f0     stwu    r1,-16(r1)
   4:   d8 21 00 08     stfd    f1,8(r1)
   __builtin_memcpy(dest, &src, sizeof(src));
   8:   89 21 00 08     lbz     r9,8(r1)
   c:   99 23 00 00     stb     r9,0(r3)
  10:   89 21 00 09     lbz     r9,9(r1)
  14:   99 23 00 01     stb     r9,1(r3)
  18:   89 21 00 0a     lbz     r9,10(r1)
  1c:   99 23 00 02     stb     r9,2(r3)
  20:   89 21 00 0b     lbz     r9,11(r1)
  24:   99 23 00 03     stb     r9,3(r3)
  28:   89 21 00 0c     lbz     r9,12(r1)
  2c:   99 23 00 04     stb     r9,4(r3)
  30:   89 21 00 0d     lbz     r9,13(r1)
  34:   99 23 00 05     stb     r9,5(r3)
  38:   89 21 00 0e     lbz     r9,14(r1)
  3c:   99 23 00 06     stb     r9,6(r3)
  40:   89 21 00 0f     lbz     r9,15(r1)
}
  44:   38 21 00 10     addi    r1,r1,16
   __builtin_memcpy(dest, &src, sizeof(src));
  48:   99 23 00 07     stb     r9,7(r3)
}
  4c:   4e 80 00 20     blr
```

Testing with

```sh
cat <<EOF > copyme.c 
void test_double(char *dest, double src)
{
   __builtin_memcpy(dest, &src, sizeof(src));
}

void test_int64(char *dest, long long src)
{
   __builtin_memcpy(dest, &src, sizeof(src));
}
EOF
powerpc-linux-gnu-gcc -mcpu=8540 -g -O3 -o copyme.o -c copyme.c  && powerpc-linux-gnu-objdump -dS copyme.o
```

mdavidsaver, 2019-07-05:
So it looks like GCC's `__buildin_memcpy()` assumes the the destination is aligned at least as well as the source pointer.  @anjohnson Is this a correct assumption?  (I think not)

mdavidsaver, 2019-07-05:
I should also mention that I see the same core output with `-O3` and `-00`.  So I think the only optimization which makes a difference is whether the builtin is used.

anjohnson, 2019-07-05:
It certainly appears that the optimizer is making that assumption, and I don't think it should be. Is that a GCC bug that should be reported?

Wind River recommends that PowerPC targets should always be built with `-mstrict-align` so all our vxWorks-ppc* targets have always had that in ARCH_DEP_CFLAGS.

mdavidsaver, 2019-07-05:
`-mstrict-align` seem to result in the same byte-wise access for all CPU types I tested (powerpc, 7400, and 8540).

@hir12111 Does `-mstrict-align` alone work for you?  If not, can you repeat your test and report the faulting instruction?

> Wind River recommends that PowerPC targets should always be built with `-mstrict-align`

Is any reason provided?  Solely because the (mis)alignment exception fails hard?

I hesitate to turn on/off code generation flags without a clear reason.  
This is necessary to allow the decision to be understood, and reconsidered, in future.

[GCC documentation](https://gcc.gnu.org/onlinedocs/gcc/RS_002f6000-and-PowerPC-Options.html#RS_002f6000-and-PowerPC-Options)

```
-mno-strict-align
-mstrict-align
    On System V.4 and embedded PowerPC systems do not (do) assume that unaligned memory references are handled by the system. 
```

Also, there are more occurrences of "assume" on that page than I expected...

mdavidsaver, 2019-07-05:
Also interesting that some, but not all, RTEMS PPC targets set `-mstrict-align`.

https://github.com/RTEMS/rtems/search?q=-mstrict-align&type=Code

hir12111, 2019-07-05:
> @hir12111 Does -mstrict-align alone work for you? If not, can you repeat your test and report the faulting instruction?

Yes, it works alone, I did one commit deleting the `noinline` part when I noticed that.
Which means we have 4 different workarounds for this problem.

> Also interesting that some, but not all, RTEMS PPC targets set -mstrict-align.

could that mean that newer PPC based CPUs allow unaligned memory access with the `stfd` instruction?

mdavidsaver, 2019-07-05:
> could that mean that newer PPC based CPUs allow unaligned memory access with the `stfd` instruction?

Those BSPs which do include the older motorola_powerpc (mvme2100 et al.) and the far newer qoiq.  I suspect this is an omission for some BSPs.  I'll ask.

mdavidsaver, 2019-07-05:
https://lists.rtems.org/pipermail/users/2019-July/033387.html

anjohnson, 2019-07-05:
> > Wind River recommends that PowerPC targets should always be built with -mstrict-align

> Is any reason provided?

I grepped through the VxWorks 6.9 documentation and can't find that actually written down any more, but this flag is used in their build system for all PowerPC targets. The build file says:

```makefile
# Common default GCC flags across all the PPC CPU variants.
#
# -mstrict-align
# Unaligned memory references must be handled by the compiler itself.
# (The default on GCC is -mno-strict-align.)

CC_ARCH_SPEC    += -mstrict-align
```

The place where I originally saw the recommendation was printed in an MPC74xx/MPC750 Release Notes and Architectural Supplement for Tornado 2.1 (VxWorks 5.4) from 2001. In a section **Building Applications** → _Compiling C and C++ modules_ it says about the compiler options:

> **-mstrict-align**
> Required; avoids generating code which may result in misaligned memory accesses.

mdavidsaver, 2019-07-17:
https://devel.rtems.org/ticket/3767

mdavidsaver, 2019-07-17:
> Testing with gcc 4.4.7 (circa RTEMS 4.10) and 6.3.0 (circa debian 9) shows some (I think) inconsequential differences.

Also with gcc 8.3.0 (circa debian 10).

anjohnson, 2022-09-28:
@hjunkes is this issue still a problem, or do you have a workaround in place that resolves it?

anjohnson, 2023-06-27:
We appear to have just hit this again with an RTEMS-6 build for the RTEMS-MVME2100 target, the libCom tests faulted from an epicsCalc literal “2”, I think it was the 5th epicCalcTest program case.

anjohnson, 2024-07-15:
RTEMS issue is now https://gitlab.rtems.org/rtems/rtos/rtems/-/issues/3767