epics-base/epics-base #49: use CA Client in Xilinx MicroBlaze system

xlgforever, 2019-09-08:
i want to use a CA Client in the Xilinx MicroBlaze System(mean i want to use some API to write a CA client), i don't know if i have to build a EPICS base for that system.(Actually, i don't know if there is any way to build a EPICS base for that system).

could you please give me some advice?

mdavidsaver, 2019-09-08:
Use the CA client library does require building code for the target system.  Building EPICS is more a question of operating system than of cpu architecture.  EPICS supports Linux, Windows, OSX, RTEMS, and vxWorks.

Based on my past experience with microblaze, your system may not have sufficient resources to meet requirements with the overhead of an OS.  One example of working around such limits is a [past project](http://mdavidsaver.github.io/pscdrv/) of mine, including an example [target program](https://github.com/mdavidsaver/pscdrv/tree/master/targetApp) using LWIP.

xlgforever, 2019-09-09:
> Use the CA client library does require building code for the target system. Building EPICS is more a question of operating system than of cpu architecture. EPICS supports Linux, Windows, OSX, RTEMS, [...]

i read about some files in the EPICS base R3.14.12.8, Such as CONFIG_SITE:
```
#  The host architecture performing the build, in the form
#       <operating system>-<architecture>[-<toolset>]
#
#  Currently Supporting:
#	aix-ppc                (IBM compiler used for host builds)
#	aix-ppc-gnu            (GNU compiler used for host builds)
#	cygwin-x86             (cygwin compiler used for host builds)
#	cygwin-x86_64          (cygwin compiler used for host builds)
#	darwin-ppc             (PowerPC based Apple running OSX)
#	darwin-ppcx86          (Universal binaries for both CPUs)
#	darwin-x86             (Intel based Apple running OSX)
#	freebsd-x86            (GNU compiler used for host builds)
#	freebsd-x86_64         (GNU compiler used for host builds)
#	linux-arm              (GNU compiler used for host builds)
#	linux-ppc              (GNU compiler used for host builds)
#	linux-ppc64            (GNU compiler used for host builds)
#	linux-x86              (GNU compiler used for host builds)
#	linux-x86_64           (GNU compiler used for host builds)
#	linux-x86-borland      (Borland C++ compiler used for host builds)
#	solaris-sparc          (Sun compiler used for host builds)
#	solaris-sparc-gnu      (GNU compiler used for host builds)
#	solaris-sparc64        (Sun compiler used for host builds)
#	solaris-sparc64-gnu    (GNU compiler used for host builds)
#	solaris-x86            (Sun compiler used for host builds)
#	solaris-x86-gnu        (GNU compiler used for host builds)
#	solaris-x86_64         (Sun compiler used for host builds)
#	solaris-x86_64-gnu     (GNU compiler used for host builds)
#	win32-x86              (MS Visual C++ compiler used for host builds)
#	win32-x86-cygwin       (WIN32 API with cygwin GNU compiler used for host builds)
#	win32-x86-mingw        (MinGW compiler used for host builds)
#	win32-x86-static       (MS Visual C++ compiler used for host builds)
#	windows-x64            (MS Visual C++ compiler used for host builds)
#	windows-x64-mingw      (MinGW compiler used for host builds)
#	windows-x64-static     (MS Visual C++ compiler used for host builds)

#  Debugging builds:
#	linux-arm-debug        (GNU compiler used for host builds)
#	linux-x86-debug        (GNU compiler with -g option for host builds)
#	linux-x86_64-debug     (GNU compiler with -g option for host builds)
#	solaris-sparc-debug    (sun compiler no optimization,-g for debugging info)
#	win32-x86-debug        (MS Visual C++ compiler with debug option for host builds)
#	windows-x64-debug      (MS Visual C++ compiler with debug option for host builds)

#  EPICS_HOST_ARCH is a required environment variable
#  Do not set EPICS_HOST_ARCH in this file. 
#  Use base/startup files to set EPICS_HOST_ARCH or 
#  provide EPICS_HOST_ARCH on the GNU make command line.

#  The cross-compiler architectures to build EPICS for
#
#  Currently Supporting:

#       ios-arm                 (darwin-x86 host)
#       ios-386                 (darwin-x86 host)
#       linux-386               (linux-x86 host)
#       linux-486               (linux-x86 host)
#       linux-586               (linux-x86 host)
#       linux-686               (linux-x86 host)
#       linux-arm               (linux-x86 or -x86_64 host)
#       linux-arm_eb            (linux-x86 host)
#       linux-arm_el            (linux-x86 host)
#       linux-athlon            (linux-x86 host)
#       linux-cris              (Axis GNU crosscompiler on linux-x86 host)
#       linux-cris_v10          (Axis GNU crosscompiler on linux-x86 host)
#       linux-cris_v32          (Axis GNU crosscompiler on linux-x86 host)
#       linux-microblaze
#       linux-xscale_be
#       vxWorks-486
#       vxWorks-68040
#       vxWorks-68040lc
#       vxWorks-68060
#       vxWorks-pentium
#       vxWorks-ppc32           (32-bit PowerPC CPUs with full FPU)
#       vxWorks-ppc32sf         (32-bit PowerPC CPUs without FPU)
#       vxWorks-ppc603
#       vxWorks-ppc603_long
#       vxWorks-ppc604
#       vxWorks-ppc604_long
#       vxWorks-ppc604_altivec
#       vxWorks-mpc8540
#       vxWorks-mpc8548
#       RTEMS-at91rm9200ek
#       RTEMS-beatnik
#       RTEMS-gen68360
#       RTEMS-mcp750
#       RTEMS-mvme167
#       RTEMS-mvme2100
#       RTEMS-mvme2700
#       RTEMS-mvme3100
#       RTEMS-mvme5500
#       RTEMS-pc386
#       RTEMS-psim
#       RTEMS-uC5282
#       win32-x86-mingw         (linux-x86 or -x86_64 host)
#
```
there is a cross complier architecture, linux-microblaze. does that mean i can run epics base on MicroBlaze-based linux system? i know that Xilinx provide some documents to run a linux on microblaze.

and when i use a CA client API(such as ca_get, ca_put) on a windows PC, it seems that i don't need to build EPICS base on the windows PC. the CA client API only needs the ca.dll and Com.dll(for windows PC). if i use libca.a and libCom.a(for linux system) in a microblaze-based linux system, can i do the same thing as i do on windows PC?

All i want to do is reading/writing some PVs in  some IOCs which is run on other PCs.  i have learned about the pscdrv project that you told me(thanks!). it seems that pscdrv has some special requirements for records in IOCs(such as the message ID), otherwise it can't work well(mens that i can't read or write some PVs that don't have ID. 

i can't change the records in the IOCs, because it will cause many other problems (other system also use these PVs). the only thing i can do is to read or write the PVs' values.

ralphlange, 2019-09-09:
Please consider posting your question to the [Tech-Talk mailing list](https://epics.anl.gov/tech-talk/index.php) - it will get a much wider audience there (~1000 EPICS users compared to a handful of developers here).

anjohnson, 2019-09-09:
We have cross-built the linux-microblaze target here at APS on a linux-x86_64 host, although it should be possible to do so from almost any EPICS host architecture (even Windows, but I would recommend against that unless you have no other choice).

You first need to configure and build Base for your host architecture (there are instructions for that available online), then add `linux-microblaze` to the `CROSS_COMPILER_TARGET_ARCHS` variable in configure/os/CONFIG_SITE and configure the microblaze target by editing configure/os/CONFIG_SITE.Common.linux-microblaze and setting GNU_DIR to point to your cross-compiler installation — see the comments in that file, although they are fairly old now. It's possible that newer cross-compiler versions might need some changes to the file configure/os/CONFIG.Common.linux-microblaze, you'll have to try it out to see.

As @ralphlange said you are likely to get a faster and probably better response by posting to the tech-talk mailing list instead of filing issues here.

Good luck!

anjohnson, 2019-09-09:
Closing this and responding on tech-talk since you've posted the question there, sorry I didn't see that before, my Inbox is rather full at the moment...

closed 2019-09-09 by anjohnson