epics-base/epics-base #69: [PATCH] Allow the IOC to use a specificd TCP port while keeping UDP port 5064

CasperVector, 2020-04-16:
```diff
diff -ur base-7.0.3-orig/modules/database/src/ioc/rsrv/caservertask.c base-7.0.3/modules/database/src/ioc/rsrv/caservertask.c
--- base-7.0.3-orig/modules/database/src/ioc/rsrv/caservertask.c	2019-08-01 01:11:22.000000000 +0800
+++ base-7.0.3/modules/database/src/ioc/rsrv/caservertask.c	2020-04-16 21:20:56.272178932 +0800
@@ -582,6 +582,10 @@
     }
 
     {
+        if ( envGetConfigParamPtr ( &EPICS_CAS_TCP_PORT ) ) {
+            ca_server_port = envGetInetPortConfigParam ( &EPICS_CAS_TCP_PORT,
+                (unsigned short) CA_SERVER_PORT );
+        }
         unsigned short sport = ca_server_port;
         socks = rsrv_grab_tcp(&sport);
 
diff -ur base-7.0.3-orig/modules/libcom/src/env/envDefs.h base-7.0.3/modules/libcom/src/env/envDefs.h
--- base-7.0.3-orig/modules/libcom/src/env/envDefs.h	2019-08-01 01:11:22.000000000 +0800
+++ base-7.0.3/modules/libcom/src/env/envDefs.h	2020-04-16 21:20:56.273178940 +0800
@@ -57,6 +57,7 @@
 epicsShareExtern const ENV_PARAM EPICS_CAS_AUTO_BEACON_ADDR_LIST;
 epicsShareExtern const ENV_PARAM EPICS_CAS_BEACON_ADDR_LIST;
 epicsShareExtern const ENV_PARAM EPICS_CAS_SERVER_PORT;
+epicsShareExtern const ENV_PARAM EPICS_CAS_TCP_PORT; /* non-standard */
 epicsShareExtern const ENV_PARAM EPICS_CA_BEACON_PERIOD; /* deprecated */
 epicsShareExtern const ENV_PARAM EPICS_CAS_BEACON_PERIOD;
 epicsShareExtern const ENV_PARAM EPICS_CAS_BEACON_PORT;
```
(For EPICS v3, change `modules/database/src/ioc/rsrv/caservertask.c` to `src/ioc/rsrv/caservertask.c`, and `modules/libcom/src/env/envDefs.h` to `src/libCom/env/envDefs.h`.)

With this patch, we can configure multiple IOCs on a same host to use pre-determined TCP ports while sharing UDP port 5064, making it easier to specify firewall rules.  Following is a little wrapper script that helps to start the IOCs with automatically increasing TCP port number.
```sh
#!/bin/sh
# Usage (remember to run `rm -f /tmp/ioc.*' on machine shutdown):
#   EPICS_CAS_TCP_PORT=$( \
#       flock /tmp/ioc.lock /path/to/wrapper /tmp/ioc.port \
#   ) ../../bin/linux-x86_64/someApp st.cmd

[ -s "$1" ] || echo 5066 > "$1"
port=$(cat "$1")
expr "$port" + 1 > "$1"
echo "$port"
```

anjohnson, 2020-04-16:
HI Casper, and welcome. This is an interesting idea, I can see it being desirable for some sites and it is a relatively simple code change. I would want to do this on the 7.0 branch as it's a new feature.

I'm not quite so sure about the name `EPICS_CAS_TCP_PORT` though, which doesn't quite match the other CAS parameter names in terms of the language it uses (TCP). I think I would prefer to keep the name `EPICS_CAS_SERVER_PORT` for the TCP port number, and to add a new name such as `EPICS_CAS_SEARCH_PORT` for the UDP port number if it's being set separately. That would need a slightly different change to the code, and we'd have to make sure we don't break existing usage so it should default to the old parameter's value if not set.

Other core developers might disagree with me though, would anybody else like to chime in? Which port should get the new name? I couldn't find any other suitable words to match the functionality of the TCP port...

There should also be an entry in the file `configure/CONFIG_ENV` to provide an empty default value for the new parameter, whatever its name is.

ralphlange, 2020-04-16:
What should happen when the configured port is already in use?

mdavidsaver, 2020-04-16:
@anjohnson The PVA equivalents use `EPICS_PVA_BROADCAST_PORT` and `EPICS_PVAS_SERVER_PORT`.  So I would like `EPICS_CAS_BROADCAST_PORT` for consistency.

anjohnson, 2020-04-16:
@mdavidsaver That seems reasonable. It's still the UDP port that we're naming there.

ralphlange, 2020-04-16:
It's consistent, agreed.
Calling a receiving port 'BROADCAST' is perfectly confusing and making me cringe, but... consistency rules.

mdavidsaver, 2020-04-16:
Of course I was forgetting.  "BROADCAST_PORT" is ambiguous for CA (could be 5064 or 5065).  Never mind.

ralphlange, 2020-04-17:
Wouldn't it be more robust to define a _**port range**_ for the TCP connection?
All IOCs could be configured the same and the firewall should be happy with a range, too.

The code changes would be more complex, but the configuration would be a lot easier (dropping completely all by-ioc parts) and less error prone (no risk of mis-configuring two IOCs to use the same port; clashes only happen if the number of IOCs on the host is larger than the width of the port range).

The implementation of configuring one specific port for each IOC seems considerably more restrictive than the requirement to know the exact range of used TCP port numbers.

CasperVector, 2020-04-17:
> I think I would prefer to keep the name `EPICS_CAS_SERVER_PORT` for the TCP port number, and to add a new name such as `EPICS_CAS_SEARCH_PORT` for the UDP port number if it's being set separately.

I think the most common application scenario would be varying the TCP port while fixing the UDP port, and in this scenario directly setting the TCP port instead of setting both TCP/UDP ports then reverting to the original UDP port would be simpler.  But, as has been noted by multiple people above, consistency rules.

> That would need a slightly different change to the code, and we'd have to make sure we don't break existing usage so it should default to the old parameter's value if not set.

Then for backward compatibility, I think we would have to fall back to `EPICS_CAS_SERVER_PORT` for UDP as well if ``EPICS_CAS_[name_for_udp]_PORT`` is unspecified. (Hence the reason above for choosing to modify the TCP port in the original patch.)

CasperVector, 2020-04-17:
> Wouldn't it be more robust to define a _**port range**_ for the TCP connection?
> All IOCs could be configured the same and the firewall should be happy with a range, too.

Sure; the original patch is more of a proof of concept. We can also recognise ranges like "5066-" (no upper limit), and we only need to write the upper limit once in the firewall rule. But we still need to handle the case when the range is exhausted; if we follow the old behaviour, the port would then be chosen randomly while giving a warning (but I personally prefer a fatal error in this case so that the sysadmin would easily notice something is wrong).

anjohnson, 2025-02-19:
Core Group: This idea still seems reasonable, we would encourage someone to develop a PR.