epics-base/epics-base #80: Windows target: vs2015 and later have working strtod()

anjohnson, 2020-06-30:
The `epicsParseDouble()` routine calls `epicsStrtod()` which we introduced because the `strtod()` implementation on some OSs doesn't meet standards. This is implemented in libCom in an osi/os/ header osdStrtod.h, and our own implementation in misc/epicsStdlib.c which is defined if the header requests it. The `epicsStdlibTest` program checks the OS version of `strtod()` if we are currently using our own, so it's easy to tell when we can switch back.

The WIN32 version of osdStrtod.h has selected our own implementation for a while, but recent versions have been fixed:

+ [vs2019 - pass](https://ci.appveyor.com/api/buildjobs/7m6ik8urd3lxs43j/artifacts/epicsStdlibTest.tap)
+ [vs2017 - pass](https://ci.appveyor.com/api/buildjobs/a2g63mqunp49lama/artifacts/epicsStdlibTest.tap)
+ [vs2015 - pass](https://ci.appveyor.com/api/buildjobs/m7j3h581dyyi8dqc/artifacts/epicsStdlibTest.tap)
+ [vs2013 - fail](https://ci.appveyor.com/api/buildjobs/02v8s9oac3sc3txc/artifacts/epicsStdlibTest.tap)
+ [vs2012 - fail](https://ci.appveyor.com/api/buildjobs/tjaisgt6c16v4bmo/artifacts/epicsStdlibTest.tap)
+ [vs2010 - fail](https://ci.appveyor.com/api/buildjobs/amnwqc1f0lnhbsth/artifacts/epicsStdlibTest.tap)
+ [mingw - pass](https://ci.appveyor.com/api/buildjobs/u8cjire3w0c6eear/artifacts/epicsStdlibTest.tap)

This issue is a reminder for someone to fix osi/os/WIN32/osdStrtod.h so we use the OS version when compiling with vs2015 or later. This fix should probably be done on the 3.15 branch.

rerpha, 2021-03-09:
the discussion didn't link it, but I will have a go at this now

anjohnson, 2021-06-27:
EPICS R3.15.9 released.

PR #125 (2021-03-10): #80 - Use OS strtod if MSVC 1900 or above
Closes #80 

Hopefully i've done this correctly! 

Should now use the OS strtod implementation if using MINGw or MSVC 1900 (2015) or above and still use our own impl if not.
closed 2021-06-27 by anjohnson