epics-base/epics-base #83: [⁨libcom⁩/⁨osi] osdTimeGetCurrent doesn't work for subprocess on macos

juanfem, 2020-07-31:
Hello,

I am having some issues using pyepics on macos, when using CAProcess to create a subprocess. I managed to track the issue down to libcom/osi in base.

In macos, calling osdTimeGetCurrent() from a subprocess gives wrong results if osdTimeGetCurrent() was called beforehand.
The reason is that in order to use clock_get_time(), one first need to call host_get_clock_service() to get a port to the host clock. This is done in osi/os/Darwin/osdTime.cpp in the static function timeRegister(), but the subprocess would require a different port.

I've seen that the high-resolution time provider was added in 2013:
https://github.com/epics-base/epics-base/commit/738b8ca55f3d4e67c7247ccb4de1745bc144fdfe

The previous implementation using gettimeofday() works fine, although with a 1us resolution. The simplest solution would be to revert to that implementation, if the lower resolution is acceptable.

This snippet below can be used to prove the issue:
```
#include <stdlib.h>
#include <stdio.h>
#include <mach/mach.h>
#include <mach/clock.h>
#include <unistd.h>

int main()
{
    clock_serv_t host_clock;

    host_get_clock_service(mach_host_self(), CALENDAR_CLOCK, &host_clock);

    mach_timespec_t mts, mts2;

    clock_get_time(host_clock, &mts);

    printf("s=%d\tns=%d\n", mts.tv_sec, mts.tv_nsec);

    pid_t pid = fork();
    clock_get_time(host_clock, &mts2);

    if (pid == 0)
    {
        printf("Child:\ns=%d\tns=%d\n", mts2.tv_sec, mts2.tv_nsec);
    }
    else
    {
        printf("Parent:\ns=%d\tns=%d\n", mts2.tv_sec, mts2.tv_nsec);
    }

    if (pid == 0)
    {
        host_get_clock_service(mach_host_self(), CALENDAR_CLOCK, &host_clock);
        clock_get_time(host_clock, &mts2);

        printf("Child after getting new clock service port:\ns=%d\tns=%d\n", mts2.tv_sec, mts2.tv_nsec);
    }

    return 0;
}
```

juanfem, 2020-07-31:
I have just realised that clock_gettime() was introduced in macos high sierra. We could use it to get nano seconds precision...

This is my proposal, I can make a pull request if you'd like
```
--- base-7.0.4/modules/libcom/src/osi/os/Darwin/osdTime.cpp	2020-07-04 22:30:48.000000000 +0200
+++ epics-base/modules/libcom/src/osi/os/Darwin/osdTime.cpp	2020-07-31 14:21:22.000000000 +0200
@@ -18,40 +18,22 @@
 #define EPICS_EXPOSE_LIBCOM_MONOTONIC_PRIVATE
 #include "osiSock.h"
 
-#define epicsExportSharedSymbols
 #include "cantProceed.h"
 #include "epicsTime.h"
 #include "generalTimeSup.h"
 
-static clock_serv_t host_clock;
-
 extern "C" {
 int osdTimeGetCurrent (epicsTimeStamp *pDest)
 {
-    mach_timespec_t mts;
+
     struct timespec ts;
 
-    clock_get_time(host_clock, &mts);
-    ts.tv_sec = mts.tv_sec;
-    ts.tv_nsec = mts.tv_nsec;
+    clock_gettime( CLOCK_REALTIME, &ts);
+
     *pDest = epicsTime(ts);
     return epicsTimeOK;
 }
-} // extern "C"
-
-
-static int timeRegister(void)
-{
-    host_get_clock_service(mach_host_self(), CALENDAR_CLOCK, &host_clock);
-
-    generalTimeCurrentTpRegister("MachTime", \
-        LAST_RESORT_PRIORITY, osdTimeGetCurrent);
-
-    osdMonotonicInit();
-    return 1;
-}
-static int done = timeRegister();
-
+} 
 
 int epicsTime_gmtime(const time_t *pAnsiTime, struct tm *pTM)
 {
@@ -65,10 +47,10 @@
         epicsTimeOK : errno;
 }
 
-extern "C" epicsShareFunc void
+extern "C" LIBCOM_API void
 convertDoubleToWakeTime(double timeout, struct timespec *wakeTime)
 {
-    mach_timespec_t now;
+    struct timespec now;
     struct timespec wait;
 
     if (timeout < 0.0)
@@ -76,7 +58,7 @@
     else if (timeout > 60 * 60 * 24 * 3652.5)
         timeout = 60 * 60 * 24 * 3652.5;    /* 10 years */
 
-    clock_get_time(host_clock, &now);
+    clock_gettime( CLOCK_REALTIME, &now);
 
     wait.tv_sec  = static_cast< time_t >(timeout);
     wait.tv_nsec = static_cast< long >((timeout - (double)wait.tv_sec) * 1e9);
```

mdavidsaver, 2020-07-31:
Which of the three [methods for the multiprocessing module](https://docs.python.org/3.7/library/multiprocessing.html#contexts-and-start-methods) are you using?  Use of `CAProcess` from pyepics with the `fork` method is known not to work properly and should be avoided.

FYI. https://epics.anl.gov/tech-talk/2020/msg01358.php

The change you suggest maybe desirable anyway as a simplification.  Have you tried simply removing `modules/libcom/src/osi/os/Darwin/osdTime.cpp`?  This will cause a fallback to `modules/libcom/src/osi/os/posix/osdTime.cpp` which uses `clock_gettime()` in some cases (and `gettimeofday()` in others).

> I have just realised that clock_gettime() was introduced in macos high sierra

What does this mean in terms of portability?  (I'm not a mac user)

anjohnson, 2020-07-31:
My laptop is running Mojave, the major release after High Sierra. If I remove the osi/os/Darwin/osdTime.c file (but not osdTime.h which is still necessary) I can build libCom, but attempting to run any executable results in a dyld error:
```
woz$ ./epicsTimeTest 
dyld: Symbol not found: _osdTimeGetCurrent
  Referenced from: /Users/anj/Software/epics/base-7.0/lib/darwin-x86/libCom.3.18.1.dylib
  Expected in: flat namespace
 in /Users/anj/Software/epics/base-7.0/lib/darwin-x86/libCom.3.18.1.dylib
Abort trap: 6
```
This occurs even for test programs that don't use the current time like `epicsMathTest` and `epicsTypesTest`. Thus Michael's trivial fix unfortunately won't work. I don't think the lower resolution of the `gettimeofday()` API is acceptable in the regular Base release given that the OS provides higher resolution clocks; we do measure and use time differences to nanosecond accuracy.

I just tried switching the osi/os/Darwin/osdTime.cpp code to use `clock_gettime()` based on your changes above. The main difference reported by `epicsTimeTest` is that fetching the current time seems to be about 28 times faster! The measurement of 10,000 calls to `epicsTime::getCurrent()` in our test code gives an average of 0.058 µsec instead of 1.606 µsec, so this definitely seems worth doing.

I want to make the code changes in a way that doesn't break builds on older versions of macOS if possible though, since I suspect there may be some in the community who want to run EPICS on older releases despite them having been dropped by Apple (support for Sierra stopped in October 2019).

Unfortunately I don't have access to any machines still running Sierra or earlier so I can't test any modifications there or be certain that I'm telling them apart properly. I'm pretty sure though that checking whether `CLOCK_REALTIME` gets defined by including sys/time.h should be sufficient – apparently time.h was new in Sierra; it gets included by sys/time.h, and this is how we checked for it before. This works for me on Mojave:

```diff
diff --git a/modules/libcom/src/osi/os/Darwin/osdTime.cpp b/modules/libcom/src/osi/os/Darwin/osdTime.cpp
index 88abd00d6..2be0cbb72 100644
--- a/modules/libcom/src/osi/os/Darwin/osdTime.cpp
+++ b/modules/libcom/src/osi/os/Darwin/osdTime.cpp
@@ -12,8 +12,25 @@
 #include <stdio.h>
 #include <string.h>
 #include <errno.h>
-#include <mach/mach.h>
-#include <mach/clock.h>
+#include <sys/time.h>
+
+#ifndef CLOCK_REALTIME
+  #include <mach/mach.h>
+  #include <mach/clock.h>
+
+  static clock_serv_t host_clock;
+
+  #define HOST_GETCLOCK host_get_clock_service(mach_host_self(), \
+              CALENDAR_CLOCK, &host_clock)
+  #define TIMESPEC mach_timespec_t
+  #define CLOCK_GETTIME(ts) clock_get_time(host_clock, &ts)
+  #define TP_NAME "MachTime"
+#else
+  #define HOST_GETCLOCK
+  #define TIMESPEC struct timespec
+  #define CLOCK_GETTIME(ts) clock_gettime(CLOCK_REALTIME, ts)
+  #define TP_NAME "OS Clock"
+#endif
 
 #define EPICS_EXPOSE_LIBCOM_MONOTONIC_PRIVATE
 #include "osiSock.h"
@@ -22,17 +39,19 @@
 #include "epicsTime.h"
 #include "generalTimeSup.h"
 
-static clock_serv_t host_clock;
-
 extern "C" {
 int osdTimeGetCurrent (epicsTimeStamp *pDest)
 {
-    mach_timespec_t mts;
     struct timespec ts;
 
-    clock_get_time(host_clock, &mts);
+#ifndef CLOCK_REALTIME
+    TIMESPEC mts;
+    CLOCK_GETTIME(&mts);
     ts.tv_sec = mts.tv_sec;
     ts.tv_nsec = mts.tv_nsec;
+#else
+    CLOCK_GETTIME(&ts);
+#endif
     *pDest = epicsTime(ts);
     return epicsTimeOK;
 }
@@ -41,9 +60,9 @@ int osdTimeGetCurrent (epicsTimeStamp *pDest)
 
 static int timeRegister(void)
 {
-    host_get_clock_service(mach_host_self(), CALENDAR_CLOCK, &host_clock);
+    HOST_GETCLOCK;
 
-    generalTimeCurrentTpRegister("MachTime", \
+    generalTimeCurrentTpRegister(TP_NAME, \
         LAST_RESORT_PRIORITY, osdTimeGetCurrent);
 
     osdMonotonicInit();
@@ -67,7 +86,7 @@ int epicsTime_localtime(const time_t *clock, struct tm *result)
 extern "C" LIBCOM_API void
 convertDoubleToWakeTime(double timeout, struct timespec *wakeTime)
 {
-    mach_timespec_t now;
+    TIMESPEC now;
     struct timespec wait;
 
     if (timeout < 0.0)
@@ -75,7 +94,7 @@ convertDoubleToWakeTime(double timeout, struct timespec *wakeTime)
     else if (timeout > 60 * 60 * 24 * 3652.5)
         timeout = 60 * 60 * 24 * 3652.5;    /* 10 years */
 
-    clock_get_time(host_clock, &now);
+    CLOCK_GETTIME(&now);
 
     wait.tv_sec  = static_cast< time_t >(timeout);
     wait.tv_nsec = static_cast< long >((timeout - (double)wait.tv_sec) * 1e9);
```

Please let me know if this fixes the issue for you. If you have access to a Sierra or earlier machine I would also appreciate if you can see whether it builds and runs properly there.

mdavidsaver, 2020-07-31:
Maybe relevant.  The [list of mac os versions](https://docs.travis-ci.com/user/reference/osx/#macos-version) available on travis-ci.

juanfem, 2020-08-03:
> Which of the three [methods for the multiprocessing module](https://docs.python.org/3.7/library/multiprocessing.html#contexts-and-start-methods) are you using? Use of `CAProcess` from pyepics with [...]

The code I am trying to use was developed at another lab. The use CAProcess, so I guess by default is using the fork method. Thanks for the link, I'll let the developers know. Does this mean that pyepics should use the `spawn` method by default?

> The change you suggest maybe desirable anyway as a simplification. Have you tried simply removing `modules/libcom/src/osi/os/Darwin/osdTime.cpp`? This will cause a fallback to `modules/libcom/src/osi/os/posix/osdTime.cpp` which uses `clock_gettime()` in some cases (and `gettimeofday()` in others).

As @anjohnson said, I confirm that removing osdTime.cpp it is not enough to fallback to the posix implementation.

> Please let me know if this fixes the issue for you. If you have access to a Sierra or earlier machine I would also appreciate if you can see whether it builds and runs properly there.

I confirm your code works fine in my laptop running Mojave, and it fixes my issue.
I found an old macbook running OS X Lion 10.7.5 and I'll test there later today.

juanfem, 2020-08-03:
@anjohnson, your code works fine on OS X Lion 10.7.5 with a minor modification:
```
 #define CLOCK_GETTIME(ts) clock_gettime(CLOCK_REALTIME, &ts)
```
becomes
```
 #define CLOCK_GETTIME(ts) clock_gettime(CLOCK_REALTIME, ts)
```

```
diff --git a/modules/libcom/src/osi/os/Darwin/osdTime.cpp b/modules/libcom/src/osi/os/Darwin/osdTime.cpp
index 88abd00d6..2be0cbb72 100644
--- a/modules/libcom/src/osi/os/Darwin/osdTime.cpp
+++ b/modules/libcom/src/osi/os/Darwin/osdTime.cpp
@@ -12,8 +12,25 @@
 #include <stdio.h>
 #include <string.h>
 #include <errno.h>
-#include <mach/mach.h>
-#include <mach/clock.h>
+#include <sys/time.h>
+
+#ifndef CLOCK_REALTIME
+  #include <mach/mach.h>
+  #include <mach/clock.h>
+
+  static clock_serv_t host_clock;
+
+  #define HOST_GETCLOCK host_get_clock_service(mach_host_self(), \
+              CALENDAR_CLOCK, &host_clock)
+  #define TIMESPEC mach_timespec_t
+  #define CLOCK_GETTIME(ts) clock_get_time(host_clock, ts)
+  #define TP_NAME "MachTime"
+#else
+  #define HOST_GETCLOCK
+  #define TIMESPEC struct timespec
+  #define CLOCK_GETTIME(ts) clock_gettime(CLOCK_REALTIME, ts)
+  #define TP_NAME "OS Clock"
+#endif
 
 #define EPICS_EXPOSE_LIBCOM_MONOTONIC_PRIVATE
 #include "osiSock.h"
@@ -22,17 +39,19 @@
 #include "epicsTime.h"
 #include "generalTimeSup.h"
 
-static clock_serv_t host_clock;
-
 extern "C" {
 int osdTimeGetCurrent (epicsTimeStamp *pDest)
 {
-    mach_timespec_t mts;
     struct timespec ts;
 
-    clock_get_time(host_clock, &mts);
+#ifndef CLOCK_REALTIME
+    TIMESPEC mts;
+    CLOCK_GETTIME(&mts);
     ts.tv_sec = mts.tv_sec;
     ts.tv_nsec = mts.tv_nsec;
+#else
+    CLOCK_GETTIME(&ts);
+#endif
     *pDest = epicsTime(ts);
     return epicsTimeOK;
 }
@@ -41,9 +60,9 @@ int osdTimeGetCurrent (epicsTimeStamp *pDest)
 
 static int timeRegister(void)
 {
-    host_get_clock_service(mach_host_self(), CALENDAR_CLOCK, &host_clock);
+    HOST_GETCLOCK;
 
-    generalTimeCurrentTpRegister("MachTime", \
+    generalTimeCurrentTpRegister(TP_NAME, \
         LAST_RESORT_PRIORITY, osdTimeGetCurrent);
 
     osdMonotonicInit();
@@ -67,7 +86,7 @@ int epicsTime_localtime(const time_t *clock, struct tm *result)
 extern "C" LIBCOM_API void
 convertDoubleToWakeTime(double timeout, struct timespec *wakeTime)
 {
-    mach_timespec_t now;
+    TIMESPEC now;
     struct timespec wait;
 
     if (timeout < 0.0)
@@ -75,7 +94,7 @@ convertDoubleToWakeTime(double timeout, struct timespec *wakeTime)
     else if (timeout > 60 * 60 * 24 * 3652.5)
         timeout = 60 * 60 * 24 * 3652.5;    /* 10 years */
 
-    clock_get_time(host_clock, &now);
+    CLOCK_GETTIME(&now);
 
     wait.tv_sec  = static_cast< time_t >(timeout);
     wait.tv_nsec = static_cast< long >((timeout - (double)wait.tv_sec) * 1e9);
```

Let me know if you want me to run any specific test on that machine.

anjohnson, 2020-08-03:
Oh yes, thanks for checking and pointing out that fix. The fact that this builds on Lion should be sufficient, but best way to test this if you want to do that is to cd into Base's `modules/libcom/test/O.darwin-x86` directory and run `./epicsTimeTest` which should end with this:
```
    Results
    =======
       Tests: 212
      Passed: 212 = 100.00%
```
In case you're interested, between the output of tests 15 and 16 you will find the measurement of how long it takes EPICS to read the current time, which on my Mojave laptop gives results like this:
```
# Running 10 loops
# 100000 calls to epicsTime::getCurrent() averaged  0.060 usec each
```

I will commit this change to the 7.0 branch of Base. I'm not sure if it will make it into the 7.0.4.1 release, but it will certainly be in the one after that.

Thanks!

closed 2020-08-03 by epics-jenkins