epics-base/epics-base #94: Output from console interrupting typing

bsobhani, 2020-11-30:
Is there a technical reason why output from the EPICS console has to interrupt what you type in the "epics>" prompt. This can make troubleshooting an IOC very frustrating. Why not isolate the "epics>" from the rest of the console like a linux or windows command line?

anjohnson, 2020-11-30:
If you can explain how to do that we'd be happy to hear your suggestions, and even better a pull request implementing that. I'm not sure that I understand what you mean, what Linux shell do you use?

The complexity that we're dealing with though is that an EPICS IOC has many background threads running which normally all send their stdout and stderr streams to the console, so messages from them can appear at any time. We wouldn't want to stop the output from background thread messages from appearing just because the user has started to type a command, and trying to coordinate the output of multiple threads isn't easy without adversely affecting performance or potentially freezing the IOC in some circumstances. Our code also runs on several different platforms, some of which don't support anything like curses for moving the cursor around on the screen. We do use a wrapper epicsReadline for the 3 different APIs that the IOC can use to prompt the user for commands and that support command-line recall and editing, but it doesn't have a way of notifying them when something else sends output to the screen that might mess up the command-line.

Note that you can turn off some IOC messages by running the iocsh command `eltc 0` (errLog to Console off) but that only stops messages that are output using `errlogPrintf()` or `epicsPrintf()`, not those from regular `printf()` calls – that's why we encourage writers of drivers and device support to use those.

anjohnson, 2022-09-28:
Core Group 9/28: Not something we can address in the EPICS IOC, maybe an external tool might be able to handle this.

closed 2022-09-28 by anjohnson