epics-base/epics-base #97: Segfault when reading from compress record into aai record

krisztianloki, 2021-01-05:
Commit 542353aedbf521e822f06d04d737f94dffc80088 seems to have introduced a regression and segfault when reading from `compress` records into an `aai` record. Test db:
```
record(compress, "compress")
{
  field(ALG,  "Circular Buffer")
  field(INP,  "jaai PP")
  field(NSAM, "5")
  field(PINI, "YES")
}

record(aai, "caai")
{
  field(INP,  "42")
  field(NELM, "5")
  field(FTVL, "DOUBLE")
}

record(aai, "jaai")
{
  field(INP,  {const:[1,2,3,4,5]})
  field(NELM, "5")
  field(FTVL, "DOUBLE")
}

record(aai, "aai")
{
  field(INP,  "compress")
  field(NELM, "5")
  field(FTVL, "DOUBLE")
}

record(ao, "kill")
{
  field(FLNK, "aai")
}
```
`dbpf kill 1` results in a segfault.

Changing `if (dbLinkIsConstant(plink))` to `if (plink->type == CONSTANT || plink->type == JSON_LINK)` and moving `dbInitLink(plink, DBF_INLINK);` inside the `if` seems to fix the issue and correctly initializes the array:
```
diff --git a/modules/database/src/std/dev/devAaiSoft.c b/modules/database/src/std/dev/devAaiSoft.c
index 55975ba..99b46a7 100644
--- a/modules/database/src/std/dev/devAaiSoft.c
+++ b/modules/database/src/std/dev/devAaiSoft.c
@@ -47,13 +47,13 @@ static long init_record(dbCommon *pcommon)
     aaiRecord *prec = (aaiRecord *)pcommon;
     DBLINK *plink = &prec->inp;
 
-    /* This is pass 0, link hasn't been initialized yet */
-    dbInitLink(plink, DBF_INLINK);
-
-    if (dbLinkIsConstant(plink)) {
+    if (plink->type == CONSTANT || plink->type == JSON_LINK) {
         long nRequest = prec->nelm;
         long status;
 
+        /* This is pass 0, link hasn't been initialized yet */
+        dbInitLink(plink, DBF_INLINK);
+
         /* Allocate a buffer, record support hasn't done that yet */
         if (!prec->bptr) {
             prec->bptr = callocMustSucceed(nRequest, dbValueSize(prec->ftvl),
```
I don't know if it causes other problems though...

anjohnson, 2021-01-06:
Thanks for the report and database which also seg-faults for me on Linux just as you describe. I don't think your fix is quite right, but I haven't worked out exactly what's happening yet.

krisztianloki, 2021-01-06:
Hmm... forcing the `INP` link to be `CA` "fixes" the issue; i.e. no segfault and the array is copied correctly.
Reversing `iterateRecords()` in `iocInit.c` (so it initializes compress records before aai records) also fixes the issue... so it seems that normally the `dbInitLink()` call while initializing`aai` happens before the `compress` record is initialized so the link ends up trying to read from a `NULL` pointer

mdavidsaver, 2021-01-06:
Please always include a stack trace when reporting a crash.  If nothing else, it can help others to realize they are seeing the same issue.

> ... also seg-faults for me on Linux ...

I'm not able to reproduce a crash loading the provided .db into a fresh build of a9e3fa74aa5ed62304e38da82408bdb0ac12c5c1 (aka. `R7.0.4.1-121-ga9e3fa74a`) on Debian 10/amd64.  The various VAL fields look like what I would expect afterwards.

```
$ ./bin/linux-x86_64/softIoc -d crash.db 
...
epics> dbpf kill 1
DBF_DOUBLE:         1         
epics> dbgf compress
DBF_DOUBLE[5]:      1         2         3         4         5         
epics> dbgf kill
DBF_DOUBLE:         1         
epics> dbgf caai
DBF_DOUBLE:         42        
epics> dbgf jaai
DBF_DOUBLE[5]:      1         2         3         4         5         
epics> dbgf aai
DBF_DOUBLE[5]:      1         2         3         4         5
```

I also tried with -debug, and running in valgrind doesn't show any errors.

anjohnson, 2021-01-06:
```
tux% gdb softIoc
...
Reading symbols from /home/phoebus4/ANJ/epics/base/7.0/bin/linux-x86_64/softIoc...done.
(gdb) run -d db/97.db
Starting program: /home/phoebus/ANJ/Linux64/bin/softIoc -d db/97.db
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff5b87700 (LWP 20518)]
[Thread 0x7ffff5b87700 (LWP 20518) exited]
[Detaching after fork from child process 20519]
[New Thread 0x7ffff7fa0700 (LWP 20520)]
Starting iocInit
############################################################################
## EPICS R7.0.4.2-DEV
## Rev. R7.0.4.1-121-ga9e3fa74aa5ed62304e3-dirty
############################################################################
[New Thread 0x7ffff7f1f700 (LWP 20521)]
[New Thread 0x7ffff5386700 (LWP 20522)]
[New Thread 0x7ffff5b87700 (LWP 20523)]
[New Thread 0x7ffff5285700 (LWP 20524)]
[New Thread 0x7ffff5084700 (LWP 20525)]
[New Thread 0x7ffff4e83700 (LWP 20526)]
[New Thread 0x7ffff4c82700 (LWP 20527)]
[New Thread 0x7ffff4a81700 (LWP 20528)]
[New Thread 0x7ffff4880700 (LWP 20529)]
[New Thread 0x7ffff467f700 (LWP 20530)]
[New Thread 0x7ffff447e700 (LWP 20531)]
[New Thread 0x7ffff427d700 (LWP 20532)]
[New Thread 0x7fffc7fff700 (LWP 20533)]
[New Thread 0x7fffc7dfe700 (LWP 20534)]
[New Thread 0x7fffc7bfd700 (LWP 20536)]
[New Thread 0x7fffc7afc700 (LWP 20537)]
[New Thread 0x7ffff7e9e700 (LWP 20538)]
iocRun: All initialization complete
epics> dbpf kill 1

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff699a930 in __memmove_ssse3 () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install glibc-2.17-317.el7.x86_64 libgcc-4.8.5-44.el7.x86_64 libstdc++-4.8.5-44.el7.x86_64 ncurses-libs-5.9-14.20130511.el7_4.x86_64 readline-6.2-11.el7.x86_64
(gdb) bt
#0  0x00007ffff699a930 in __memmove_ssse3 () from /lib64/libc.so.6
#1  0x00007ffff7923fcf in getDoubleDouble ()
   from /home/phoebus4/ANJ/epics/base/7.0/lib/linux-x86_64/libdbCore.so.3.18.2
#2  0x00007ffff79178e8 in dbGet (paddr=paddr@entry=0x6c7720, 
    dbrType=dbrType@entry=10, pbuffer=pbuffer@entry=0x6b5b60, 
    options=options@entry=0x0, nRequest=nRequest@entry=0x7fffffffd4c0, 
    pflin=pflin@entry=0x0) at ../db/dbAccess.c:1000
#3  0x00007ffff791a7fc in dbChannelGet (chan=chan@entry=0x6c7718, 
    type=type@entry=10, pbuffer=pbuffer@entry=0x6b5b60, 
    options=options@entry=0x0, nRequest=nRequest@entry=0x7fffffffd4c0, 
    pfl=pfl@entry=0x0) at ../db/dbChannel.c:661
#4  0x00007ffff7925334 in dbDbGetValue (plink=0x6b6cf8, 
    dbrType=<optimized out>, pbuffer=0x6b5b60, pnRequest=0x7fffffffd4c0)
    at ../db/dbDbLink.c:217
#5  0x00007ffff7928607 in dbGetLink (plink=plink@entry=0x6b6cf8, 
    dbrType=<optimized out>, pbuffer=<optimized out>, 
    poptions=poptions@entry=0x0, pnRequest=pnRequest@entry=0x7fffffffd4c0)
    at ../db/dbLink.c:328
#6  0x00007ffff7bbfde0 in readLocked (pinp=0x6b6cf8, dummy=<optimized out>)
    at ../dev/devAaiSoft.c:77
#7  0x00007ffff7bbff51 in read_aai (prec=0x6b6a90) at ../dev/devAaiSoft.c:99
#8  0x00007ffff7b92282 in readValue (prec=0x6b6a90) at ../rec/aaiRecord.c:370
#9  process (pcommon=0x6b6a90) at ../rec/aaiRecord.c:158
#10 0x00007ffff7916b48 in dbProcess (precord=precord@entry=0x6b6a90)
    at ../db/dbAccess.c:589
#11 0x00007ffff7925111 in processTarget (pdst=0x6b6a90, psrc=0x6b6e60)
    at ../db/dbDbLink.c:488
#12 dbScanPassive (pfrom=0x6b6e60, pto=0x6b6a90) at ../db/dbDbLink.c:420
#13 0x00007ffff7934140 in recGblFwdLink (precord=precord@entry=0x6b6e60)
    at ../db/recGbl.c:251
#14 0x00007ffff7b96a58 in process (pcommon=0x6b6e60) at ../rec/aoRecord.c:235
#15 0x00007ffff7916b48 in dbProcess (precord=precord@entry=0x6b6e60)
    at ../db/dbAccess.c:589
#16 0x00007ffff7918988 in dbPutField (paddr=paddr@entry=0x7fffffffd870, 
    dbrType=dbrType@entry=0, pbuffer=pbuffer@entry=0x6fa81a, nRequest=1)
    at ../db/dbAccess.c:1260
#17 0x00007ffff792ff70 in dbpf (pname=0x6fa815 "kill", pvalue=0x6fa81a "1")
    at ../db/dbTest.c:402
#18 0x00007ffff745ea50 in iocshBody ()
   from /home/phoebus4/ANJ/epics/base/7.0/lib/linux-x86_64/libCom.so.3.18.2
#19 0x0000000000406296 in main (argc=3, argv=0x7fffffffdbf8)
    at ../softIoc/softMain.cpp:245
```

anjohnson, 2021-01-06:
```
tux% valgrind softIoc -d db/97.db 
==20951== Memcheck, a memory error detector
==20951== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==20951== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==20951== Command: softIoc -d db/97.db
==20951== 
Starting iocInit
############################################################################
## EPICS R7.0.4.2-DEV
## Rev. R7.0.4.1-121-ga9e3fa74aa5ed62304e3-dirty
############################################################################
iocRun: All initialization complete
epics> dbpf kill 1
==20951== Syscall param sendmsg(msg.msg_name) points to uninitialised byte(s)
==20951==    at 0x60FC9ED: ??? (in /usr/lib64/libc-2.17.so)
==20951==    by 0x65FD1FE: readline (in /usr/lib64/libreadline.so.6.2)
==20951==    by 0x55A7BF1: epicsReadline (in /home/phoebus4/ANJ/epics/base/7.0/lib/linux-x86_64/libCom.so.3.18.2)
==20951==    by 0x559856E: iocshBody (in /home/phoebus4/ANJ/epics/base/7.0/lib/linux-x86_64/libCom.so.3.18.2)
==20951==    by 0x406295: main (softMain.cpp:245)
==20951==  Address 0x1ffefff7d2 is on thread 1's stack
==20951==  in frame #1, created by readline (???:)
==20951== 
==20951== Invalid read of size 8
==20951==    at 0x4C2E68C: memcpy@GLIBC_2.2.5 (vg_replace_strmem.c:1034)
==20951==    by 0x50B5FCE: getDoubleDouble (in /home/phoebus4/ANJ/epics/base/7.0/lib/linux-x86_64/libdbCore.so.3.18.2)
==20951==    by 0x50A98E7: dbGet (dbAccess.c:1000)
==20951==    by 0x50B7333: dbDbGetValue (dbDbLink.c:217)
==20951==    by 0x50BA606: dbGetLink (dbLink.c:328)
==20951==    by 0x4E6EDDF: readLocked (devAaiSoft.c:77)
==20951==    by 0x4E6EF50: read_aai (devAaiSoft.c:99)
==20951==    by 0x4E41281: readValue (aaiRecord.c:370)
==20951==    by 0x4E41281: process (aaiRecord.c:158)
==20951==    by 0x50A8B47: dbProcess (dbAccess.c:589)
==20951==    by 0x50B7110: processTarget (dbDbLink.c:488)
==20951==    by 0x50B7110: dbScanPassive (dbDbLink.c:420)
==20951==    by 0x50C613F: recGblFwdLink (recGbl.c:251)
==20951==    by 0x4E45A57: process (aoRecord.c:235)
==20951==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==20951== 
==20951== 
==20951== Process terminating with default action of signal 11 (SIGSEGV)
==20951==  Access not within mapped region at address 0x0
==20951==    at 0x4C2E68C: memcpy@GLIBC_2.2.5 (vg_replace_strmem.c:1034)
==20951==    by 0x50B5FCE: getDoubleDouble (in /home/phoebus4/ANJ/epics/base/7.0/lib/linux-x86_64/libdbCore.so.3.18.2)
==20951==    by 0x50A98E7: dbGet (dbAccess.c:1000)
==20951==    by 0x50B7333: dbDbGetValue (dbDbLink.c:217)
==20951==    by 0x50BA606: dbGetLink (dbLink.c:328)
==20951==    by 0x4E6EDDF: readLocked (devAaiSoft.c:77)
==20951==    by 0x4E6EF50: read_aai (devAaiSoft.c:99)
==20951==    by 0x4E41281: readValue (aaiRecord.c:370)
==20951==    by 0x4E41281: process (aaiRecord.c:158)
==20951==    by 0x50A8B47: dbProcess (dbAccess.c:589)
==20951==    by 0x50B7110: processTarget (dbDbLink.c:488)
==20951==    by 0x50B7110: dbScanPassive (dbDbLink.c:420)
==20951==    by 0x50C613F: recGblFwdLink (recGbl.c:251)
==20951==    by 0x4E45A57: process (aoRecord.c:235)
==20951==  If you believe this happened as a result of a stack
==20951==  overflow in your program's main thread (unlikely but
==20951==  possible), you can try to increase the size of the
==20951==  main thread stack using the --main-stacksize= flag.
==20951==  The main thread stack size used in this run was 8388608.
==20951== 
==20951== HEAP SUMMARY:
==20951==     in use at exit: 811,611 bytes in 12,600 blocks
==20951==   total heap usage: 48,769 allocs, 254 frees, 1,288,463 bytes allocated
==20951== 
==20951== LEAK SUMMARY:
==20951==    definitely lost: 226 bytes in 11 blocks
==20951==    indirectly lost: 0 bytes in 0 blocks
==20951==      possibly lost: 15,917 bytes in 38 blocks
==20951==    still reachable: 775,500 bytes in 12,686 blocks
==20951==                       of which reachable via heuristic:
==20951==                         stdstring          : 254 bytes in 2 blocks
==20951==         suppressed: 0 bytes in 0 blocks
==20951== Rerun with --leak-check=full to see details of leaked memory
==20951== 
==20951== Use --track-origins=yes to see where uninitialised values come from
==20951== For lists of detected and suppressed errors, rerun with: -s
==20951== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
Segmentation fault
```

krisztianloki, 2021-02-23:
I think the problem is that trying to initialize links before record initialization results in `cvt_dbaddr` returning NULL. So something like this "delays" link initialization until after record initialization (but `CONSTANT` and `JSON_LINK`s are initialized):
```diff
diff --git a/modules/database/src/ioc/db/dbLink.c b/modules/database/src/ioc/db/dbLink.c
index ef83368..a781286 100644
--- a/modules/database/src/ioc/db/dbLink.c
+++ b/modules/database/src/ioc/db/dbLink.c
@@ -49,6 +49,8 @@
 #include "recSup.h"
 #include "special.h"

+extern int recordsInitialized;
+
 /* How to identify links in error messages */
 const char * dbLinkFieldName(const struct link *plink)
 {
@@ -97,22 +99,24 @@ void dbInitLink(struct link *plink, short dbfType)
     /* Only initialize link once */
     if (plink->flags & DBLINK_FLAG_INITIALIZED)
         return;
-    else
-        plink->flags |= DBLINK_FLAG_INITIALIZED;

     if (plink->type == CONSTANT) {
+        plink->flags |= DBLINK_FLAG_INITIALIZED;
         dbConstInitLink(plink);
         return;
     }

     if (plink->type == JSON_LINK) {
+        plink->flags |= DBLINK_FLAG_INITIALIZED;
         dbJLinkInit(plink);
         return;
     }

-    if (plink->type != PV_LINK)
+    if (plink->type != PV_LINK || ! recordsInitialized)
         return;

+    plink->flags |= DBLINK_FLAG_INITIALIZED;
+
     if (plink == &precord->tsel)
         TSEL_modified(plink);

diff --git a/modules/database/src/ioc/misc/iocInit.c b/modules/database/src/ioc/misc/iocInit.c
index c26fd0e..b8c8d61 100644
--- a/modules/database/src/ioc/misc/iocInit.c
+++ b/modules/database/src/ioc/misc/iocInit.c
@@ -95,6 +95,8 @@ typedef void (*recIterFunc)(dbRecordType *rtyp, dbCommon *prec, void *user);

 static void iterateRecords(recIterFunc func, void *user);

+int recordsInitialized = 0;
+
 int dbThreadRealtimeLock = 1;
 epicsExportAddress(int, dbThreadRealtimeLock);

@@ -561,8 +563,10 @@ static void initDatabase(void)
 {
     dbChannelInit();
     iterateRecords(doInitRecord0, NULL);
+    recordsInitialized = 1;
     iterateRecords(doResolveLinks, NULL);
     iterateRecords(doInitRecord1, NULL);
+    recordsInitialized = 2;

     epicsAtExit(exitDatabase, NULL);
     return;
```

mdavidsaver, 2021-02-24:
Trying again, I'm not able to reproduce a crash.  In `getDoubleDouble()`, which argument is `NULL`?  I only see one call being made and neither `pto` nor `paddr->pfield` is `NULL`.

```gdb
Thread 1 hit Breakpoint 3, getDoubleDouble (paddr=(dbAddr*)(0x55d2d65476c0, "jaai.VAL"), pto=0x55d2d654f530, nRequest=5, 
    no_elements=5, offset=0) at ../db/dbConvert.c:831
831     static long getDoubleDouble GET_NOCONVERT(epicsFloat64, epicsFloat64)
(rr) p paddr->pfield
$4 = (void *) 0x55d2d6534cc0
```

Also, I'm not sure if it is relevant, but `getDoubleDouble()` isn't calling `memcpy()` in my builds.

anjohnson, 2021-02-24:
MacOS doesn't crash for me, but RHEL7 does, same stack trace as before (look at the earlier gdb one, it calls a memmove variant).

mdavidsaver, 2021-02-24:
> isn't calling `memcpy()` in my builds.

My mistake.  I was looking at the wrong function as GDB wasn't showing a frame for `copyNoConvert()`

I can replicate this crash with a RHEL7 container.  So a crash with gcc 4.8, but not with gcc 8.3 nor with clang?

anjohnson, 2021-02-24:
In my `dbGet()` call `paddr->pfield` is NULL:
```
(gdb) up
#3  0x00007ffff7917938 in dbGet (paddr=paddr@entry=0x6c7d40, 
    dbrType=dbrType@entry=10, pbuffer=pbuffer@entry=0x6c3510, 
    options=options@entry=0x0, nRequest=nRequest@entry=0x7fffffffd4c0, 
    pflin=pflin@entry=0x0) at ../db/dbAccess.c:1000
1000	            status = convert(paddr, pbuf, n, capacity, offset);
(gdb) print paddr
$1 = (DBADDR *) 0x6c7d40
(gdb) print *paddr
$2 = {precord = 0x69d7d0, pfield = 0x0, pfldDes = 0x6517f0, no_elements = 5, 
  field_type = 10, field_size = 8, special = 2, dbr_field_type = 10}
```

Here's what's happening:
Array record types have to allocate their array buffers in the first pass of record init. For most record types, the `dset::init_record()` routine is called in the second pass of record init. In the aai case though it gets called in the _first_ pass to allow input device types to do their own memory allocations if they wish. For most device support that's fine, but  the `devAaiSoft::init_record()` routine is responsible for allocating and initializing the array if INP holds a constant. In order for `dbLinkIsConstant(&prec->inp)` to work it first has to call `dbInitLink(&prec->inp, DBF_INLINK)`. When it is a constant that's fine, but in this case it isn't, the link gets initialized and copies the NULL buffer pointer from the other record. The order of initialization of the record instances matters here...

anjohnson, 2021-02-24:
BTW some of this involves my changes to get constant array initialization working in the aai record, so it's not necessarily Dirk's fault.

mdavidsaver, 2021-02-24:
Ah.  I see.  difference isn't compiler version, it's record iteration order.

```
# Debian 10
epics> dbl
compress
caai
jaai
aai
kill
```

```
# centos7
epics> dbl
caai
jaai
aai
compress
kill
```

mdavidsaver, 2021-02-24:
FYI. this specific issue would also be avoided if compressRecord kept `pfield=&val` like waveformRecord and aaiRecord now do.

mdavidsaver, 2021-02-24:
In Base, the two remaining record types which set `pfield=bptr` are compress and histogram.  Maybe changing this is the way to go since `pfield=&val` removes the ordering dependency between `dbDbInitLink()` and `init_record()`?

mdavidsaver, 2021-02-24:
> ... it's record iteration order.

Specifically, it is record type iteration order, as determined by dbd file order.  I guess this is determined in `dbdExpand.pl`?

anjohnson, 2021-02-24:
Don't quite follow, are suggesting moving the code that sets `paddr->pfield` from `cvt_dbaddr()` into `get_array_info()`? That might work, but it still has to set `pfield` to the buffer address somewhere and several record types use `bptr` to hold that pointer.

mdavidsaver, 2021-02-24:
> suggesting moving the code that sets `paddr->pfield` from `cvt_dbaddr()` into `get_array_info()`?

Yes, as is done by waveformRecord et al.

> it still has to set `pfield` to the buffer address somewhere

if "it" is record support code, then no.  Retaining the default of `pfield = (char*)precord + pflddes->offset` is sufficient.  I think that default comes from `dbAllocRecord()`.

krisztianloki, 2021-02-24:
If I understand things correctly a record type iteration order independent test would involve two aai records with their INP fields pointing at each other. The first to be initialized will cause a segfault when processed

mdavidsaver, 2021-02-24:
> independent test would involve two aai records

I think this would work just fine, and not crash.  The issue is specific to the combination of aaiRecord (or aaoRecord) targeting a DB link at a compressRecord (or I guess histogramRecord, or likely most external array-like record types).

anjohnson, 2021-02-24:
`dbdExpand.pl` uses a Perl hash to store the record types, and currently it outputs the definitions in whatever order Perl iterates through that hash. For security reasons I believe the hash order can vary.

Regarding setting `pfield`, my copy of waveformRecord.c contains this:
```C
static long get_array_info(DBADDR *paddr, long *no_elements, long *offset)
{
    waveformRecord *prec = (waveformRecord *) paddr->precord;

    paddr->pfield = prec->bptr;
    *no_elements = prec->nord;
    *offset = 0;

    return 0;
}
```
Unless you're talking about some other `pfield` I don't follow what you're saying at all.

mdavidsaver, 2021-02-24:
cf. 92d52cc415599b7dffa7df1f23d5ba8227fb4d3a I'm suggesting making an analogous change to compress and histogram.

anjohnson, 2021-02-24:
Ahhh, you're talking about the value of the `db_post_events()` pointer. Okay, I follow now and agree that this is probably the best way to solve this. Do you have time to handle that? I have meetings coming up and haven't had lunch yet...

mdavidsaver, 2021-02-24:
Let's see if @krisztianloki want's to have a go.

krisztianloki, 2021-02-24:
I will take a look at it tomorrow morning if that's okay with you

krisztianloki, 2021-02-25:
I've fixed compress, histogram, and subArray. I'm not sure if there are others that need fixing (lsi? printf?)
New test database: [compress.txt](https://github.com/epics-base/epics-base/files/6043495/compress.txt)
I created a modified Output.pm (modules/database/src/tools/DBD/Output.pm) that can put `aai` records on the top or bottom of the softIOC.dbd to easily trigger a segfault [Output.txt](https://github.com/epics-base/epics-base/files/6043503/Output.txt)

closed 2021-03-09 by anjohnson