epics-base/epics-base #106: Timers expire early

mark0n, 2021-02-05:
Generally, timers are supposed to sleep for _at least_ the amount of time they were set for. However, the current EPICS timer implementation behaves differently:
```
# Testing timer accuracy
ok  2 - timerCount == nTimers
ok  3 - 0.486553 < 5.000000, delay = 1.000000 s, error = -0.004866 s (0.5 %)
ok  4 - 0.440644 < 5.000000, delay = 1.100000 s, error = -0.004847 s (0.4 %)
ok  5 - 0.404795 < 5.000000, delay = 1.200000 s, error = -0.004858 s (0.4 %)
ok  6 - 0.373543 < 5.000000, delay = 1.300000 s, error = -0.004856 s (0.4 %)
ok  7 - 0.346924 < 5.000000, delay = 1.400000 s, error = -0.004857 s (0.3 %)
ok  8 - 0.326921 < 5.000000, delay = 1.500000 s, error = -0.004904 s (0.3 %)
ok  9 - 0.303757 < 5.000000, delay = 1.600000 s, error = -0.004860 s (0.3 %)
ok 10 - 0.285826 < 5.000000, delay = 1.700000 s, error = -0.004859 s (0.3 %)
ok 11 - 0.270031 < 5.000000, delay = 1.800000 s, error = -0.004861 s (0.3 %)
ok 12 - 0.255543 < 5.000000, delay = 1.900000 s, error = -0.004855 s (0.3 %)
ok 13 - 0.242964 < 5.000000, delay = 2.000000 s, error = -0.004859 s (0.2 %)
ok 14 - 0.231398 < 5.000000, delay = 2.100000 s, error = -0.004859 s (0.2 %)
ok 15 - 0.220428 < 5.000000, delay = 2.200000 s, error = -0.004849 s (0.2 %)
ok 16 - 0.211106 < 5.000000, delay = 2.300000 s, error = -0.004855 s (0.2 %)
ok 17 - 0.202014 < 5.000000, delay = 2.400000 s, error = -0.004848 s (0.2 %)
ok 18 - 0.194164 < 5.000000, delay = 2.500000 s, error = -0.004854 s (0.2 %)
ok 19 - 0.186316 < 5.000000, delay = 2.600000 s, error = -0.004844 s (0.2 %)
ok 20 - 0.181187 < 5.000000, delay = 2.700000 s, error = -0.004892 s (0.2 %)
ok 21 - 0.173014 < 5.000000, delay = 2.800000 s, error = -0.004844 s (0.2 %)
ok 22 - 0.167593 < 5.000000, delay = 2.900000 s, error = -0.004860 s (0.2 %)
ok 23 - 0.163954 < 5.000000, delay = 3.000000 s, error = -0.004919 s (0.2 %)
ok 24 - 0.156422 < 5.000000, delay = 3.100000 s, error = -0.004849 s (0.2 %)
ok 25 - 0.152041 < 5.000000, delay = 3.200000 s, error = -0.004865 s (0.2 %)
ok 26 - 0.149199 < 5.000000, delay = 3.300000 s, error = -0.004924 s (0.1 %)
ok 27 - 0.142764 < 5.000000, delay = 3.400000 s, error = -0.004854 s (0.1 %)
# average timer delay error -4.863972 ms
```
Note how the timers are consistently expiring ~5 ms early (I ran this on a state-of-the-art Linux system). Turns out for some esoteric reason half a quantum is subtracted in `timer::privateStart()`. If other architectures require this adjustment it should probably only be applied to these architectures or alternatively the quantum should return 0 on modern Linux systems. The existing tests do not fail since they have a very generous tolerance of +/-5% (-0% +5% would probably be a better choice).

Related issues have been discussed in the past (see [here](https://epics.anl.gov/core-talk/2019/msg00520.php) and [here](https://epics.anl.gov/tech-talk/2008/msg00987.php)) without any action taken. Considering the significant error and the popularity of Linux machines in the EPICS community I think this issue deserves to be addressed.

mdavidsaver, 2021-02-05:
[Another](https://bugs.launchpad.net/epics-base/+bug/1861612) relevant issue.  I've [abandoned](https://bugs.launchpad.net/epics-base/+bug/1861612/comments/31) my attempt to sort this out as I wasn't able to handled the RTEMS/vxWorks case without limiting the max timer rate to half to tick rate, which at least one user (@MarkRivers) is depending on.

mdavidsaver, 2021-02-05:
Long story short, I think the `-quantum/2` is a somewhat intentional hack covering for rounding issues on RTEMS, vxWorks, and WIN32.  Simply removing it is likely to add an extra tick of delay.

mark0n, 2021-02-05:
Maybe we should fix this for Linux (and possibly Windows) only? Both seem to be affected and their behavior seems to be well understood.

mark0n, 2021-02-05:
Also see [this Stack Exchange question](https://unix.stackexchange.com/questions/524917/find-out-timer-resolution-on-old-and-new-linux-systems).

mdavidsaver, 2021-02-05:
quoting https://bugs.launchpad.net/epics-base/+bug/1861612/comments/22

> ...
> 1. epicsEventWaitWithTimeout() doesn't function as specified on RTEMS, vxWorks, or Windows.
...
> 3. Windows is peculiar
> 
> The Windows impl. uses WaitForSingleObject() which takes a timeout in milli-seconds,
> but seems to behave as if it were internally doing a similar truncation.
> The delays I've observed seem to be round down to multiples of epicsThreadSleepQuantum().
> ...

The windows primitive we currently use was not intended to be used for high resolution timing.

FreddieAkeroyd, 2021-03-08:
Results for running PR #108 on my windows 10 desktop
```
epicsTimerTest.t ........... 1/58
not ok  1 - timer expired at the expected time (error = -7.011100 ms, tolerance = +16.625000/-0.0 ms)
epicsTimerTest.t ........... 2/58 not ok  3 - timer expired at the expected time (error = -7.344300 ms, tolerance = +16.625000/-0.0 ms)
not ok  5 - timer expired at the expected time (error = -7.312500 ms, tolerance = +16.625000/-0.0 ms)
not ok  7 - timer expired at the expected time (error = -7.323800 ms, tolerance = +16.625000/-0.0 ms)
epicsTimerTest.t ........... 8/58 not ok  9 - timer expired at the expected time (error = -7.601600 ms, tolerance = +16.625000/-0.0 ms)
not ok 11 - timer expired at the expected time (error = -7.568600 ms, tolerance = +16.625000/-0.0 ms)
not ok 13 - timer expired at the expected time (error = -7.600200 ms, tolerance = +16.625000/-0.0 ms)
not ok 14 - timer expired at the expected time (error = -7.448500 ms, tolerance = +16.625000/-0.0 ms)
epicsTimerTest.t ........... Dubious, test returned 1 (wstat 256, 0x100)
Failed 8/58 subtests
```
after removing `this->exp = expire - ( this->queue.notify.quantum () / 2.0 );` from timer::privateStart()
```
epicsTimerTest.t ........... 1/58
not ok  1 - timer expired at the expected time (error = -299.890100 ms, tolerance = +16.625000/-0.0 ms)
not ok  3 - timer expired at the expected time (error = -999.923900 ms, tolerance = +16.625000/-0.0 ms)
not ok  5 - timer expired at the expected time (error = -1199.924600 ms, tolerance = +16.625000/-0.0 ms)
not ok  7 - timer expired at the expected time (error = -1099.924700 ms, tolerance = +16.625000/-0.0 ms)
not ok  9 - timer expired at the expected time (error = -1099.874600 ms, tolerance = +16.625000/-0.0 ms)
not ok 11 - timer expired at the expected time (error = -1199.875200 ms, tolerance = +16.625000/-0.0 ms)
not ok 13 - timer expired at the expected time (error = -999.875300 ms, tolerance = +16.625000/-0.0 ms)
not ok 14 - timer expired at the expected time (error = -299.956000 ms, tolerance = +16.625000/-0.0 ms)
epicsTimerTest.t ........... 15/58 not ok 20 - 99.987390 < 5.000000, delay = 1.000000 s, error = -0.999874 s (100.0%)
not ok 21 - 99.988609 < 5.000000, delay = 1.100000 s, error = -1.099875 s (100.0 %)
not ok 22 - 99.989575 < 5.000000, delay = 1.200000 s, error = -1.199875 s (100.0 %)
not ok 23 - 99.990377 < 5.000000, delay = 1.300000 s, error = -1.299875 s (100.0 %)
not ok 24 - 99.991071 < 5.000000, delay = 1.400000 s, error = -1.399875 s (100.0 %)
not ok 25 - 99.991673 < 5.000000, delay = 1.500000 s, error = -1.499875 s (100.0 %)
not ok 26 - 99.992200 < 5.000000, delay = 1.600000 s, error = -1.599875 s (100.0 %)
not ok 27 - 99.992665 < 5.000000, delay = 1.700000 s, error = -1.699875 s (100.0 %)
not ok 28 - 99.993078 < 5.000000, delay = 1.800000 s, error = -1.799875 s (100.0 %)
not ok 29 - 99.993447 < 5.000000, delay = 1.900000 s, error = -1.899875 s (100.0 %)
not ok 30 - 99.993780 < 5.000000, delay = 2.000000 s, error = -1.999876 s (100.0 %)
not ok 31 - 99.994081 < 5.000000, delay = 2.100000 s, error = -2.099876 s (100.0 %)
not ok 32 - 99.994355 < 5.000000, delay = 2.200000 s, error = -2.199876 s (100.0 %)
not ok 33 - 99.994604 < 5.000000, delay = 2.300000 s, error = -2.299876 s (100.0 %)
not ok 34 - 99.994833 < 5.000000, delay = 2.400000 s, error = -2.399876 s (100.0 %)
not ok 35 - 99.995044 < 5.000000, delay = 2.500000 s, error = -2.499876 s (100.0 %)
not ok 36 - 99.995238 < 5.000000, delay = 2.600000 s, error = -2.599876 s (100.0 %)
not ok 37 - 99.995419 < 5.000000, delay = 2.700000 s, error = -2.699876 s (100.0 %)
not ok 38 - 99.995586 < 5.000000, delay = 2.800000 s, error = -2.799876 s (100.0 %)
not ok 39 - 99.995741 < 5.000000, delay = 2.900000 s, error = -2.899877 s (100.0 %)
not ok 40 - 99.995887 < 5.000000, delay = 3.000000 s, error = -2.999877 s (100.0 %)
not ok 41 - 99.996023 < 5.000000, delay = 3.100000 s, error = -3.099877 s (100.0 %)
not ok 42 - 99.996150 < 5.000000, delay = 3.200000 s, error = -3.199877 s (100.0 %)
not ok 43 - 99.996270 < 5.000000, delay = 3.300000 s, error = -3.299877 s (100.0 %)
not ok 44 - 99.996382 < 5.000000, delay = 3.400000 s, error = -3.399877 s (100.0 %)
not ok 50 - cancelVerify::expireCount == 0
epicsTimerTest.t ........... 52/58 not ok 52 - cancelVerify::expireCount == 0
epicsTimerTest.t ........... Dubious, test returned 1 (wstat 256, 0x100)
Failed 35/58 subtests
```

anjohnson, 2021-03-08:
Hi Freddie, you still need `this->exp = expire` just without the subtraction of half the quantum; was that what you meant?

FreddieAkeroyd, 2021-03-08:
Ah, I had commented out the whole line, It's late  :-D With just `this->exp = expire;` the test all pass, have ran them twice so far.

FreddieAkeroyd, 2021-03-09:
I think there may be a way to get a higher resolution on  epicsEventWaitWithTimeout(). A waitable timer configured through SetWaitableTimer() claims to have a 100ns resolution, so I think the following could work: 
-	Create a waitable timer for each thread, either on thread creation or in static TLS of the called function 
-	Use SetWaitableTimer() and then  WaitForMultipleObjects() on both event and timer
 
The same approach may also be possible with epicsThreadSleep() if a higher resolution was needed – so having the waitable timer as a thread resource may be best as it could also be used in epicsThreadSleep()

I'm happy to try this if it hasn't been looked at before, just because it lets you set a 100ns resolution may not mean you actually get it of course...

MarkRivers, 2021-03-09:
6 years ago I added code to asyn to allow setting the minimum timer period on Windows:
https://github.com/epics-modules/asyn/commit/181f85dd27ba6886a7234b16e6eebfbbdbf8448e

It has been changed since to avoid the use of shareLib.h.
https://github.com/epics-modules/asyn/blob/bc97ac15c36f216f89c833b143d779ace3eac356/asyn/miscellaneous/asynShellCommands.c#L1196

I said at the time that this functionality really belongs in base.  Since you are currently working on timers in base maybe now is the time to move it?

FreddieAkeroyd, 2021-03-09:
Hi Mark, thank you - I will take a look.  I have been cautious of `timeBeginPeriod` in the past as until very recently it affected all processes on a system not just the calling one e.g. https://randomascii.wordpress.com/2020/10/04/windows-timer-resolution-the-great-rule-change/

MarkRivers, 2021-03-09:
That article is very well written, and very useful.

FreddieAkeroyd, 2021-03-09:
I ran the test program
```
C:\devel\epics-base>waitable_timer.exe
Current timer resolution is 1.0 ms
Stopwatch frequency is already raised so some tests cannot be run.
Use powercfg /energy /duration 5 to find and close the program
that has raised the timer frequency.
1000 sleeps of 0.5 ms took 0.6 seconds.
Skipping tests without timeBeginPeriod because it's already called.

3. CREATE_WAITABLE_TIMER_HIGH_RESOLUTION is off - timeBeginPeriod is on
delay is 1000 us - slept for 1362.5 us
delay is 1000 us - slept for 730.0 us
delay is 1000 us - slept for 1771.7 us
delay is 100 us - slept for 689.9 us
delay is 100 us - slept for 751.4 us
delay is 100 us - slept for 732.1 us
delay is 10 us - slept for 756.1 us
delay is 10 us - slept for 734.3 us
delay is 10 us - slept for 739.2 us
delay is 1 us - slept for 737.4 us
delay is 1 us - slept for 733.9 us
delay is 1 us - slept for 731.7 us

4. CREATE_WAITABLE_TIMER_HIGH_RESOLUTION is on - timeBeginPeriod is on
delay is 1000 us - slept for 1126.1 us
delay is 1000 us - slept for 1700.9 us
delay is 1000 us - slept for 1697.2 us
delay is 100 us - slept for 598.6 us
delay is 100 us - slept for 588.4 us
delay is 100 us - slept for 588.8 us
delay is 10 us - slept for 572.5 us
delay is 10 us - slept for 78.0 us
delay is 10 us - slept for 584.5 us
delay is 1 us - slept for 67.1 us
delay is 1 us - slept for 73.3 us
delay is 1 us - slept for 78.6 us

C:\devel\epics-base>
```
which looks pretty variable even with a high resolution timer. I'm running windows 10 1803, would be interesting to know what  W10 2004 gave

FreddieAkeroyd, 2021-03-10:
So looking at https://groups.google.com/a/chromium.org/g/scheduler-dev/c/0GlSPYreJeY using the high resolution timer removes a dependency on `timeBeginPeriod`  and also looks to stop early wakeups. I'm note sure if there may be other `timeBeginPeriod` dependencies that need to be considered and could be affected when the setting changes from system wide to process wide

mark0n, 2021-03-11:
I tried using C++11's `std::this_thread::wait_for()` (which guarantees that timers don't expire early) on Linux and Windows. Works great on Linux (on average timers expire ~100 µs late) but that alone isn't sufficient to bring down the huge delta on Windows (on average timers expire on the order of 10 ms late there). However, combining `std::this_thread::wait_for()` with calling `timeBeginPeriod(1)` brought the average delta down to ~600 µs.

I tested on an old i7-6700 running Windows 10 20H2.

FreddieAkeroyd, 2021-03-11:
20H2 is very recent and you have just given us a nice demonstration of the new feature of no longer getting the best `timeBeginPeriod` any process on the system has asked for already. It would be interesting to know if `select()` is also affected in the same way as `std::this_thread::wait_for()`. The high resolution waitable timer doesn't need a call to `timeBeginPeriod` but the lower resolution one does - maybe that is what `std::this_thread::wait_for` is using?

FreddieAkeroyd, 2021-03-12:
@mark0n does `std::condition_variable::wait_for` behave well too with timeouts?

mdavidsaver, 2022-09-28:
Whenever I have attempted to do performance testing with [fast counter](https://github.com/epics-base/epics-base/files/9660277/spam.db.txt) records I have seen unexpectedly high results.  I finally put `2+2` and figured out why this sometimes gives me `20000`.

Here I set `ODLY`, waiting 2 seconds, then reading the measured counter rate.  Both unmodified (with the `- .quantum () / 2.0`), and removing it.  As can be seen, the effective count/scan rate is always high, and "saturates" somewhere between 150 and 200Hz.

[image]

On this (Linux) host:

```
$ getconf CLK_TCK
100
```

So `epicsThreadSleepQuantum()` returns `0.01`, and so a delay <=0.005 results in `expire` being in the past.  Some further playing around confirms that there is a sharp jump at 200 Hz (`ODLY=0.005`).

I also notice that this triggers the worst case of `timer::privateStart()`, which does an insertion sort from the far end.

anjohnson, 2022-09-28:
AI on ANJ: Try implementing [this suggestion](https://bugs.launchpad.net/epics-base/+bug/1861612/comments/18) to see if it helps.

mdavidsaver, 2023-01-09:
> $ getconf CLK_TCK
> 100

Looking further into where `100` comes from I find that on Linux the value returned for `_SC_CLK_TCK` actually comes from the kernel via the magic [auxiliary vector](http://articles.manugarg.com/aboutelfauxiliaryvectors.html) (cf. `AT_CLKTCK`).

```
$ /usr/bin/getconf CLK_TCK
100
$ LD_SHOW_AUXV=1 /bin/true | grep AT_CLKTCK
AT_CLKTCK:            100 
```

Looking further into the Linux kernel source.  (official/vanilla, not any of the RT forks)

```
$ git describe 
v5.10
$ git grep AT_CLKTCK
fs/binfmt_elf.c:        NEW_AUX_ENT(AT_CLKTCK, CLOCKS_PER_SEC);
fs/binfmt_elf_fdpic.c:  NEW_AUX_ENT(AT_CLKTCK,  CLOCKS_PER_SEC);
include/uapi/linux/auxvec.h:#define AT_CLKTCK 17        /* frequency at which times() increments */
$ git grep 'define\s*CLOCKS_PER_SEC'
arch/alpha/include/asm/param.h:# define CLOCKS_PER_SEC  USER_HZ /* frequency at which times() counts */
arch/ia64/include/asm/param.h:# define CLOCKS_PER_SEC   HZ      /* frequency at which times() counts */
include/asm-generic/param.h:# define CLOCKS_PER_SEC     (USER_HZ)       /* in "ticks" like times() */
$ git grep 'define\s*USER_HZ'
arch/alpha/include/asm/param.h:# define USER_HZ 1024
arch/ia64/include/asm/param.h:# define USER_HZ  HZ
include/asm-generic/param.h:# define USER_HZ    100             /* some user interfaces are */...
```

The definitions in `include/asm-generic/param.h` is unconditional.  I don't find either of these macros defined in any generated header for amd64.

As I read this, with the exception of alpha and perhaps ia64, `sysconf(_SC_CLK_TCK)` is a constant 100 regardless of tick timer configuration.  To my mind, this means our usage of `_SC_CLK_TCK` in `epicsThreadSleepQuantum()` is broken on Linux.

anjohnson, 2025-02-19:
Core Group: The tick adjustment in the timer code is only needed for embedded OSs that have a relatively coarse tick timer. We think this issue should disappear when we drop RTEMS-4 and VxWorks, so in the 7.1 series, when there will need to be a fix applied.