epics-base/epics-base #116: Conflict with libpcap on Centos-8

anjohnson, 2021-03-03:
A build of the synApps MCA module R7-8 on Centos-8 died with this error, duplicate definitions of `SOCKET` and `INVALID_SOCKET` in libpcap-1.9.1 and osiSock.h:

```
make[4]: Entering directory '/home/anj/ioc-modules/build/MCA/7.8-C2-2+WEED-001/mcaApp/CanberraSrc/O.linux-x86_64'
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE         -DUSE_LIBNET -g `libnet-config --defines`    -D_X86_64_  -DUNIX  -Dlinux      -O3   -Wall     -DUSE_LIBNET -g `libnet-config --defines`   -mtune=generic      -m64 -fPIC -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include   -I/home/anj/ioc-modules/build/CALC/3.7.2-C2-1+WEED-001/include   -I/home/anj/ioc-modules/build/SSCAN/2.11.2-C2-1+WEED-001/include   -I/home/anj/ioc-modules/build/SEQ/2.2.6-C2-2+WEED-001/include   -I/home/anj/ioc-modules/build/SCALER/4.0-C2-1+WEED-001/include   -I/home/anj/ioc-modules/build/BUSY/1.7.1-C2-1+WEED-001/include  -I/home/anj/ioc-modules/build/AUTOSAVE/5.10-C2-1+WEED-001/include/os/Linux -I/home/anj/ioc-modules/build/AUTOSAVE/5.10-C2-1+WEED-001/include   -I/home/anj/ioc-modules/build/ASYN/4.40-C2-2+WEED-001/include -I/home/anj/ioc-modules/build/BASE/7.0.4-C2-1+WEED-001/include/compiler/gcc -I/home/anj/ioc-modules/build/BASE/7.0.4-C2-1+WEED-001/include/os/Linux -I/home/anj/ioc-modules/build/BASE/7.0.4-C2-1+WEED-001/include        -c ../nmc_comm_subs_1.c
In file included from /usr/include/pcap/pcap.h:87,
                 from /usr/include/pcap.h:43,
                 from ../nmc_sys_defs.h:57,
                 from ../nmc_comm_subs_1.c:72:
/home/anj/ioc-modules/build/BASE/7.0.4-C2-1+WEED-001/include/os/Linux/osdSock.h:32:33: error: two or more data types in declaration specifiers
 typedef int                     SOCKET;
                                 ^~~~~~
In file included from /home/anj/ioc-modules/build/BASE/7.0.4-C2-1+WEED-001/include/osiSock.h:19,
                 from ../nmc_comm_subs_1.c:103:
/home/anj/ioc-modules/build/BASE/7.0.4-C2-1+WEED-001/include/os/Linux/osdSock.h:32:1: warning: useless type name in empty declaration
 typedef int                     SOCKET;
 ^~~~~~~
/home/anj/ioc-modules/build/BASE/7.0.4-C2-1+WEED-001/include/os/Linux/osdSock.h:33: warning: "INVALID_SOCKET" redefined
 #define INVALID_SOCKET          (-1)
 
In file included from /usr/include/pcap/pcap.h:87,
                 from /usr/include/pcap.h:43,
                 from ../nmc_sys_defs.h:57,
                 from ../nmc_comm_subs_1.c:72:
/usr/include/pcap/socket.h:89: note: this is the location of the previous definition
     #define INVALID_SOCKET -1
```

Mark Rivers has fixed this in MCA R7-9 with a `#undef SOCKET` immediately after the `#include <pcap.h>`, but this shows an issue with our osi/os/Linux/osiSock.h header which we should probably address. Those definitions are new since libpcap-1.5.3 as was shipped with RHEL-7.

mdavidsaver, 2021-03-03:
cf. https://github.com/the-tcpdump-group/libpcap/blob/master/pcap/socket.h

```c
  #ifndef SOCKET
    #define SOCKET int
  #endif
...
  #ifndef INVALID_SOCKET
    #define INVALID_SOCKET -1
  #endif
```

mdavidsaver, 2021-03-03:
pcap has a `#define` while Com has a typedef.

https://github.com/epics-base/epics-base/blob/f41276bef8e1a98da18f146428ad2c1ada6a9b6b/modules/libcom/src/osi/os/Linux/osdSock.h#L36-L37

closed 2022-09-28 by anjohnson