epics-base/epics-base #137: Avoid intentional memory leak in `epicsEnvSet()`

mdavidsaver, 2021-03-10:
`epicsEnvSet()` has different implementations, some of which use `setenv()`, and some `putenv()`.  Those using `putenv()` currently leak on each call since the buffer passed to `putenv()` may be referenced by the libc for some indeterminate time.  Currently this is the case for: default, WIN32, and vxWorks.  The others use `setenv()`

The default impl (used by eg. Linux) could probably be switched to use `setenv()` at this point.  Windows has [`_putenv_s`](https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/putenv-s-wputenv-s?view=msvc-160) which seems to work like `setenv()`.  @anjohnson says that with at least recent vxWorks (>=6.8), `putenv()` is making a copy of the provided buffer, which we could then free.

```
$ git ls-files|grep osdEnv
modules/libcom/src/osi/os/Darwin/osdEnv.c
modules/libcom/src/osi/os/RTEMS/osdEnv.c
modules/libcom/src/osi/os/WIN32/osdEnv.c
modules/libcom/src/osi/os/default/osdEnv.c
modules/libcom/src/osi/os/iOS/osdEnv.c
modules/libcom/src/osi/os/solaris/osdEnv.c
modules/libcom/src/osi/os/vxWorks/osdEnv.c
```

dirk-zimoch, 2021-03-11:
> @anjohnson says that with at least recent vxWorks (>=6.8), putenv() is making a copy of the provided buffer, which we could then free.
The same was already true for vxWorks 5. No need to keep a copy of the buffer in vxWorks.

dirk-zimoch, 2021-03-11:
From the VxWorks 5.5 manual:

>    STATUS putenv  ( char * pEnvString  )
>
>    This routine sets an environment variable to a value by altering an existing variable or creating a new one. The parameter points to a string of the form "variableName=value". Unlike the UNIX implementation, the string passed as a parameter is copied to a private buffer.
> RETURNS
>    OK, or ERROR if space cannot be malloc'd.

closed 2021-06-02 by anjohnson