epics-base/epics-base #155: Apply some GCC attributes

mdavidsaver, 2021-04-06:
GCC (and usually clang) has a number of `__attribute__(())`s which it might be useful to (conditionally) apply to some Base functions to facilitate static analysis/optimization by the compiler.

https://gcc.gnu.org/onlinedocs/gcc/Common-Function-Attributes.html#Common-Function-Attributes

eg. `alloc_size(...)` to provide size hints, or `malloc(deallocator, ptr-index)` to match alloc and dealloc functions, could be added to various memory allocation functions.  eg.

```c
__attribute__((malloc(freeListFree,2)))
void* freeListMalloc(void *pvt)

__attribute__((alloc_size(1,2),malloc(free,1),returns_nonnull))
void* callocMustSucceed(size_t count, size_t size,  const char *errorMessage);
```

Some of the simpler, or more common, might be abstracted through `compilerDependencies.h` (eg. `deprecated/error/warning` or `warn_unused_result`).  Though the more specialized/complicated probably have to be special cases.

Complications include identifying supported GCC versions ranges.  (although GCC usually treats unknown attributes as a warning).

FreddieAkeroyd, 2021-04-07:
Visual Studio has some similar notation  https://docs.microsoft.com/en-us/cpp/code-quality/using-sal-annotations-to-reduce-c-cpp-code-defects?view=msvc-160 but I think it is just for static analysis. I'm not sure if it is possible to combine the two notations in a common way

mdavidsaver, 2021-04-07:
Are the MSVC annotations meant to be used on declarations, definitions, or both?  The examples I've seen so far are on definitions.  The GCC annotations are exclusively for declarations, and it would actually be nice if the MSVC were definitions only as this could avoid some extra clutter in headers.

Also, the MSVC notation examples look more preprocessor friendly (no extra commas).  So it might be easier to add than the GCC notation, which might need `#ifdef __GNUC__` in more cases.

FreddieAkeroyd, 2021-04-07:
The examples on https://docs.microsoft.com/en-us/cpp/code-quality/best-practices-and-examples-sal?view=msvc-160 seem to have a mixture of use on declarations and definitions. I'm afraid i've never actually used this annotations, it may be some parts  only make sense on the definition, some may work well on the declaration and allow static analysis to check both caller usage and implementation. Maybe you can annotate both declaration and definition separately and it combines the relevant bits during static analysis

dirk-zimoch, 2022-05-02:
Different gcc versions support different sets of attributes. On older compilers you may end up with lots of warnings like `warning: 'deprecated' attribute directive ignored`. Have a method to check if the attribute is actually available, That may mean to check for each attribute since which gcc version is is supported. A simple `#ifdef __GNUC__` is definitely not sufficient.

mdavidsaver, 2022-05-02:
@dirk-zimoch If this is a nuisance to you, can you help with investigating the history GCC of attributes?

eg. I know that the `__attribute__((deprecated("message")))` form with a message needs at least GCC >= 4.4.5.  As I recall, I chose this by looking through old version of the GCC manual for the first version which mentioned this variant.   From pvDataCPP:

```
#if defined(PVD_INTERNAL)
#  define PVD_DEPRECATED(msg)
#elif __GNUC__ > 4 || __GNUC__ == 4 && __GNUC_MINOR__ >= 5
#  define PVD_DEPRECATED(msg) __attribute__((deprecated(msg)))
#else
#  define PVD_DEPRECATED(msg) EPICS_DEPRECATED
#endif
```

dirk-zimoch, 2022-05-02:
For `deprecated` `__GNUC__ > 2` seems sufficient. For `deperated(msg)` I need to test.
Write down a list of attributes you plan to use and I will have a look during the Codeathon.
I typically compile for a wide range of architectures, including old Linuxes and VxWorks 5 but as well newer ones with gcc 11.2.0. Windows via wine too.

dirk-zimoch, 2022-05-09:
I have checked the gcc manuals here https://gcc.gnu.org/onlinedocs/.
`deprecated` appears first in gcc version 3.1 while `deprecated(msg)` is new in 4.5 as you already found out. 
`format` and `format_arg` are already there in version 2.95.3.
`malloc` shows up in 3.0 but has no arguments. Those appear first only in version 11.3.
`alloc_size` is available since verision 4.3.
`returns_nonnull` shows up first in version 4.9.

RAYemelyanova, 2023-03-07:
I am interested in doing this for the codeathon. I have little experience with gcc though, so I am using this more as a learning experience. If anyone else wants to join in, please let me know! Thanks

RAYemelyanova, 2023-03-07:
Currently looking at implementing these for the `callocMustSucceed` function above. Looking through the [documentation](https://www.rowleydownload.co.uk/arm/documentation/gnu/gcc/Common-Function-Attributes.html) for the malloc attribute,

> Attribute malloc indicates that a function is malloc-like, i.e., that the pointer P returned by the function cannot alias any other pointer valid when the function returns, and moreover no pointers to valid objects occur in any storage addressed by P. In addition, the GCC predicts that a function with the attribute returns non-null in most cases.

@mdavidsaver Does this mean we do not technically need the attribute returns_nonnull? Or would you prefer to have it anyways just incase?

RAYemelyanova, 2023-03-07:
^ talked to David about this, and looked at the returns_nonnull attribute. Technically malloc can return null, in which case we should not be using it in `callocMustSucceed` or `mallocMustSucceed`.

closed 2025-02-19 by anjohnson