epics-base/epics-base #167: caRepeater init not using `FINAL_LOCATION`

mdavidsaver, 2021-05-07:
`S99caRepeater@` and `caRepeater.service@` in `modules/ca/src/client/` are being expanded with `INSTALL_BIN` based on `INSTALL_LOCATION` instead of `FINAL_LOCATION`.

mdavidsaver, 2021-05-07:
Correcting this would mean making a third copy of the `FINAL_LOCATION ?=` definition.  (after `modules/database/src/std/softIoc/Makefile` and `modules/database/src/tools/Makefile`)

@anjohnson Do you think it is worthwhile yet to promote `FINAL_LOCATION` to `configure/`?

anjohnson, 2021-05-12:
Sure, it should be included in `configure/CONFIG_SITE` just below the commented-out `INSTALL_LOCATION`; the description for it there needs to be understandable so new users will set the right one (adjust the existing wording if necessary to clarify).

I know we're deprecating it and planned to remove our log-server sometime, but the same problem exists with the `S99logServer` script installed from `modules/libcom/src/log`.

mdavidsaver, 2021-05-23:
Now fixed by #170.

Worth noting that the example inits are running things as root, which is unnecessary for the functions being performed.  Also, the sysv scripts are exceptionally minimal, eg. without `status`.  So not what I would consider examples of current best practice.

closed 2021-07-21 by anjohnson