epics-base/epics-base #169: Cross builds not setting SONAME

mdavidsaver, 2021-05-12:
While shared libraries are always installed with the `SHRLIB_VERSION` suffix, this is only being used on a couple of host targets where the SONAME is actually being set.

```sh
$ readelf -d lib/linux-x86_64/libCom.so.3.19.0 |grep SONAME
 0x000000000000000e (SONAME)             Library soname: [libCom.so.3.19.0]
$ readelf -d lib/linux-aarch64/libCom.so.3.19.0 |grep SONAME
```

It looks like the `-Wl,-h` is only passed for certain host builds.  Maybe this can move into `configure/os/CONFIG.Common.linuxCommon`?

```make
$ git grep '\-Wl,-h'
configure/os/CONFIG.linux-x86.linux-x86:SHRLIB_LDFLAGS += -Wl,-h$@
configure/os/CONFIG.linux-x86.linux-x86:LOADABLE_SHRLIB_LDFLAGS += -Wl,-h$@
configure/os/CONFIG.solaris-sparc-gnu.solaris-sparc-gnu:SHRLIB_LDFLAGS += -Wl,-z,defs -Wl,-z,text -Wl,-h,$@
configure/os/CONFIG.solaris-sparc-gnu.solaris-sparc-gnu:LOADABLE_SHRLIB_LDFLAGS += -Wl,-z,text -Wl,-h,$@
```

anjohnson, 2021-05-12:
Unfortunately I suspect the assignments in `configure/CONFIG.gnuCommon` will override their values if you set `SHRLIB_LDFLAGS` and `LOADABLE_SHRLIB_LDFLAGS` in `configure/os/CONFIG.Common.linuxCommon`. I think we will have to duplicate those settings in all of the `configure/os/CONFIG.linux-*.linux-*` files for them to work properly. I have no objection to our doing that though.

mdavidsaver, 2021-05-13:
Right, I keep forgetting how late gnuCommon is included.  (how do you keep this straight?)

Still, does this mean that `-Wl,-h` could just be moved to gnuCommon?  It looks like the targets with different linkers already provide alternate definitions.

```
$ git grep '^\s*SHRLIB_LDFLAGS\s*=' -- configure/
configure/CONFIG.gnuCommon:SHRLIB_LDFLAGS = -shared -fPIC
configure/os/CONFIG.Common.iosCommon:SHRLIB_LDFLAGS = -dynamiclib -flat_namespace -undefined suppress \
configure/os/CONFIG.Common.vxWorksCommon:SHRLIB_LDFLAGS =
configure/os/CONFIG.cygwin-x86.cygwin-x86:SHRLIB_LDFLAGS = -shared -Wl,--out-implib,$(LIB_PREFIX)$*$(LIB_SUFFIX)
configure/os/CONFIG.darwinCommon.darwinCommon:SHRLIB_LDFLAGS = -dynamiclib -flat_namespace -undefined dynamic_lookup \
configure/os/CONFIG.linux-x86.win32-x86-mingw:SHRLIB_LDFLAGS = -shared \
configure/os/CONFIG.solarisCommon.solarisCommon:SHRLIB_LDFLAGS = -z defs -G -h $@ -z text
configure/os/CONFIG.win32-x86-mingw.win32-x86-mingw:SHRLIB_LDFLAGS = -shared \
```

mdavidsaver, 2021-05-13:
> ... duplicate those settings in all of the `configure/os/CONFIG.linux-*.linux-*` files ...

```sh
$ ls configure/os/CONFIG.linux-*.linux-*| wc -l
24
```

anjohnson, 2021-05-13:
I didn't mean to suggest that all the `CONFIG.linux-*.linux-*` files should have the setting added. Not quite sure why but I'm wary of cross-building shared libraries, and I suspect all our cross-targets may have them turned off. Your PR #171 looks like a reasonable fix.

mdavidsaver, 2021-05-13:
> ,,, Not quite sure why but I'm wary of cross-building shared libraries ,,,

This leaves me not quite sure how to help you with this :)

> ,.. I suspect most cross-targets may have them turned off.

Perhaps why no one has noticed (and cared) about this previously.

PR #171 (2021-05-13): set SONAME for cross builds
Attempt to fix #169.
closed 2021-05-13 by epics-jenkins