epics-base/epics-base #174: bo and mbbo records don't raise STATE_ALARM in init_record()

anjohnson, 2021-06-01:
The bo and mbbo record types don't raise a `STATE_ALARM` in their `init_record()` routines even if the `*SV` field corresponding to the current state is set. It is often undesirable to set PINI on output records (which would also fix this), to avoid writing to hardware unnecessarily.

Other record types might have similar issues with their initial states.
```
woz$ cat do-init-sev.db 
record(bo, bo) {
  field(ZSV, MINOR)
  field(ZNAM, Zero)
  field(VAL, 0)
}

record(mbbo, mbbo) {
  field(ZRSV, MINOR)
  field(ZRST, Zero)
  field(VAL, 0)
}
```

```
woz$ caget -a bo mbbo
bo                             <undefined> Zero UDF NO_ALARM
mbbo                           <undefined> Zero UDF NO_ALARM
```

After processing both records:
```
woz$ caget -a bo mbbo
bo                             2021-06-01 12:40:27.046965 Zero STATE MINOR
mbbo                           2021-06-01 12:40:41.207460 Zero STATE MINOR
```

Noticed while investigating [epics-modules/asyn#136](/epics-modules/asyn/issues/136).

These kinds of changes to record types should probably be discussed and agreed (on core-talk or tech-talk) before making them though.

anjohnson, 2021-06-01:
This does require `init_record()` to call `recGblResetAlarms(prec)` but I'm starting to think it should be doing that anyway.

anjohnson, 2021-06-03:
This particular issue also applies to the ao, longout and int64out output record types which have level alarms (HIGH, HIHI etc.) that probably should be raised based on the initial contents of VAL.

The Core Developers group discussed this, and I proposed that the affected output record types should export their `checkAlarms()` routines as for example `void boRecord_checkAlarms(boRecord *)` for bo-specific device support to call directly, but @mdavidsaver requested that a new method be added to the RSET for this instead, so it could be more easily called by generic device support code. The only Base record types that would need this method are ao, bo, mbbo, longout and int64out.

I have the necessary changes for those record types, but I'm not convinced that a new RSET routine is really the right place for this. There is other record-specific code that device support could also make use of instead of having to copy it out of the record, e.g. any ai device support that sets VAL is also supposed to apply the SMOO filter to that value itself, since the record normally does that inside `convert()` which isn't called if the `read_ai()` returns 2.

These kinds of routines would all be heavily record-specific and would probably need to be called at different times. We wouldn't want each record type to be allowed to add its own special methods to RSET, and other than the above `checkAlarms()` routine which is common to several (but not all) output record types I suspect there wouldn't be much commonality. That's why I still think it makes more sense for each record type to directly export any routines that device support might need, or for us to add some kind of generic `ioctl()`-like API to the RSET (which wouldn't really help a generic device support very much either).

Should I just give this idea up completely and make @MarkRivers copy more record code into Asyn?

MarkRivers, 2021-06-03:
Adding a new boRecord_checkAlarms(boRecord *) or a new RSET would only help with new versions of base, and asyn is intended to be used with anything 3.14 and later.  So maybe putting it in asyn is the best solution?

anjohnson, 2021-06-03:
If you did that in Asyn I would still have to apply the changes that call `checkAlarms()` and `recGblResetAlarms()` from the record's `init_record()` routine so the behavior is the same whether the read-back alarm check happens synchronously at record init or asynchronously through your callback mechanism. Actually that might be a reason to _not_ add the code to Asyn, so these initial alarms would only happen when running on newer versions of Base. We could severely confuse our users otherwise (and I now hope that I'm not severely confused instead).

MarkRivers, 2021-06-03:
I'll trust that you are less confused than I, and leave it to your judgement.  These are corner cases, and Torsten was the first one to notice a problem, which has existed for many years.

anjohnson, 2022-09-28:
Core Group 9/28: Closing until someone comes up with a worthwhile solution.

closed 2022-09-28 by anjohnson