epics-base/epics-base #175: RTEMS5 network issues/TODOs

mdavidsaver, 2021-06-15:
A catalog of issues I'm seeing with an RTEMS5 build.

## FHI defaults

These defaults shouldn't be hidden away in C code.

https://github.com/epics-base/epics-base/blob/256babf9614a199709e2728f0ecd79bf57b668ec/modules/libcom/RTEMS/posix/rtems_init.c#L96-L99

## Timeouts

DHCP has a timeout, but NTP fetch does not.  imo. this should be the other way around.  This is problematic for me since with QEMU 3.1.0 at least, `-net user` the DHCP server doesn't seem to respond to `option ntp_servers`.  (may be related by the next item)

## DHCP option names

There may be an issue with `option ntp_servers` vs. `options ntp-servers`.  cf. `freebsd/sbin/dhclient/tables.c` in the libbsd source, or `/etc/dhcp/dhclient.conf`.

https://github.com/epics-base/epics-base/blob/256babf9614a199709e2728f0ecd79bf57b668ec/modules/libcom/RTEMS/posix/rtems_init.c#L819-L822

## NIC emulation

`-net nic,model=e1000` works for me... eventually.

`-net nic,model=rtl8139` doesn't work for me.  I see `reason=CARRIER`, but never `reason=BOUND`.

mdavidsaver, 2021-06-15:
wrt. `-net nic,model=rtl8139`, it's not straightforward to troubleshoot, but I don't think the QEMU DHCP server ever sees a request.

```
qemu-system-i386 -m 64 -no-reboot -serial stdio -display none -net nic,model=rtl8139 -net nic,model=ne2k_pci \
 -net user -append --console=/dev/com1 -kernel epicsTypesTest
```

Also, I do eventually see a `reason=IPV4LL` with `new_ip_address=169.254.205.11`.  This looks like dhcpcd timing out and using a auto-config address, which I don't think we ever want.

mdavidsaver, 2021-06-23:
`-mtune=pentiumpro -march=pentium` from `$(CPU_FLAGS)` in the RTEMS makefiles shows up in `CPPFLAGS` instead of `CFLAGS`/`CXXFLAGS` where I would expect to find them.  I don't think this will break anything though.

```sh
$ make -s -C configure/O.RTEMS-pc686-qemu PRINT.CPPFLAGS
CPPFLAGS =   -mtune=pentiumpro -march=pentium       -D_GNU_SOURCE -D_DEFAULT_SOURCE            -DUNIX   
$ make -s -C configure/O.RTEMS-pc686-qemu PRINT.CFLAGS
CFLAGS =   -O2 -g -ffunction-sections -fdata-sections  -pipe -Wall -Werror-implicit-function-declaration     -Werror=implicit-function-declaration      -D__LINUX_ERRNO_EXTENSIONS__
$ make -s -C configure/O.RTEMS-pc686-qemu PRINT.CXXFLAGS
CXXFLAGS =   -O2 -g -ffunction-sections -fdata-sections  -pipe -Wall          -D__LINUX_ERRNO_EXTENSIONS__
```

mdavidsaver, 2021-06-29:
[One](https://github.com/mdavidsaver/epics-base/runs/2938789923?check_suite_focus=true#step:10:153) of my test builds had an unexpected failure.

```
  ------------------------------------------------------------------------------
  /home/runner/work/epics-base/epics-base/modules/libcom/test/O.RTEMS-pc686-qemu
...
  epicsMessageQueueTest.tap  (Wstat: 0 Tests: 73 Failed: 0)
    Parse errors: Tests out of sequence.  Found (66) but expected (65)
                  Tests out of sequence.  Found (67) but expected (66)
                  Tests out of sequence.  Found (68) but expected (67)
                  Tests out of sequence.  Found (69) but expected (68)
                  Tests out of sequence.  Found (70) but expected (69)
```

```sh
$ git ls-files|grep epicsMes
modules/libcom/src/osi/epicsMessageQueue.cpp
modules/libcom/src/osi/epicsMessageQueue.h
modules/libcom/test/epicsMessageQueueTest.cpp
```

anjohnson, 2021-06-29:
That part of the `.tap` file is shown below. There's a timeout message at the relevant point which I assume must have come from your additions and is missing a newline at the end, so the Perl test harness doesn't recognize the `ok 65 - ...` test result which follows it:

```
# sleepySender: sending every 0.010 seconds
ok 63 - Sent 500 (should be 500)
ok 64 - Received 500 (should be 500)
# sleepySender: sending every 0.011 seconds
timed outallowing 8 seconds for IPv4LL timeouttimed outok 65 - Sent 500 (should be 500)
ok 66 - Received 500 (should be 500)
```

mdavidsaver, 2021-06-29:
> There's a timeout message at the relevant point which I assume must have come from your additions

Well, yes.  I suppose enabling the test to run can be seen as an addition on my part ;)

> timed outallowing 8 seconds for IPv4LL timeouttimed out ...

This is likely coming from the dhcp client.  I'll look into why it isn't going through same redirection as the other logging from libbsd.

anjohnson, 2021-06-29:
Sorry, without actually looking back at the committed code I guessed the timeouts had come from your [epicsUnitTest: add $EPICS_TEST_TIMEOUT](https://github.com/mdavidsaver/epics-base/commit/24b44ac8791e449011dc98894b8803485de59ca1) changes.

mdavidsaver, 2021-06-29:
Looks like `syslog()` on RTEMS uses `printf()`.  So any local output goes to `stdout`, and I can't see a way to redirect this to errlog.  I guess it needs to be disabled at least for unittests (cf. `setlogmask()`).

```sh
$ grep -R 'seconds for.*timeout'
dhcpcd/dhcpcd.c:        syslog(LOG_WARNING, "allowing %d seconds for IPv4LL timeout", timeout);
```

From `cpukit/libnetworking/lib/syslog.c` in the RTEMS tree, it looks like `syslog()` should add a newline when printing, but this doesn't seem to happen.  At least I don't see any stray newlines later in the log, so this doesn't look like a synchronization issue.

mdavidsaver, 2021-06-30:
`setlogmask()` does not seem to have the desired effect...

mdavidsaver, 2024-06-19:
My list is likely out of date by now.  I am going to tag this issue as "help wanted" as I don't have a plan to revisit.

closed 2025-02-19 by anjohnson