epics-base/epics-base #178: epicsAtomic TODOs

mdavidsaver, 2021-06-23:
It looks like epicsAtomic operations for clang targets always use the (inefficient) interlocked fallback.  A first task would be to investigate which clang versions provide the GCC `__sync_*` and/or `__atomic_*` builtins.

https://github.com/epics-base/epics-base/blob/1cacd058cd3bd1d9b6cabc6219c48217913eb1e8/modules/libcom/src/osi/compiler/clang/epicsAtomicCD.h#L17-L24

GCC >=4.8 adds the [`__atomic_*`](https://gcc.gnu.org/onlinedocs/gcc-4.7.1/gcc/_005f_005fatomic-Builtins.html#_005f_005fatomic-Builtins) builtins.  The [`__sync_*`](https://gcc.gnu.org/onlinedocs/gcc-4.7.1/gcc/_005f_005fsync-Builtins.html#_005f_005fsync-Builtins) are then described as "legacy", but not deprecated.  It would be worth investigating if switching to `__atomic_*` would have any immediate benefit, or if we would only be getting ahead of an eventual deprecation.

https://github.com/epics-base/epics-base/blob/1cacd058cd3bd1d9b6cabc6219c48217913eb1e8/modules/libcom/src/osi/compiler/gcc/epicsAtomicCD.h#L115-L118

anjohnson, 2021-06-23:
Agreed, I was surprised to see the other day that the Clang implementations aren't using the built-ins, so most IOCs running on Apple devices probably aren't making the best use of the OS & CPU's capabilities.

mdavidsaver, 2021-06-24:
```
$ clang -E -dM -x c /dev/null | grep SYNC
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1
$ clang --version
clang version 7.0.1-8+deb10u2 (tags/RELEASE_701/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

anjohnson, 2021-06-24:
```
woz$ clang --version
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
woz$ clang -E -dM -x c /dev/null | grep SYNC
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_16 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1
```

anjohnson, 2021-06-24:
RHEL-7
```
tux% clang --version
clang version 3.4.2 (tags/RELEASE_34/dot2-final)
Target: x86_64-redhat-linux-gnu
Thread model: posix
tux% clang -E -dM -x c /dev/null | grep SYNC
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1
```

anjohnson, 2021-06-24:
RHEL-8
```
$ clang --version
clang version 10.0.1 (Red Hat 10.0.1-1.module+el8.3.0+7459+90c24896)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /bin
$ clang -E -dM -x c /dev/null | grep SYNC
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_8 1
```

dirk-zimoch, 2021-06-25:
SL-6
```
$ clang --version
clang version 2.8 (branches/release_28)
Target: x86_64-redhat-linux-gnu
Thread model: posix
$ clang -E -dM -x c /dev/null |  egrep 'ATOMIC|SYNC'
#define __SIG_ATOMIC_WIDTH__ 32
```

anjohnson, 2021-08-11:
Did we fix this issue with the #192 branch which just got merged?

anjohnson, 2021-08-16:
I noticed earlier that the APS Jenkins build of RTEMS-4.10.2 for the uC5282 (m68k) has been failing since #192 was merged:
```
/usr/local/vw/rtems/rtems-4.10.2/bin/m68k-rtems4.10.2-gcc --pipe -B/usr/local/rtems/rtems-4.10.2/m68k-rtems4.10.2/uC5282/lib/ -specs bsp_specs -qrtems   -fasm   -mcpu=5282     -D_GNU_SOURCE -D_DEFAULT_SOURCE            -DUNIX      -O2 -g -g   -Wall -Werror-implicit-function-declaration       -DMY_DO_BOOTP=NULL    -D__LINUX_ERRNO_EXTENSIONS__ -DRTEMS_LEGACY_STACK  -I. -I../O.Common -I. -I../osi/compiler/gcc -I../osi/compiler/default -I. -I../osi/os/RTEMS-score -I../osi/os/RTEMS -I../osi/os/posix -I../osi/os/default -I.. -I../as -I../bucketLib -I../calc -I../cvtFast -I../cppStd -I../cxxTemplates -I../dbmf -I../ellLib -I../env -I../error -I../fdmgr -I../flex -I../freeList -I../gpHash -I../iocsh -I../log -I../macLib -I../misc -I../osi -I../pool -I../ring -I../taskwd -I../timer -I../yacc -I../yacc -I../yajl -I../../../../include/compiler/gcc -I../../../../include/os/RTEMS -I../../../../include         -c ../osi/os/RTEMS-score/osdThread.c
In file included from ../osi/compiler/gcc/epicsAtomicCD.h:32,
                 from ../osi/epicsAtomic.h:107,
                 from ../osi/os/RTEMS-score/osdThread.c:40:
../osi/os/RTEMS-score/epicsAtomicOSD.h: In function 'epicsAtomicReadMemoryBarrier':
../osi/os/RTEMS-score/epicsAtomicOSD.h:38: error: implicit declaration of function 'epicsAtomicMemoryBarrierFallback'
../../../../configure/RULES_BUILD:258: recipe for target 'osdThread.o' failed
make[4]: *** [osdThread.o] Error 1
```
Not sure why we didn't get an email about that last week, but I just triggered another build after fixing the MacOS agent so we _should_ get an email about that soon.

Anyway, that particular compiler (gcc 4.4.7) has no `__GCC_HAVE_SYNC_COMPARE_AND_SWAP_` macros defined.

anjohnson, 2022-03-23:
Check whether this has been fixed, I remember discussing it last year.

closed 2022-03-23 by anjohnson