epics-base/epics-base #183: DB link DBF_MENU to DBF_STRING conversions broken

anjohnson, 2021-06-30:
From a [bug report](https://epics.anl.gov/tech-talk/2021/msg01253.php) by Bayan Sobhani (Alex) I derived this test database:
```
record(ai, "ai") {}
record(stringin, "si") {
  field(INP, "ai.SEVR")
}
```
This shows the break on the 7.0 branch:
```
woz$ ./base-7.0/bin/darwin-x86/softIoc -d ~/db/test.db
Starting iocInit
############################################################################
## EPICS R7.0.5.1-DEV
## Rev. R7.0.5-264-g7ca25d3c888c56844877-dirty
############################################################################
iocRun: All initialization complete
epics> dbpf si.PROC 1
missing record support entry table Record Support Routine (get_enum_str) Record Type ai PV ai.SEVR  dbGetField

DBF_UCHAR:          1 = 0x1   
epics>
```
The database works properly with 3.15.9, 7.0.2 and 7.0.4.

`SEVR` is a `DBF_MENU` field, so the code shouldn't be trying to call the `RSET::get_enum_str()` routine for it anyway.

FreddieAkeroyd, 2021-07-01:
Ah, I should have checked here earlier, just been writing an email to report the same thing. In our case it was reading FTVL from a waveform record into a STRING aSub record field.

anjohnson, 2021-07-01:
@dirk-zimoch It seems likely to me that this bug was introduced by your dbChannelForDBLinks changes from last November, could you find time to take a look at it? We could probably use some additional test cases to match as well.

mdavidsaver, 2021-07-01:
> ... We could probably use some additional test cases to match as well.

fyi. [dbDbLinkTest.c](https://github.com/epics-base/epics-base/blob/7.0/modules/database/test/ioc/db/dbDbLinkTest.c), and the associated `dbDbLinkTest.db`, would be one obvious place to put additional tests like this.

dirk-zimoch, 2021-07-01:
I‘ll have a look…

dirk-zimoch, 2021-07-02:
Database links now use the same code path as CA links. So I tested what happens when the INP link is a CA link. In that case it fails in 3.14.12 as well with the error message  "dbpf: Value conversion from STRING to UCHAR failed". (Totally misleading of course. What happens is that processing the si record fails because reading INP fails and dbpf thinks that writing the string "1" to .PROC (a DBF_UCHAR) went wrong. ) In 7.0.5 this fails silently.

So the problem is the missing conversion from ENUM to STRING in dbChannel, which had always been missing in CA links. But `caget -d0 ai.SEVR` works as expected (ENUM is converted into string). Thus, dbChannel does something differently for (CA) links than for CA. I will dig into the code now...

dirk-zimoch, 2021-07-02:
'caget' calls `dbGet` which call `dbFastGetConvertRoutine[DBF_MENU][DBF_STRING]`. This works.
But the links calls `dbDbGetValue` which calls `dbFastGetConvertRoutine[DBF_ENUM][DBF_STRING]`. That s wrong. It calls `cvt_e_st_get` which works for .VAL enums but not for other fields as it tries to find `get_enum_str` of the record support.

dirk-zimoch, 2021-07-02:
The reason for this is that `dbChannelOpen()` initializes `chan->final_type` with `dbChannelExportType()` (=`chan->addr.dbr_field_type`) instead of `dbChannelFieldType()` (=`chan->addr.field_type`). In case of native DBF_MENU fields, these become `DBF_ENUM` which then calls the wrong `dbFastConvertRoutine`.

No idea what breaks when I change this. All tests still succeed and `caget ai.SEVR` still works.

The code duplication beween `dbGet()` and `dbDbGetValue()` bothers me.

dirk-zimoch, 2021-07-02:
```
record(ai, "ai") {}
record(stringin, "si") {
  field(INP, "ai.SEVR")
  field(PINI, "YES")
}
record(stringin, "si2") {
  field(INP, "ai.SEVR CP")
}
```
```
$ caget ai.SEVR si si2
ai.SEVR                        INVALID
si                             INVALID
si2                            INVALID
```

dirk-zimoch, 2021-07-02:
```
diff --git a/modules/database/src/ioc/db/dbChannel.c b/modules/database/src/ioc/db/dbChannel.c
index a806f3c..3c089b8 100644
--- a/modules/database/src/ioc/db/dbChannel.c
+++ b/modules/database/src/ioc/db/dbChannel.c
@@ -577,7 +577,7 @@ long dbChannelOpen(dbChannel *chan)
     /* Set up type probe */
     probe.type = dbfl_type_val;
     probe.ctx = dbfl_context_read;
-    probe.field_type  = dbChannelExportType(chan);
+    probe.field_type  = dbChannelFieldType(chan);
     probe.no_elements = dbChannelElements(chan);
     probe.field_size  = dbChannelFieldSize(chan);
     probe.sevr = NO_ALARM;
```

FreddieAkeroyd, 2021-07-07:
Hi @dirk-zimoch I tried this change on our system and noticed that `caget -n ai.SEVR` no longer prints the numerical value of the severity

dirk-zimoch, 2021-07-07:
Maybe one of the inventors of dbChannel needs to have a look?

FreddieAkeroyd, 2021-07-07:
For info I added
```
record(longin, "li") {
  field(INP, "ai.SEVR")
  field(PINI, "YES")
}
record(longin, "li2") {
  field(INP, "ai.SEVR CP")
}
```
and `li` and `li2` get the numerical value OK.

ralphlange, 2021-07-07:
You're saying the reluctant maintainer of `caget` should have a look?

mdavidsaver, 2021-08-12:
> Hi @dirk-zimoch I tried this change on our system and noticed that `caget -n ai.SEVR` no longer prints the numerical value of the severity

Specifically, the native type reported by `cainfo ai.SEVR` would change from `DBF_ENUM` to `DBF_STRING`.

mdavidsaver, 2021-08-12:
#196 adds a test to demonstrate this issue.  Maybe useful to others investigating.

anjohnson, 2021-08-12:
I'm wondering if [lp:1495417](https://bugs.launchpad.net/epics-base/+bug/1495417) is due to the same fundamental issue or needs a similar fix, although I haven't looked closely at it recently to confirm that.

mdavidsaver, 2021-08-13:
I've updated #196 with a candidate fix in dbDbLink.c.  Since this case is only reached if there are no channel filters, it seems reasonable to ignore the probed final type, and just use the original (`DBF_MENU`) field type.

```diff
@@ -188,8 +188,11 @@ static long dbDbGetValue(struct link *plink, short dbrType, void *pbuffer,
                 && dbChannelSpecial(chan) != SPC_ATTRIBUTE
                 && ellCount(&chan->filters) == 0)
     {
-        /* simple scalar: set up shortcut */
-        unsigned short dbfType = dbChannelFinalFieldType(chan);
+        /* Simple scalar w/o filters, so *Final* type has no additional information.
+         * Needed to correctly handle DBF_MENU fields, which become DBF_ENUM during
+         * probe of dbChannelOpen().
+         */
+        unsigned short dbfType = dbChannelFieldType(chan);
 
         if (dbrType < 0 || dbrType > DBR_ENUM || dbfType > DBF_DEVICE)
             return S_db_badDbrtype;
```

mdavidsaver, 2022-07-28:
Should be fixed by 7ce69155a9736ab02ea1265901c55d9044a19350

closed 2022-07-28 by mdavidsaver