epics-base/epics-base #184: osiSockTest udpSockFanoutTest() failure on macOS when VPN running

anjohnson, 2021-07-02:
This issue is minor and occurs on my laptop running macOS Catalina when I have the VPN connected:

```
woz$ cd modules/libcom/test/O.darwin-x86/
woz$ ./osiSockTest 
1..24
ok  1 - osiSockAttach
# udpSockTest()
ok  2 - epicsSocketCreate INET, DGRAM, 0
ok  3 - setsockopt BROADCAST := 1
ok  4 - getsockopt BROADCAST => 32
ok  5 - setsockopt BROADCAST := 0
ok  6 - getsockopt BROADCAST => 0
ok  7 - setsockopt MULTICAST_LOOP := 1
ok  8 - getsockopt MULTICAST_LOOP => 1
ok  9 - setsockopt MULTICAST_LOOP := 0
ok 10 - getsockopt MULTICAST_LOOP => 0
ok 11 - setsockopt IP_MULTICAST_TTL := 2
ok 12 - getsockopt IP_MULTICAST_TTL => 2
ok 13 - setsockopt IP_MULTICAST_TTL := 1
ok 14 - getsockopt IP_MULTICAST_TTL => 1
# udpSockFanoutBindTest()
# First test if epicsSocketEnableAddressUseForDatagramFanout() is necessary
ok 15 - bind() to port 49288
ok 16 - bind() to 49288 error -1, 48
# Now the real test
ok 17 - bind() to port 49288
ok 18 - bind() to port 49288
# udpSockFanoutTest()
# Interface 10.234.58.255:5064
# Not LO
# RX1 start
# RX2 start
# recvfrom error (35)
# RX1 end
# recvfrom error (35)
# RX2 end
# Result: RX1 0:0 RX2 0:0
# Interface 130.202.224.244:5064
# Not LO
# RX1 start
# RX1 success 2011178490
# RX2 start
# RX1 success 0
# RX1 success 1
# RX1 success 2
# RX1 success 3
# RX1 success 4
# RX1 end
# recvfrom error (35)
# RX2 end
# Result: RX1 1f:0 RX2 0:0
ok 19 - Found non-loopback interface
not ok 20 - Successes 0
# tcpSockReuseBindTest(0)
ok 21 - bind() to port 64874
ok 22 - bind() to 64874 error -1, 48
# tcpSockReuseBindTest(1)
# epicsSocketEnableAddressReuseDuringTimeWaitState
ok 23 - bind() to port 64875
ok 24 - bind() to 64875 error -1, 48

    Results
    =======
       Tests: 24 
      Passed:  23 = 95.83%
      Failed:   1 =  4.17%
```

The 10.234.58.* interface is my home WiFi, the second interface is my work VPN, which disappears when not connected.

The test passes if I disconnect from the VPN, or if I run it on my office workstation which doesn't have a VPN. Looking at older releases, it passed on 7.0.4 before `udpSockFanoutTest()` was added, failed on 7.0.4.1 which was the first release to add those checks and on 7.0.5.

mdavidsaver, 2021-07-02:
Do you know if beacons are being received by `caRepeater`?

Failure of this test may indicate that this isn't the case (oh no!).

```
# Interface 10.234.58.255:5064
```

I'm guessing that your wifi has firewall rules which are blocking CA traffic (quite reasonable).

```
# Interface 130.202.224.244:5064
```

`.244` (`0b11110100`) doesn't look like a valid broadcast address.  Is your VPN is setting up a point to point link?  It wouldn't surprise me at all to find a bug related to p2p links.

What does the IP configuration for this interface look like?

fyi. to avoid similar problems with my laptop (Linux) I configure a dummy software bridge interface with no firewall rules.

```
$ cat /etc/network/interfaces
...
auto br0
iface br0 inet static
    address 192.168.110.1
    netmask 255.255.255.0
    pre-up brctl addbr br0
    post-down brctl delbr br0
$ /sbin/ifconfig br0
br0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.110.1  netmask 255.255.255.0  broadcast 192.168.110.255
...
```

anjohnson, 2021-07-02:
> Do you know if beacons are being received by `caRepeater`?

`netstat` says there is one listening on `*.ca-2` (the process was started on June 1st), `casw` shows beacon anomalies when I start IOCs on it, and a disconnected `camonitor` does recognize and connect within a couple of seconds when I start an IOC with PVs that it's waiting for. CA clients connect fine to a single IOC when the VPN is up, but don't see PVs on a second IOC that's using the non-default TCP port. PVA can connect to multiple IOCs though.

> Is your VPN is setting up a point to point link?

Yes, but I suspect it also prevents packets from being routed through the WiFi interface at all, I've heard that's common for institutional VPNs. `ifconfig -a` shows a total of 14 interfaces including the loop-back; here are all the interfaces that have inet addresses:
```
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384
	options=1203<RXCSUM,TXCSUM,TXSTATUS,SW_TIMESTAMP>
	inet 127.0.0.1 netmask 0xff000000 
	inet6 ::1 prefixlen 128 
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1 
	nd6 options=201<PERFORMNUD,DAD>
en0: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=400<CHANNEL_IO>
	inet 10.234.58.108 netmask 0xffffff00 broadcast 10.234.58.255
	media: autoselect
	status: active
utun2: flags=80d1<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST> mtu 1300
	inet 130.202.224.241 --> 130.202.224.241 netmask 0xffff0000 
	inet6 fe80::5626:96ff:fece:bdfb%utun2 prefixlen 64 scopeid 0xf 
	inet6 fe80::53d0:980f:a847:74e1%utun2 prefixlen 128 scopeid 0xf 
	nd6 options=201<PERFORMNUD,DAD>
```

It would be nice if the test could detect this case and explain the limitations without actually failing, if only to avoid other people reporting the same kind of problem, but I'm fine with it failing if that's tricky.

anjohnson, 2022-09-28:
Core Group 9/28: Not worth working on.

closed 2022-09-28 by anjohnson