epics-base/epics-base #202: Posix implementation of epicsEvent is unsuitable for RTEMS-posix targets

anjohnson, 2021-09-22:
The code in libCom/osi/os/posix/osdEvent.c was written for workstation OSs that don't support true ISRs. The `struct epicsEventOSD` in that file which implements the epicsEvent object contains both a `pthread_mutex_t` and a `pthread_cond_t`. Unfortunately neither of these can be used in ISR context on an RTEMS-posix target. The `epicsEventTrigger()` routine must be callable from ISR context when the target supports them.

Discovered by @hjunkes and @mdavidsaver.

hjunkes, 2021-09-22:
For a first try I copied osdEvent.c from RTEMS-kernel to RTEMS-posix.
I commented out the show routine first. This uses old internal calls.
And replaced extern double rtemsTicksPerSecond_double by rtems_clock_get_ticks_per_second(). 
The test system has been running for 1.5 hours. Longer than before ;-) I have to observe this further and then adjust osdEvent.c properly.

anjohnson, 2021-09-22:
Can the `rtems_semaphore_*()` routines be used on SMP systems?

hjunkes, 2021-09-30:
My first suggestion for osdEvent.c for libcom/src/osi/os/RTEMS-posix
It uses self contained objects (rtems_binary_semaphore): 
https://docs.rtems.org/branches/master/c-user/self_contained_objects.html#binary-semaphore

[osdEvent.c.txt](https://github.com/epics-base/epics-base/files/7259201/osdEvent.c.txt)

anjohnson, 2021-09-30:
Seems like a reasonable start. Please change the Author at the top and remove the Saskatchewan copyright (you can add an FHI or one of your own if you need/wish to).

Note that `&(pSem->rbs)` is equivalent to `&pSem->rbs` since `->` [takes precedence](https://en.cppreference.com/w/c/language/operator_precedence) over `&` — I find the latter easier to read at a glance since I don't have to match up the parentheses.

Please look at the VxWorks implementation of `epicsEventWaitWithTimeout()` which shows the issues that can arise with the conversion of seconds into ticks. The `timeOut` parameter may be negative or zero (these both mean to poll the event and return immediately without waiting, so you could call `rtems_binary_semaphore_try_wait()` in that case), or can overflow when converted to ticks. An overflow is allowed to mean wait forever (as long as its's some reasonably large time period), and it looks like `rtems_binary_semaphore_wait_timed_ticks()` accepts 0 ticks to mean that or you could just call `rtems_binary_semaphore_wait()` in that case.

Thanks!

PR #206 (2022-01-24): Added Heinz's new osdEvent.c to RTEMS-posix
Fixes #202
closed 2022-01-24 by epics-jenkins