epics-base/epics-base #207: SingletonUntyped::pInstance warning

tboegi, 2021-10-13:
Folks,
compiling EPICS under Mingw gives a bunch of these warnings
(Any hints how to fix it ? I can prepare a PR)

g++             -D_MINGW -D__USE_MINGW_ANSI_STDIO   -DBUILDING_libCa_API   -O3   -Wall      -m64    -DEPICS_BUILD_DLL -DEPICS_CALL_DLL   -I. -I../O.Common -I. -I. -I.. -I../test -I../../../../../include/compiler/gcc -I../../../../../include/os/WIN32 -I../../../../../include         -o netSubscription.obj -c ../netSubscription.cpp
In file included from ../localHostName.h:27,
                 from ../cac.h:32,
                 from ../netSubscription.cpp:30:
../../../../../include/epicsSingleton.h:164:15: warning: 'void* SingletonUntyped::pInstance() const' redeclared without dllimport attribute after being referenced with dll linkage
  164 | inline void * SingletonUntyped :: pInstance () const
      |               ^~~~~~~~~~~~~~~~

FreddieAkeroyd, 2021-10-13:
Hi, I compiled with visual studio and `-Wall` in debug and non-debug mode and I don't get a similar looking warning, so it looks like a MinGW only issue.  If you remove the `inline` does that help get rid of the warning?

tboegi, 2021-10-19:
Interesting approach from the Core developpers.
A simple patch, which passes the appveyor tests:
https://ci.appveyor.com/project/epics-base-7/epics-base/builds/41143108
Is replaced by a commit directly to the 7.0 branch:
https://github.com/epics-base/epics-base/commit/2b3c6f2e262a7dfc506cce66547ad2dc74d8a80b
And failing the tests:
https://ci.appveyor.com/project/mdavidsaver/epics-base/builds/41187871

I think I am done here