epics-base/epics-base #211: Lack of `fork()` safety

mdavidsaver, 2021-11-03:
Documenting the current state of things.  As it stands now, there are a number places where the Base APIs may not safely be used by a child process after a `fork()`.  Adventurous users while investigate the safety of particular APIs are encouraged to note this, either in documentation on this issue.  Less adventurous users should assume that no API is safe from a child process.

The general [recommendation](https://www.google.com/search?q=mixing+fork+and+pthread) is to avoid usage of `fork()` when multiple threads exist as the new child process.  A newly `fork()`d child process consists of only a single thread and a copy of parent memory.  So any other threads in the parent process effectively "disappear", leaving memory in an indeterminate state (eg. a global mutex may be permanently locked).  Further, some (but perhaps not all) parent file descriptors are inherited by the child.  Unlike memory, these are not copies are changes by the child may effect the parent.

mdavidsaver, 2021-11-03:
References

- https://github.com/epics-base/epics-base/issues/83#issuecomment-667919934
- "Re: Pure Python IOC (CAProcess issue)" https://epics.anl.gov/tech-talk/2020/msg01358.php

anjohnson, 2022-01-24:
Can this issue be closed now that #214 has been merged?

mdavidsaver, 2022-09-28:
https://github.com/epics-base/epics-base/issues/286#issuecomment-1261203282 has a survey of global c++ constructors/destructors currently in libCom.

mdavidsaver, 2023-12-23:
Update circa 7.0.8.  Three remaining global constructors in libCom on linux-x86_64.

```
$ nm lib/linux-x86_64/libCom.a |grep GLOBAL__sub
0000000000000000 t _GLOBAL__sub_I_fdManager.cpp
0000000000000000 t _GLOBAL__sub_I_epicsThread.cpp
0000000000000000 t _GLOBAL__sub_I_osdTime.cpp
```

#451 would remove the ctor in epicsThread.cpp.

mdavidsaver, 2025-03-14:
Status update as of 7.0.9:  https://github.com/epics-base/epics-base/issues/286#issuecomment-2646472250