epics-base/epics-base #212: Error during build when using INSTALL_LOCATION

mfurseman, 2021-11-04:
I've got a docker based build system that defines `INSTALL_LOCATION` but _not_ `EPICS_BASE`. I assume https://github.com/epics-base/epics-base/commit/7ca25d3c888c56844877d194d91f1f47b69f323c was supposed to help prevent unhelpful errors further down in the build, but it has the side affect of making a previously working build now error for me.

anjohnson, 2021-11-04:
Hi Matthew, what software are you building when this fails, where are you defining `INSTALL_LOCATION`, and why can't you define `EPICS_BASE` in a RELEASE file as well?

The EPICS build system expects to have `EPICS_BASE` defined in a module's configure/RELEASE file (or some other RELEASE file that gets included by that) because it runs our convertRelease.pl script which also parses the RELEASE files to provide more information about the modules being linked with to various parts of the build. It very deliberately does _not_ look at environment variables to get the path to Base. Could your build system create a configure/RELEASE.local file that sets EPICS_BASE perhaps?

We do know of [one bug](https://bugs.launchpad.net/epics-base/+bug/1868769) related to modules setting INSTALL_LOCATION. It doesn't sound like you have the same problem as described there, but would you please check to confirm that.

mfurseman, 2021-11-04:
The full list of modules we're currently building is:
- base
- autosave
- ipac
- seq
- sscan
- calc
- asyn
- busy
- iocStats
- StreamDevice
- areaDetector

There's nothing to stop us adding in `EPICS_BASE` as well as `INSTALL_LOCATION`, however my objective was to provide the minimal set of configuration on top of the community repositories to do a build.

We have found that libtirpc.a, needed by Asyn, does not get included in the linker path for IOC's based on StreamDevice. Could this be because we have not been using `EPICS_BASE`?

We're setting the `INSTALL_LOCATION` variable the same everywhere, so I don't think we are hitting bug #1868769.

mfurseman, 2021-11-04:
Two answer your first question, we're building base, specifically pvData:
```
make[2]: Entering directory '/mnt/build/base/modules/pvData'
/mnt/build/prefix/configure/CONFIG:21: *** EPICS_BASE must be set in a configure/RELEASE file.  Stop.
make[2]: Leaving directory '/mnt/build/base/modules/pvData'
make[1]: *** [../configure/RULES_DIRS:85: pvData.linux-x86_64] Error 2
```

mdavidsaver, 2021-11-04:
How is `INSTALL_LOCATION` being set?  Are you running `make INSTALL_LOCATION=...`?  Are you making use of any `configure/RELEASE.local` files?

anjohnson, 2021-11-04:
Ahh, slightly different issue —  does your base/modules directory have a `RELEASE.*.local` file in it after that error? That file should get created with a definition of EPICS_BASE in it by the build. You aren't doing a `make distclean` or something similar are you?

BTW It seems very unlikely that you should need to be building ipac; it's actually optional for Asyn, and is really only needed for VxWorks IOCs (or Linux IOCs that have hardware connecting them to a VMEbus). It won't do you any harm though, so this is just an observation.

Yeah, linking to libtirpc is a bit of a pain. I have a solution but it will need some changes to Asyn, streamDevice and probably to the downstream IOCs as well, and I haven't talked to Mark Rivers or Dirk about it yet.

mfurseman, 2021-11-04:
> How is `INSTALL_LOCATION` being set? Are you running `make INSTALL_LOCATION=...`? Are you making use of any `configure/RELEASE.local` files?

`INSTALL_LOCATION` is set from our build script when invoking `make`.

Ideally I wanted a single top-level CONFIG_SITE and RELEASE file. In the end there are quite a few `*.local` files for AreaDetector. I've hard coded a definition of `EPICS_BASE`  in a `RELEASE.local` which sits above all the support modules.

mfurseman, 2021-11-04:
> Ahh, slightly different issue — does your base/modules directory have a `RELEASE.*.local` file in it after that error? That file should get created with a definition of EPICS_BASE in it by the build. You aren't doing a `make distclean` or something similar are you?

We are not doing any cleaning following the build. I'm not sure I understand what you're asking. I can run the build against a previous version of base and see if a `*.local` file is created? I'd like to share the build scripts we have for context, but I doubt we can get managerial agreement on licencing.

> BTW It seems very unlikely that you should need to be building ipac; it's actually optional for Asyn, and is really only needed for VxWorks IOCs (or Linux IOCs that have hardware connecting them to a VMEbus). It won't do you any harm though, so this is just an observation.

Thanks, I hadn't realised this! We do have some VME systems at UKAEA  (although no immediate plans to target them with EPICS) so I'll leave it in for now.

> Yeah, linking to libtirpc is a bit of a pain. I have a solution but it will need some changes to Asyn, streamDevice and probably to the downstream IOCs as well, and I haven't talked to Mark Rivers or Dirk about it yet.

This sounds good. We're doing our builds against CentOS 8 and this has cropped up a few times, I guess most sites are only beginning to look at moving beyond RHEL7. I brought this up in https://github.com/paulscherrerinstitute/StreamDevice/pull/82 where I've asked Dirk to unexpetadly include libtirpc.a for static builds.

anjohnson, 2021-11-04:
Early in the build of Base, a rule in modules/Makefile creates the file `modules/RELEASE.$(EPICS_HOST_ARCH).local` which is essential for building the "external submodules" (modules/pvData, modules/pvAccess etc.). These "external submodules" are all git submodules of the main git repository and have there own configure subdirectories, because it has to be possible to build them stand-alone against a separate Base-3.15 installation as well inside the Base-7.0.x tree where you see them. The libcom, ca and database sub-trees of base/modules are "internal submodules" and are part of the main git repository; they don't have their own configure subdirectories.

Back to that modules/RELEASE.*local file — if doesn't get created properly it will cause an error like the one you're reporting when the build descends into modules/pvData. It should set EPICS_BASE to your INSTALL_LOCATION path; it does that for me when I set INSTALL_LOCATION in a base/configure/CONFIG_SITE.local file, but it doesn't if I only set INSTALL_LOCATION on the make command-line. Setting INSTALL_LOCATION just as an environment variable has no effect at all, which doesn't surprise me.

I would guess that when you wrote
> `INSTALL_LOCATION` is set from our build script when invoking `make`.

you meant that you provide it on the command line. Please change that to putting it in a CONFIG_SITE file. Are you aware of the ability to override settings by creating a configure/CONFIG_SITE.local file instead of having to use `sed` to modify the configure/CONFIG_SITE file?

Don't worry about my cleaning question, I was trying to determine if something might have been deleting the RELEASE.*.local file — cleaning up after the build is also a tricky thing, which I don't recommend doing if you don't need to.

BTW I created a [PR for areaDetector](https://github.com/areaDetector/areaDetector/pull/72) which simplifies the configuration of that subsystem back to just standard configure/RELEASE and CONFIG_SITE files (you also need a configure/SITE_MODULES file which replaces what the configure/RELEASE file is currently doing). If you like this idea, please take a look and add comments about it to the PR, Mark Rivers hasn't sent me any indication about merging it, but someone else liking it might trigger him to look again.

mfurseman, 2021-11-05:
> Early in the build of Base, a rule in modules/Makefile creates the file modules/RELEASE.$(EPICS_HOST_ARCH).local which is essential for building the "external submodules".

I have a CONFIG_SITE.local in base/configure:
```
# Build all libraries as static
SHARED_LIBRARIES=NO
STATIC_BUILD=YES

# Don't install RELEASE files as we're using INSTALL_LOCATION
CONFIG_INSTALLS := $(filter-out ../RELEASE*, $(CONFIG_INSTALLS))
```
which is then installed to `INSTALL_LOCATION` and used for subsequent build steps. We're invoking make as
```
make linux-x86_64 INSTALL_LOCATION=$(pwd)/../prefix
```
which has been working for us up until https://github.com/epics-base/epics-base/commit/7ca25d3c888c56844877d194d91f1f47b69f323c. The modules/RELEASE.linux-x86_64.local _is_ created.

anjohnson, 2021-11-05:
So the only thing seems to be that you aren't setting INSTALL_LOCATION inside that CONFIG_SITE.local file; as I said above if do that it should work. Adding this to your CONFIG_SITE.local file should do the same as your command-line:
```makefile
INSTALL_LOCATION = $(TOP)/../prefix
```

Make sure that the modules/RELEASE.*.local file doesn't already exist if you're testing this by hand not from a clean `git clone` since it won't get updated if it already exists.

Are you trying to make your build of Base be position-independent, or will your final installation will be to a different path than you're building it for? We have other tricks to support those kinds of things if you need them.

mfurseman, 2021-11-05:
Yes, our aim is for a position independent build; that seems to be working now. In the IOC builds one just needs to write `EPICS_BASE=/install/location/path` and all the dependencies fall together without any extra configuration. Using the static build means that when deploying only the built IOC directory is needed. We do not have the support of sysadmins for an NFS or similar infrastructure at UKAEA; we also have MAST and JET with very different software systems. This Docker based build with a cross platform deployable is working well for us at the moment. 

I think if we use a relative path for the `INSTALL_LOCATION` we'll run into problems with AreaDetector modules where `TOP` is redefined at the module level? 

I'm still unsure about the need to write the `INSTALL_LOCATION` in the file. Out build has been working without this 'sanity check' until this change. Could this be because we specify the architecture target on the command line instead of leaving make to run the default `install` rule...?

mdavidsaver, 2021-11-05:
> ... Using the static build ...

While optional, if you did want to use shared libraries in some cases, you could look into `LINKER_USE_RPATH=ORIGIN`.

> ... AreaDetector modules where `TOP` is redefined at the module level?

Yup, I've run into this with all modules which have "recursive" builds (embedded TOP) in one way or another.

> ... I'm still unsure about the need to write the `INSTALL_LOCATION` in the file ...

The check is being overly conservative.  This used to work (I did this with Debian pacakges).  Maybe because `$(TOP)` is a special case in convertRelease.pl?  If so, this probably wasn't intentional.  Our (Base developers) position has been that all paths, and only paths, should go into a RELEASE* file where convertRelease.pl can parse them.

The RELEASE.local convention exists to help with this.  However, for builds intended to be reproducible (from a fresh VCS clone) I often just overwrite RELEASE and CONFIG_SITE, which also works for modules where `configure/` haven't been updated recently.  eg. [build-epics.sh](https://github.com/mdavidsaver/build-epics/blob/2b1851e045835d64297a3f55d868d9575d142703/build-epics.sh#L98)

anjohnson, 2021-11-05:
If you can put an absolute path in INSTALL_LOCATION then do so, from your make command-line I assumed that you might not want to burn one into your build scripts, but the path for EPICS_BASE that appears in the generated modules/RELEASE.*.local file will be an absolute version of the INSTALL_LOCATION path anyway. I had actually tested that and a relative path worked for building Base, but I agree that relative paths are more likely to cause problems in other modules. 

Like Michael I usually completely replace configure/RELEASE files in automated builds.

Note that this change derived from PR #163 although I think the particular commit was actually a later follow-up to it. Your approach might have worked once but it was never guaranteed to.

mfurseman, 2021-11-08:
Okay, thank you Andrew and Michael, I think I've got a much better idea how to configure our build process now. I hope you don't mind if I close this issue now while I continue tidying up our build.

Without being familiar with all the details, your https://github.com/areaDetector/areaDetector/pull/72 looks like a good idea. Personally I found the number of options for configuring areaDetector and the apparently differences in configuration from other EPICS modules quite difficult.

closed 2021-11-08 by mfurseman