epics-base/epics-base #227: /src/tools/makeRPath.py shebang does not specify python version

daykin, 2022-03-29:
The 'makeRPath' python script in /src/tools uses a `#!/usr/bin/env python` shebang. This results in using the default python2 on Debian systems, which is deprecated with the release of Debian 11 (bullseye). Does this need to be run on python 2 (in which case the shebang should be changed to `#!/usr/bin/env python2` or can python3 be specified?

mdavidsaver, 2022-03-29:
Also, I suggest pointing `/usr/bin/python` to `python3`.

```sh
apt-get install python-is-python3
```

daykin, 2022-03-29:
OK, thank you for the advice. I guess I should add a bit more context: I am not running into any issues with it directly, however, I am creating a debian package for Base 7.0.6 and Lintian throws an error about it. Would you rather I patch this only for our local version, or would you accept a PR to change it to a specific version?

mdavidsaver, 2022-03-30:
Well... personally I'm not enthused about replacing `/usr/bin/python`.  It seems like unnecessary churn to me.  Were I in your place, I'd just patch it out, or add a lintian override, and revisit the question after the next debian stable release.

daykin, 2022-03-30:
OK, will do. thanks.

closed 2022-03-30 by daykin