epics-base/epics-base #238: Recently-deprecated epicsThreadExitMain() is used by i-Tech Libera IOC code

anjohnson, 2022-04-27:
We just rebuilt the IOC for one or our Libera devices, updating from Base 7.0.4 to 7.0.6.1 which broke the IOC (running on Ubuntu 18.04, x86_64).

Their software calls `epicsThreadExitMain()` at the end of some code that I suspect is processing records in their own threads, the C++ symbol `istd::thread::thread_fnc(void*)` appears below but I believe there's tail-call optimization going on, I have the source for the call which appears at the end of a different more application-specific function.

We can change their code, but my guess is that they are calling this to free up thread-specific resources that the IOC is allocating. Given the frequency of the message repeats I suspect their threads are short-lived, so they need some call to release the stuff we allocate in `createImplicit()`, and I don't see any other API we provide which can do that.

Can we undeprecate this, or provide some other way of undoing `createImplicit()`? I do see that as a legitimate need.

```
iocInit
Starting iocInit
############################################################################
## EPICS R7.0.6.1-C2-1
## Rev. 7.0.6.1-C2-1
############################################################################
iocRun: All initialization complete
## Do iocStd post-initialization
< "/C2/ioc-modules/build/IOCSTD/4.1.0-C2-1+LIB-003/bin/linux-x86_64/postInit.iocsh"
# iocStdPostInit
#<aoi name="aoi_site_controls_iocstd_ioc-data-dump">
# Record our boot parameters
stdPrtBootParam "/C2/iocinfo/bootparams/sioc2k2llrf"
# Log the reboot time and reason
rebootLog "/C2/iocinfo/bootlog/sioc2k2llrf",1
epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.
Thread non-EPICS_140536976410368 (0x7fd0fc002020) can't proceed, suspending.
Dumping a stack trace of thread 'non-EPICS_140536976410368':
epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.Thread non-EPICS_140536974305024 (0x7fd1000026d0) can't proceed, suspending.
epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.Thread non-EPICS_140536970094336 (0x7fd0f0002640) can't proceed, suspending.
epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.Thread non-EPICS_140535820953344 (0x7fd0c0002a60) can't proceed, suspending.
epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.Thread non-EPICS_140536972199680 (0x7fd0f80018e0) can't proceed, suspending.
epicsThreadExitMain() has been deprecated for lack of usage.  Please report if you see this message.errlog: 77 messages were discarded
[    0x559e03ab28f3]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsStackTrace+0x73)
[    0x559e03aa54c5]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(cantProceed+0xc5)
[    0x559e03aafce6]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsThreadExitMain+0x56)
[    0x7fd15d189ebc]: /opt/libera/lib/libliberaistd.so.3.16(_ZN4istd6thread10thread_fncEPv+0x1c)
[    0x7fd15b8416db]: /lib/x86_64-linux-gnu/libpthread.so.0(start_thread+0xdb)
[    0x7fd15a5d6a3f]: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f)
Dumping a stack trace of thread 'non-EPICS_140536974305024':
[    0x559e03ab28f3]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsStackTrace+0x73)
[    0x559e03aa54c5]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(cantProceed+0xc5)
[    0x559e03aafce6]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsThreadExitMain+0x56)
[    0x7fd15d189ebc]: /opt/libera/lib/libliberaistd.so.3.16(_ZN4istd6thread10thread_fncEPv+0x1c)
[    0x7fd15b8416db]: /lib/x86_64-linux-gnu/libpthread.so.0(start_thread+0xdb)
[    0x7fd15a5d6a3f]: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f)
Dumping a stack trace of thread 'non-EPICS_140536970094336':
[    0x559e03ab28f3]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsStackTrace+0x73)
[    0x559e03aa54c5]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(cantProceed+0xc5)
[    0x559e03aafce6]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsThreadExitMain+0x56)
[    0x7fd15d189ebc]: /opt/libera/lib/libliberaistd.so.3.16(_ZN4istd6thread10thread_fncEPv+0x1c)
[    0x7fd15b8416db]: /lib/x86_64-linux-gnu/libpthread.so.0(start_thread+0xdb)
[    0x7fd15a5d6a3f]: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f)
Dumping a stack trace of thread 'non-EPICS_140535820953344':
[    0x559e03ab28f3]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsStackTrace+0x73)
[    0x559e03aa54c5]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(cantProceed+0xc5)
[... cut ...]
[    0x559e03ab28f3]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsStackTrace+0x73)
[    0x559e03aa54c5]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(cantProceed+0xc5)
[    0x559e03aafce6]: /C2/iocs/afp/liberaLLRF/master/bin/linux-x86_64/liberaIOC(epicsThreadExitMain+0x56)
[    0x7fd15d189ebc]: /opt/libera/lib/libliberaistd.so.3.16(_ZN4istd6thread10thread_fncEPv+0x1c)
[    0x7fd15b8416db]: /lib/x86_64-linux-gnu/libpthread.so.0(start_thread+0xdb)
[    0x7fd15a5d6a3f]: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f)
```

mdavidsaver, 2022-04-27:
imo. there is some "bad code smell" here (or maybe just stackoverflow).  I only see one frame in `libliberaistd.so.3.16`.  Is this a case of someone translating an example calling `pthread_exit(0)` instead of `return 0`?

anjohnson, 2022-04-27:
I assumed you would know what tail-call optimization is, their routine looks like this:
```cpp
void ...::ProcessThreadFnc()
{
    // omitted ...

    epicsThreadExitMain();
}
```
Thus the `istd::thread::thread_fnc()` calls the above method, but the compiler can clean up the local stack first and turn that call into a jump straight to our routine. Totally legit.

anjohnson, 2022-04-27:
We've just confirmed that removing the call to `cantProceed()` that was added to the Posix implementation of `epicsThreadExitMain()` allows the Libera IOC to work normally again.

mdavidsaver, 2022-05-04:
@anjohnson tells me (by private mail) that the underlying issue is that calls to some `epicsThread*()` by a non-EPICS thread leaks to  `epicsThreadOSD` allocated by `createImplicit()`.  So `epicsThreadExitMain()` was used (imo.) as a workaround since it was the only way for user code to trigger `free_threadInfo()`.

https://github.com/epics-base/epics-base/blob/cbae8d37b3da486ac8a68ad6ef9d2028cd98cca0/modules/libcom/src/osi/os/posix/osdThread.c#L632-L635

From some reading, I think that using the (second) `destructor` argument of [`pthread_key_create()`](https://linux.die.net/man/3/pthread_key_create) is the way to address this.  See #240

mdavidsaver, 2022-05-04:
> the underlying issue

cf #241

mdavidsaver, 2022-11-12:
I recently crossed paths with the 3.2 version of the libera IOC code.  It uses `std::thread` with the `ProcessThreadFnc()` method mentioned above as the entry point.  Now that a fix for #241 on Linux has been released with Base 7.0.7, the `epicsThreadExitMain()` call can simply be removed as the thread will end after `ProcessThreadFnc()` returns.

closed 2022-11-12 by mdavidsaver