epics-base/epics-base #241: Some `epicsThread*()` leak when called from a non-EPICS thread

mdavidsaver, 2022-05-04:
Several of the `epicsThread*()` functions will allocate an `epicsThreadOSD*` the first time one is called from a thread not created with `epicsThreadCreate*()`.  At present, this allocation is not `free()`'d automatically when the non-EPICS thread exits, causing a memory leak.

Status:

- [ ] posix/osdThread.c - Leak observed.  Candidate fix: #240
- [x] RTEMS-score/osdThread.c - Believed to be ok
- [ ] WIN32/osdThread.c - Believed to be an issue
- [x] vxWorks/osdThread.c - Believed to be ok

mdavidsaver, 2022-05-04:
wrt. Windows. [`TlsAlloc()`](https://docs.microsoft.com/en-us/windows/desktop/api/processthreadsapi/nf-processthreadsapi-tlsalloc) doesn't accept a destructor function like `pthread_key_create()`.  I found a [reference](https://stackoverflow.com/questions/18650956/how-to-release-the-object-in-a-tls-slot-at-thread-exit-on-windows) to using [`FlsAlloc()`](https://docs.microsoft.com/en-us/windows/win32/api/fibersapi/nf-fibersapi-flsalloc?redirectedfrom=MSDN) instead, which does.  This may have compatibility implications.  @FreddieAkeroyd fyi.

anjohnson, 2022-05-04:
The RTEMS-score implementation has not been coded to work properly from native threads anyway. It would fault or try to access data at or just above address 0 if some APIs ever get called from a non-EPICS thread. It can't be needed by any of our existing users though since that would have been triggered before now if it was, so we should wait to resolve that issue if it ever comes up (and ask the user if they can upgrade to RTEMS-posix instead).

anjohnson, 2022-05-09:
MS blog on Windows Fibers: https://devblogs.microsoft.com/oldnewthing/20191011-00/?p=102989

mdavidsaver, 2022-07-20:
I merged a fix for the POSIX targets as of 2f8272d3a17065693b2964e0975427c335b5a691.

mdavidsaver, 2024-06-19:
Win32 fix https://github.com/epics-base/epics-base/commit/f56412d6a56226524be57620f00b63519638b6aa in #348.  All fixed in R7.0.8

closed 2024-06-19 by mdavidsaver