epics-base/epics-base #256: dbLexRoutines should use synchronous logging

exzombie, 2022-05-12:
While some messages generated in `dbLexRoutines.c` are printed via `fprintf`, most are printed via various functions from `errlog.h`. So there is a mixture of synchronous and asynchronous logging, which results in messed-up output when parsing errors are encountered.

anjohnson, 2022-09-28:
Core Group 9/28: @mdavidsaver may have some fixes for this.

mdavidsaver, 2024-06-19:
I have been replacing many `errlogPrintf()` calls with `fprintf()` with the eventual goal to replace all of the calls to errlog prior to the completion of `iocInit()`.  This applies to code which is _only_ called prior to or during `iocInit()`.  I don't think I have found all call sites though.  So there is still work to be done.  I am going to rename this ticket for the larger task.

anjohnson, 2025-02-19:
Core Group: Needs a design discussion.