epics-base/epics-base #258: Undefined timestamp on first camonitor update of NORD field

bhill-slac, 2022-05-12:
SLAC is seeing a problem when client apps are monitoring the NORD field of image waveforms.
If the client app is running and has a CA monitor pending, when the IOC boots
the monitor first reports UDF as the waveform has not yet processed.
Once it does, the first update doesn't get a timestamp.
A fresh caget or camonitor does see the timestamp.
% camonitor ioc:fillingwaveform.NORD
ioc:fillingwaveform.NORD       <undefined> 0 UDF INVALID
ioc:fillingwaveform.NORD       <undefined> 3 UDF INVALID
ioc:fillingwaveform.NORD       <undefined> 3 UDF INVALID
ioc:fillingwaveform.NORD       2022-05-12 14:18:00.721198 5  
ioc:fillingwaveform.NORD       2022-05-12 14:18:00.721198 5

bhill-slac, 2022-05-12:
[test.db.txt](https://github.com/epics-base/epics-base/files/8682707/test.db.txt)

bhill-slac, 2022-05-12:
To trigger the bug, I start the camonitor then run:
$EPICS_BASE/bin/$EPICS_HOST_ARCH/softIoc -d test.db

Then
dbpf ioc:fillingwaveform.PROC 1

anjohnson, 2022-05-12:
Investigation: The waveformRecord's `readValue()` calls
```
  db_post_events(prec, &prec->nord, DBE_VALUE | DBE_LOG);
```
after NORD gets updated, that needs to be moved into process() after the timestamp is set.

bhill-slac, 2022-05-12:
Thanks, I'll give it a try!

bhill-slac, 2022-05-12:
Looks like that works for a fix.   I'll work on adding a test case and push that first.

anjohnson, 2022-09-28:
Core Group 9/29: Fixed with #263

closed 2022-09-28 by anjohnson