epics-base/epics-base #261: Improvements to EPICS_DEPRECATED?

anjohnson, 2022-05-13:
It would be nice to have a standard way to suppress `EPICS_DEPRECATED` warnings in places that we can't remove — IIRC test code and implementations of member functions in a derived class are places that are unavoidable for example.

mdavidsaver, 2022-07-28:
The way I've found to to do this involves defining more macros.  eg.

```c++
#if defined(PVD_INTERNAL)
#  define PVD_DEPRECATED(msg)
#elif __GNUC__ > 4 || __GNUC__ == 4 && __GNUC_MINOR__ >= 5
#  define PVD_DEPRECATED(msg) __attribute__((deprecated(msg)))
#else
#  define PVD_DEPRECATED(msg) EPICS_DEPRECATED
#endif
```

https://github.com/epics-base/pvDataCPP/blob/f3911d583139ac5969df6a992c424788e41d9634/src/misc/pv/serialize.h#L19-L25

Maybe this could be rolled into `makeAPIheader.pl` to take advantage of the `BUILDING_${stem}_API` flag?

RAYemelyanova, 2023-03-08:
This seems interesting, Which functions in particular would you like to see deprecation warnings be silenced on?

anjohnson, 2025-02-19:
Core Group: Most of the annoying deprecation warnings are when building pvData and pvAccess submodules.