epics-base/epics-base #280: Timestamp when monitoring non-VAL

mdavidsaver, 2022-07-28:
#263 Fixes an issue where `process()` for waveformRecord called `db_post_events()` before `TIME` was updated by `recGblGetTimeStampSimm()`.  So the posted update had the wrong timestamp.

At least `aai` and `aao` have the same code.

anjohnson, 2022-09-28:
Core Group 9/28: @bhill-slac 's fix #263 only fixed the waveform record, we need to apply the same change to aai and aao, maybe others too.

henrique-silva, 2023-03-08:
I'll work on applying this fix to the other records as well

henrique-silva, 2023-03-08:
So, while investigating this issue I noticed that currently the `NORD` field is updated twice every time the record processes. It keeps the same value, but gets a new timestamp.
This is due to  `db_post_events` being called both on the record support and device support layers.

As agreed with @mdavidsaver and @anjohnson , I'll work on removing the duplicate call on the device support layer, so we keep it only on the record support

anjohnson, 2025-02-19:
Core Group: PR #359 needs final cleanup and merging.