epics-base/epics-base #284: Constant `INP*` to aSub

mdavidsaver, 2022-08-05:
cf. https://epics.anl.gov/tech-talk/2022/msg01154.php

```
record(aSub,"$(user):aSubExample")
{
   field(INAM,"myAsubInit")
   field(SNAM,"myAsubProcess")
   field(FTA,"DOUBLE")
   field(NOA,"10")
   field(INPA,"$(user):compressExample CPP")
   field(FTB,"DOUBLE")
   field(NOB,"1")
   field(INPB,"123")
}
```

Stepping through `aSubRecord.c`, it looks like `init_record()` does the right thing, and leaves `B=123` and `NEB=1`.

https://github.com/epics-base/epics-base/blob/424d678abff19887cd2fa9ff5af82e33806a3712/modules/database/src/std/rec/aSubRecord.c#L133-L135

However, later in `fetch_values()`, there is no `dbLinkIsConstant()` test, so the call to `dbGetLink()` goes to `dbConstGetValue()` where `NEB` is set to zero.

https://github.com/epics-base/epics-base/blob/424d678abff19887cd2fa9ff5af82e33806a3712/modules/database/src/std/rec/aSubRecord.c#L278-L283

https://github.com/epics-base/epics-base/blob/424d678abff19887cd2fa9ff5af82e33806a3712/modules/database/src/ioc/db/dbConstLink.c#L212-L213

mdavidsaver, 2022-08-05:
@anjohnson Should the call to `dbGetLink()` be conditional on `!dbLinkIsConstant()` as in eg. `devAiSoft.c`?

anjohnson, 2022-08-06:
That sounds right. I wonder if this code got changed at all when I introduced the link support routines?

anjohnson, 2022-08-08:
Prior to the zero-size-arrays changes (merged into Base-7.0.5) setting aSub.NEB to zero didn't prevent a subsequent scalar get of aSub.B from returning the first array element, so the example record above actually works against Base-7.0.4.

This bug _should_ have been detected in Base-7.0.5 by `testHexConstantLinks()` in regressTest.c because that uses an aSub with several constant scalar INP links. Apparently `testdbGetFieldEqual()` isn't checking that the field actually holds exactly one element, which it probably needs to do.

mdavidsaver, 2022-08-08:
> This bug _should_ have been detected in Base-7.0.5 by `testHexConstantLinks()` in regressTest.c

Almost.  The `as1` record is never processed, so `dbConstGetValue()` is never called.

If I change `testHexConstantLinks()` to add:

```c

    testdbPutFieldOk("as1.PROC", DBR_LONG, 1);
    testdbGetFieldEqual("as1.A", DBR_LONG, 0x10);
```

then this new `testdbGetFieldEqual()` errors with

```
not ok 23 - dbGetField("as1.A", 5, ...) -> zero length
```

mdavidsaver, 2022-08-08:
Maybe a basic question, but why does `dbConstGetValue()` set `nReq=0` and return success?

Should it return an error?

mdavidsaver, 2022-08-08:
I should mention that I've looked at eg. calcRecord, where `dbGetLink()` is called without checking for a constant link.  But always without `nReq`.  So it is a no-op.

https://github.com/epics-base/epics-base/blob/424d678abff19887cd2fa9ff5af82e33806a3712/modules/database/src/std/rec/calcRecord.c#L436-L441

Is the present behavior of `dbConstGetValue()` about compatibility with external record support which might not be checking for constant links?

anjohnson, 2022-08-08:
> The as1 record is never processed

Thanks, that makes sense; in the original example the CPP link in INPA processes the record when it connects, calling the `dbGetLink()` routine at that point.

> Is the present behavior of `dbConstGetValue()` about compatibility with external record support which might not be checking for constant links?

It is about compatibility, but I didn't introduce it myself. The 3.14 version of dbAccess.c::dbGetLinkValue() does this for a CONSTANT link:
```C
    if (plink->type == CONSTANT) {
        if (poptions) *poptions = 0;
        if (pnRequest) *pnRequest = 0;
    }
```

I think it's reasonable to set `*pnRequest = 0`, it tells the caller their `dbGetLink()` call hasn't modified the buffer at all. It can't return an error for compatibility with record types that don't check for constant links at all, or with those that only do `if (plink->type == CONSTANT)` since the `{const:...}` link type also does the same thing.

anjohnson, 2022-09-28:
Core Group 9/28: @mdavidsaver wants to look at adding a warning for code that does the wrong thing.

closed 2025-02-19 by anjohnson