epics-base/epics-base #285: if[n]eq RTEMS_VERSION in configure/os/CONFIG.Common.RTEMS-target files

hjunkes, 2022-08-07:
E.g. in the file configure/os/CONFIG.Common.RTEMS-pc386 you can find the following:

```
ifneq ($(RTEMS_VERSION),5)
  $(info *** This target is not compatible with the configured RTEMS version.)
  $(info *** Build the RTEMS-pc386 (-qemu) target for RTEMS 4.x)
  $(error Can't continue)
endif
```
But when the file gets executed $RTEMS_VERSION is not defined. 
How must/can this be done correctly? Of course I just copied this construct and now I had to realize that it does not work :-(
Danke Heinz

mdavidsaver, 2022-08-07:
The issue is with `RTEMS_VERSION = 6`?

A simple change would be to reverse the condition and test for RTEMS 4.x.  (Did Base ever supported RTEMS 3.x?)

https://github.com/epics-base/epics-base/blob/424d678abff19887cd2fa9ff5af82e33806a3712/configure/os/CONFIG.Common.RTEMS-pc386#L28-L32
```
ifneq ($(RTEMS_VERSION),4)
```
https://github.com/epics-base/epics-base/blob/424d678abff19887cd2fa9ff5af82e33806a3712/configure/os/CONFIG.Common.RTEMS-pc686#L35-L39
```
ifeq ($(RTEMS_VERSION),4)
```

hjunkes, 2022-08-07:
Hmm, I had understood that the queries in these files do not make sense because RTEMS_VERSION are NULL/not defined/not set at the time of interpretation?

hjunkes, 2022-08-08:
Ok, as always. If you sleep on it for a night and look at the whole thing calmly.
The solution looks like this (the definition after the include of $(CONFIG)/os/CONFIG.Common.RTEMS )

```
OP_SYS_LDLIBS += -qrtems -Wl,--gc-sections
ARCH_DEP_LDFLAGS += -L$(RTEMS_BASE)/$(GNU_TARGET)$(RTEMS_VERSION)/beagleboneblack/lib/

include $(CONFIG)/os/CONFIG.Common.RTEMS

ifeq ($(RTEMS_VERSION),6)
# Which nfs version to use
ARCH_DEP_CFLAGS += -DNET_CFG_NFS_MOUNT_OPTIONS=\"nfsv4,minorversion=1\"
endif
```
and if I had read correctly :-(
in CONFIG.Common.RTEMS-pc386
```
# This check must appear after the above include
ifeq ($(RTEMS_VERSION),5)
  $(info *** This target is not compatible with the configured RTEMS version.)
  $(info *** Build the RTEMS-pc686 (-qemu) target for RTEMS 5.x)
  $(error Can't continue)
endif
```
Sorry for the stress.

closed 2022-08-08 by hjunkes

mdavidsaver, 2022-08-08:
> ARCH_DEP_CFLAGS += -DNET_CFG_NFS_MOUNT_OPTIONS=\"nfsv4,minorversion=1\"

I'm confused.  Where/how is this `NET_CFG_NFS_MOUNT_OPTIONS` macro used?

hjunkes, 2022-08-08:
Please excuse me for adding to the confusion here again. 
These are my current extensions for RTEMS6, libbsd, and NFSv4. They are not yet in the epics-base-repo but are currently only on a private one: https://github.com/hjunkes/epics7Rtems6Beaglebonebalck
And I believed that the query for the RTEMS version in the config-files for the individual target architectures generally does not work. It was just that in the CONFIG.Common.RTEMS-beaglebone I had not taken care to insert the query after the inclusion of CONFIG.Common.RTEMS.  Unfortunately, it was my carelessness and I just haven't seen/found it.