epics-base/epics-base #286: UB fork() warnings from legal code

anjohnson, 2022-08-08:
```C
#include <unistd.h>

int main()
{
    return fork();
}
```

Compiling and linking the program above against libCom causes it to trigger the warning added in PR #214 on MacOS. I suspect Linux will behave the same but I haven't actually tried it. Adding calls to `epicsThreadShowAll()` demonstrate that no EPICS threads are running immediately before or after the fork, so I think the current check is too simplistic. I had assumed it would only warn if a thread is actually started before `fork()` is called.

Passing `-s` to the ca-nameserver or ca-gateway (which tells them to detach from their parent session) causes them to generate this warning incorrectly. IMHO a server process which double-forks to disconnect and become a session leader before it makes _any_ EPICS calls should not be triggering the warning.

mdavidsaver, 2022-09-28:
My concern with relaxing the warning stems from concern about what could happen in global constructors running before `main()` (or on `dlopen()`).

If we care to assert that any such constructors are not, and will not, do anything `fork()` unsafe then I'm ok with relaxing the warning.  eg. not starting worker threads, or maybe even not locking a mutex.

With pvxs I make a stronger statement that no global constructors will appear in libraries at all.  In large part, this is because I know how to verify this [automatically](https://github.com/mdavidsaver/pvxs/blob/master/.ci-local/cdt-check.sh).  For GCC at least.  I don't know about clang.  (windows has no `fork()`, so I think isn't directly relevant)

From a quick check, it doesn't seem like the 4 Base libraries have any global constructors at present.  At least for my Linux build.

```
nm lib/linux-x86_64/*.a|grep __static_initialization_and_destruction
```

For comparison:

```c++
void bar();
struct Foo {
   Foo() {bar();}
} foo;
```

```
$ g++ -o ctor.o -c ctor.cpp && nm ctor.o | grep __static_initialization_and_destruction
0000000000000000 t _Z41__static_initialization_and_destruction_0ii
```

mdavidsaver, 2022-09-28:
> ... no global constructors will appear in libraries at all ...

If we don't have any legacy to contend with, I think this would be a good assurance to give for the Base libraries.

mdavidsaver, 2022-09-28:
> ... the code I posted at the top of this issue has no global constructors ...

Not directly, no.  My concern is about any global constructors in `libCom`, which would be run before `main()`.

mdavidsaver, 2022-09-28:
> ... good assurance to give for the Base libraries.

This assurance would have to be restricted to the 4 libraries built from this repository as some of the PVA libraries do have global constructors.

```
$ nm pvAccess/lib/linux-x86_64/lib*.{a,so}|grep __static_initialization_and_destruction
0000000000000000 t _Z41__static_initialization_and_destruction_0ii.constprop.0
00000000000001c4 t _Z41__static_initialization_and_destruction_0ii.constprop.0.cold
0000000000000000 t _Z41__static_initialization_and_destruction_0ii.constprop.0
0000000000000756 t _Z41__static_initialization_and_destruction_0ii.constprop.0.cold
0000000000054000 t _Z41__static_initialization_and_destruction_0ii.constprop.0
0000000000054460 t _Z41__static_initialization_and_destruction_0ii.constprop.0
0000000000044998 t _Z41__static_initialization_and_destruction_0ii.constprop.0.cold
000000000004a6be t _Z41__static_initialization_and_destruction_0ii.constprop.0.cold
```

mdavidsaver, 2022-09-28:
wrt. #214.  Seeing the warning implies that something called `epicsThreadInit()`.

In the case of your example, it looks my quick check was wrong.  There is at least one global ctor in libCom, which does trigger epicsThread* stuff.

```
gcc -o forkme forkme.c -Wl,--no-as-needed \
 -L$HOME/work/epics/base-git/lib/linux-x86_64 \
 -Wl,-rpath,$HOME/work/epics/base-git/lib/linux-x86_64 \
 -lCom
```

```
Breakpoint 1, 0x00007ffff7fa5480 in pthread_atfork () from /home/mdavidsaver/work/epics/base-git/lib/linux-x86_64/libCom.so.3.21.1
(gdb) bt
#0  0x00007ffff7fa5480 in pthread_atfork () from /home/mdavidsaver/work/epics/base-git/lib/linux-x86_64/libCom.so.3.21.1
#1  0x00007ffff7f91b08 in once () at ../osi/os/posix/osdThread.c:354
#2  0x00007ffff7d2034f in __pthread_once_slow (once_control=0x7ffff7fc8f80 <once_control>, init_routine=0x7ffff7f91ae0 <once>)
    at pthread_once.c:116
#3  0x00007ffff7d203f5 in __GI___pthread_once (once_control=once_control@entry=0x7ffff7fc8f80 <once_control>, 
    init_routine=init_routine@entry=0x7ffff7f91ae0 <once>) at pthread_once.c:143
#4  0x00007ffff7f913f7 in epicsThreadInit () at ../osi/os/posix/osdThread.c:450
#5  0x00007ffff7f934dd in epicsThreadOnce (id=id@entry=0x7ffff7fc8408 <iocshOnceId>, func=func@entry=0x7ffff7f7e110 <iocshOnce(void*)>, 
    arg=arg@entry=0x0) at ../osi/os/posix/osdThread.c:530
#6  0x00007ffff7f7f2e3 in iocshInit () at ../iocsh/iocsh.cpp:135
#7  iocshTableLock () at ../iocsh/iocsh.cpp:144
#8  iocshRegister (piocshFuncDef=piocshFuncDef@entry=0x7ffff7fc35e0 <commentFuncDef>, 
    func=func@entry=0x7ffff7f7e100 <commentCallFunc(iocshArgBuf const*)>) at ../iocsh/iocsh.cpp:166
#9  0x00007ffff7f6e4d7 in localRegister () at ../iocsh/iocsh.cpp:1492
#10 IocshRegister::IocshRegister (this=<optimized out>) at ../iocsh/iocsh.cpp:1508
#11 __static_initialization_and_destruction_0 (__priority=65535, __initialize_p=1) at ../iocsh/iocsh.cpp:1510
#12 _GLOBAL__sub_I_iocsh.cpp(void) () at ../iocsh/iocsh.cpp:1510
#13 0x00007ffff7fe1fe2 in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffe0a8, env=env@entry=0x7fffffffe0b8)
    at dl-init.c:72
#14 0x00007ffff7fe20e9 in call_init (env=0x7fffffffe0b8, argv=0x7fffffffe0a8, argc=1, l=<optimized out>) at dl-init.c:30
#15 _dl_init (main_map=0x7ffff7ffe180, argc=1, argv=0x7fffffffe0a8, env=0x7fffffffe0b8) at dl-init.c:119
#16 0x00007ffff7fd30ca in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#17 0x0000000000000001 in ?? ()
#18 0x00007fffffffe3d5 in ?? ()
#19 0x0000000000000000 in ?? ()
```

mdavidsaver, 2022-09-28:
> There is at least one global ctor in libCom,

There are five.

```
000000000001b550 t _GLOBAL__sub_I_epicsThread.cpp
000000000001b490 t _GLOBAL__sub_I_fdManager.cpp
000000000001b4c0 t _GLOBAL__sub_I_iocsh.cpp
000000000001b570 t _GLOBAL__sub_I_osdTime.cpp
000000000001b590 t _GLOBAL__sub_I_timerQueueActive.cpp
```

mdavidsaver, 2022-09-28:
https://github.com/epics-base/epics-base/blob/0f8ea3aa3671cb0433b4350020430a7740d6fdc6/modules/libcom/src/osi/epicsThread.cpp#L386-L393

I think this could simply be removed as it is redundant to work which already happens in `epicsThreadInit()`.

https://github.com/epics-base/epics-base/blob/0f8ea3aa3671cb0433b4350020430a7740d6fdc6/modules/libcom/src/fdmgr/fdManager.h#L116

This global object is more problematic in terms of API, and in terms of side-effects (its constructor calls `osiSockAttach()`).

https://github.com/epics-base/epics-base/blob/0f8ea3aa3671cb0433b4350020430a7740d6fdc6/modules/libcom/src/iocsh/iocsh.cpp#L1291-L1295

Could easily be switched to an `epicsThreadOnce()`.

https://github.com/epics-base/epics-base/blob/0f8ea3aa3671cb0433b4350020430a7740d6fdc6/modules/libcom/src/osi/os/posix/osdTime.cpp#L57-L64

Could also be `epicsThreadOnce()`, though maybe more expensive.

https://github.com/epics-base/epics-base/blob/0f8ea3aa3671cb0433b4350020430a7740d6fdc6/modules/libcom/src/timer/timerQueueActive.cpp#L33

I think this is the source of the global ctor/dtor for `timerQueueActive.cpp`.  I'm not confident that I understand how `epicsSingleton` is supposed to work.  At least this global object is not public API, and so could be made lazy with `epicsThreadOnce()`.

mdavidsaver, 2022-09-28:
Where does this leave us?

Simply loading `libCom.so` causes quite a bit to happen.  None of which is obviously unsafe wrt. `fork()`.  Although I would not be surprised to find some errlog* calls on failure paths, which of course implicitly `epicsThreadCreate()`.

So I see this warning from "legal code" as an indication that libCom does some questionable things wrt. `fork()` safety.

anjohnson, 2022-09-28:
Thanks for that analysis, I think it was a useful exercise even if we do end up dropping this particular issue. It would be good to try to make libCom `fork()`-safe, but evidently it isn't going to be a trivial thing to do. Probably a good topic for the meeting at BNL.

mdavidsaver, 2025-02-09:
Status update as of 1d19ba4cc2e8c9ab17288b684022a0682e398bde :

- [X] `epicsSingleton` constructors
  -  Fixed by 8f1243da406aebf5a8cebda65ea84e5dbeeff286 when building with `-std=c++11` or later.
- [ ] `fileDescriptorManager` singleton
  - https://github.com/epics-base/epics-base/pull/298 has an example of making the singleton lazy, now out of date.
- [X] `localRegister()` in iocsh.cpp
  - Fixed by 7830345e5927ca1f262dda17a1f819286c5b5821
- [ ] `initMainThread()`
  - Removed by #451
- [ ] `timeRegister()` really two separate instances
  - [ ] in `posix/osdTime.cpp`
  - [ ] in `Darwin/osdTime.cpp`