epics-base/epics-base #288: RTEMS and -ffunction-sections and -fdata-sections options

kiwichris, 2022-08-10:
Code and Data Sections
==================

RTEMS has supported options on most BSPs since RTEMS 5 and now 6:

- `-ffunction-sections` 
- `-fdata-sections`

These options place each function and object file scoped variable in a separate `.text` and `.data` section. The linker only includes the referenced functions and variables from an object file in the executable due to the separate sections. The unreferenced functions and variables are garbage collected. This means a static global variable in a module that is not referenced will not be included.

EPICS uses the following construct in a few places:
```
/*
 * Register commands on application startup
 */
static int Registration() {
    iocshRegisterCommon();
    iocshRegister(&rrddFuncDef, rrddCallFunc);
    return 0;
}
static int done EPICS_UNUSED = Registration();
```
The variable `EPICS_UNUSED` is not referenced so it will not be linked into the final application. As a result the automatic registration will not happen.

----------------------------------

RTEMS Versions
============

- RTEMS 5 exports these options in the BSP `Makefile.inc` compile flags. EPICS does not filter these flags so it is built with them.
-  RTEMS 6 only exports BSP machine ABI flags so these flags are not exported.
- LibBSD requires these flags when building. Applications using LibBSD are also required to provide these flags.

This means:

- Building EPICS 7 with RTEMS 5 and the legacy networking stack for the `mvme2700` BSP results in EPICS being built with these flags.
- Building EPICS 7 with RTEMS 6 and LibBSD requires EPICS have these options in its compile flags.

_Note_: the register call happens early in the BSP start up and may work now but may fail depending on the calls to RTEMS or supporting libraries made. RTEMS 6 supports a SYSINIT approach for a structured and sequenced initialization.

mdavidsaver, 2022-08-10:
> This means a static global variable in a module that is not referenced will not be included.

Are you seeing some specific breakage?

What we are doing has worked thus far.  All of the GCC+Binutils versions I've encountered did the Right Thing, or at least the same thing, wrt. `-f*-sections` and `--gc-sections`.  Including c++ global constructors in an object file as long as some symbol in that file is needed (survives `--gc-sections`), or the object file is mentioned explicitly when linking the executable.

I would expect the same logic to apply to the `__attribute__(section(...))` behind the RTEMS SYSINIT mechanism.

Has this behavior changed in some newer GCC version?

Are you thinking ahead to `-flto`?

(imo. LTO seems like a better approach to reduce code size for embedded targets, though I've tried and failed to use it with RTEMS5/arm)

kiwichris, 2022-08-10:
> > This means a static global variable in a module that is not referenced will not be included.
> 
> Are you seeing some specific breakage?

For my testing of something else in EPICS 7 I disabled the options so no I am not. I am planing to specifically test the effect and characterize.

> What we are doing has worked thus far. All of the GCC+Binutils versions I've encountered did the Right Thing, or at least the same thing, wrt. `-f*-sections` and `--gc-sections`. 

This is good to hear and welcoming. Have you tried with the latest tools for RTEMS 6?

> Including c++ global constructors in an object file as long as some symbol in that file is needed (survives `--gc-sections`), or the object file is mentioned explicitly when linking the executable.

OK. If these functions are called when the global constructors are that is fine. If it was earlier RTEMS may not be initialized.

> I would expect the same logic to apply to the `__attribute__(section(...))` behind the RTEMS SYSINIT mechanism.

I thought the RTEMS symbols are global and not private. I know global is needed for similar libbsd (ie FreeBSD) symbols. You could be right. I will check.

> Has this behavior changed in some newer GCC version?

GCC's dead code elimination is always improving and why I am interested in how this works. I recently tripped over a bug in libstdc++ related to exceptions in the exit path that turned out to be a neat optimization of gcc.

> Are you thinking ahead to `-flto`?

No.

> (imo. LTO seems like a better approach to reduce code size for embedded targets, though I've tried and failed to use it with RTEMS5/arm)

I have recently enable LTO functionality in the RSB built tools. I had not understood the enable option and corrected this. I have built and run LTO linked RTEMS executables and like you believe this will become standard. I liked what I saw especially with C++ applications.

mdavidsaver, 2022-08-11:
> > wrt. `-f*-sections` and `--gc-sections`.
> 
> This is good to hear and welcoming. Have you tried with the latest tools for RTEMS 6?

I have not yet done anything with RTEMS > 5.

Also worth noting that the Clang and MSVC linkers have (apparently) this same behavior.  So while not standardized, this handling of global constructors is actually widely portable.  I would be surprised if it were intentionally changed in GCC.

> I thought the RTEMS symbols are global and not private.  ...

I hadn't noticed.  Maybe makes a difference for `__attribute__(section`, but not for `__attribute__(constructor`?

This also reminds me of the LD option [`--gc-keep-exported`](https://sourceware.org/binutils/docs/ld/Options.html) which I first noticed earlier today.  (I also noticed `--print-memory-usage`)

kiwichris, 2022-08-11:
> > > wrt. `-f*-sections` and `--gc-sections`.
> > 
> > 
> > This is good to hear and welcoming. Have you tried with the latest tools for RTEMS 6?
> 
> I have not yet done anything with RTEMS > 5.

I will try RTEMS 6 and also libbsd. I am not sure how far I will get before I see some breakage. I know Heinz has some patches for libbsd but I am not sure what the status of them is.

> Also worth noting that the Clang and MSVC linkers have (apparently) this same behavior. So while not standardized, this handling of global constructors is actually widely portable. I would be surprised if it were intentionally changed in GCC.

Thanks. It would be nice if it is standardized and supported.

The issue then moves to LibBSD needing the options and RTEMS 6 not exporting them. EPICS would need to add the garbage collection option to `LDFLAGS` for RTEMS 6 and not for 5.

> > I thought the RTEMS symbols are global and not private.  ...
> 
> I hadn't noticed. Maybe makes a difference for `__attribute__(section`, but not for `__attribute__(constructor`?

Let me check and report back.

> This also reminds me of the LD option [`--gc-keep-exported`](https://sourceware.org/binutils/docs/ld/Options.html) which I first noticed earlier today. (I also noticed `--print-memory-usage`)

I also noticed that option and my understanding is for use with shared libraries so the garbage collector does not remove global variables. RTEMS does not support the ELF SO model so I am not sure what effect it has. We have not needed that option.