epics-base/epics-base #291: require operator delete if placement operator delete

mdavidsaver, 2022-08-22:
https://github.com/epics-base/epics-base/blob/59ee7b32bc532797fffd1fa3fa7f2283a394df6c/modules/libcom/src/timer/timer.cpp#L221-L228

This comment block is repeated a number of times in the C++ code which utilizes [class specific](https://en.cppreference.com/w/cpp/memory/new/operator_new#Class-specific_overloads) [placement new](https://en.cppreference.com/w/cpp/language/new#Placement_new), and catches my eye everything time I see it.

From what I can tell, these unused (and unusable) `T::operator delete(void*)` are required, although not MS specific.  So it is only that the comment is inaccurate.

tldr...

From what I can tell this is not an MS specific behavior.  The same holds true for GCC, at least with its current [c++ ABI](https://itanium-cxx-abi.github.io/cxx-abi/abi.html).  However, the only detailed discussions I've found are in terms of a specific OS/compiler ABI(s).  So it is not clear to me how much of this is specific by the c++ standard.

With respect to the language, what I do find are statements like [this](https://en.cppreference.com/w/cpp/memory/new/operator_delete#Class-specific_overloads)

> ...  Regardless of which deallocation function would be executed at run time, the statically visible version of operator delete must be accessible in order to compile.  ...

And indeed g++ 10.2 does fail at compilation.  From Jeff's comment about "undefined symbols", I gather that this wasn't always a compile error.

From what I can tell, if `class T` provides `void* T::operator new(size_t, ...)` then it must also provide `void T::operator delete(void*)` even when it is never called (or callable via `private`).  I think this is analogous to the need to define a pure virtual destructor.

(Note this required `operator delete` has *no arguments*.  So it is distinct from the "placement delete" operator `void T::operator delete(void*, ...)` which is only called if the corresponding placement `T::operator new(size_t, ...)` throws an exception!)

Poking around with GCC 10 for amd64 and ARM7, what I see is that if a member `T::operator new` is provided, then the implicitly generated "deleting destructor" (aka. the `_ZN*D0Ev` symbol) will switch from calling `operator delete(void*, unsigned int)` to `T::operator delete(void*, unsigned int)`.  This "deleting destructor" is referenced from the vtable (aka `_ZTV*`), thus must be defined if any instance `T` (or a sub-class) is constructed.

As far as I can tell, the c++ language (<= c++20) does not define a way to omit the "deleting destructor" when it shouldn't be used.

```c++
struct T {
    ...
    static void operator delete(void *p) = delete;
};
```

Appears to be valid syntax, but fails to compile with `error: use of deleted function`.

The closest approximation seems to be making `void T::operator delete(void*)` private, which is possible so long as `void T::operator delete(void*, ...)` is also provided and accessible wherever `new` is called.  I'm not sure how this fits with the above quote about `... operator delete must be accessible ...`.

This is done by all of the classes in Base which define a placement new that I looked at.

anjohnson, 2022-09-28:
Core Group 9/28: This needs the comments adjusting, when someone has time.

anjohnson, 2025-02-19:
Core Group: This isn't just for MS compilers, the task is just to remove that aspect from the comments.