epics-base/epics-base #293: Tests with custom .plt files do not track their dependencies

simon-ess, 2022-09-01:
There are a few tests in EPICS base that have .plt files that are meant to replace the standard .t files generated by `makeTestfile.pl`. In most cases, these are their own independent tests, but in a few cases they have source files that need to be compiled in order for the test to run:
* modules/libcom/test/epicsUnitTestTest
* modules/libcom/test/yajlTest

However, the rules from RULES_BUILD do not take this into account:
```make
# If there's a perl test script (.plt) available, use it
%.t: ../%.plt
	@$(RM) $@
	$(EXPAND_TOOL) -t $(INSTALL_LOCATION) -a $(T_A) $< $@

# Test programs (.t files) must be written in Perl.
# Generate a perl program to exec the real test binary.
%.t: %$(EXE) $(TOOLS)/makeTestfile.pl
	@$(RM) $@
	$(PERL) $(TOOLS)/makeTestfile.pl $(T_A) $(EPICS_HOST_ARCH) $@ $<
```

As a consequence, if you happen to modify or delete the source files that are used in these tests, then the tests will fail upon a repeated invocation of `make runtests`.

anjohnson, 2022-09-01:
Doesn't this just mean that we need to add something like this dependency?
```makefile
runtests: build
```
The final version won't actually be quite that, but I think it's close enough to confirm whether that does what you're looking for.

anjohnson, 2022-09-03:
Fixed with 54c02e2c247def09ebfbfc501e29c1b4e25d42a8

closed 2022-09-03 by anjohnson