epics-base/epics-base #302: Migrate lgtm.com To GHA Code Scanning

anjohnson, 2022-11-07:
lgtm.com was bought by GitHub (MS) some time back, and all their functionality has now been made available through GitHub's Code Scanning. This issue is to record the documentation I've seen recently about that in case someone else wants to work on migrating our GHA configuration to use that instead — I might not have time for a while.

* [The next step for LGTM.com: GitHub code scanning!](https://github.blog/2022-08-15-the-next-step-for-lgtm-com-github-code-scanning/)
* [Configuring code scanning](https://docs.github.com/en/code-security/code-scanning/automatically-scanning-your-code-for-vulnerabilities-and-errors/configuring-code-scanning) — Some internal links on this page that look like they would be useful don't actually work.

mdavidsaver, 2024-06-19:
@anjohnson Should this issue, and #453 be "help wanted"?

anjohnson, 2024-06-24:
#453 has been waiting for you to look at the page I linked to in my last reply, in response to your request. The results are still present at that link. If there's "help wanted" we should provide some explanation of what we're asking to be done.