epics-base/epics-base #317: Document built-in device support

anjohnson, 2022-12-28:
Some of the Base record types provide documentation for soft device support in their POD text, but not all (the Raw and Async Soft Channel supports are mostly undocumented).

The existing text should be moved to one or more `.dbd.pod` files nearer to the device support code, expanded as appropriate, and appropriate descriptions written for the support layers that are currently undocumented.

Links to the new files should be added to the `documentation/RecordReference.md` index.

marciodo, 2024-06-12:
I'm claiming this for the Codeathon 2024.

marciodo, 2024-06-12:
`documentation/RecordReference.md` changed to `documentation/ComponentReference.pod`.

marciodo, 2024-06-12:
List of pod files in `modules/database/src/std/rec` (checked items already have the description for Soft Channel and Raw Channel, or don't have device support):
- [ ] aai
- [ ] aao
- [ ] ai
- [ ] ao
- [ ] aSub
- [ ] bi
- [ ] bo
- [ ] calcout
- [ ] calc
- [ ] compress
- [ ] dfanout
- [ ] event
- [ ] fanout
- [ ] histogram
- [ ] int64in
- [ ] int64out
- [ ] longin
- [ ] longout
- [ ] lsi
- [ ] lso
- [ ] mbbiDirect
- [ ] mbbi
- [ ] mbboDirect
- [ ] mbbo
- [ ] permissive
- [ ] printf
- [ ] sel
- [ ] seq
- [ ] state
- [ ] stringin
- [ ] stringout
- [ ] subArray
- [ ] sub
- [ ] waveform