epics-base/epics-base #318: MacOS build warnings: -undefined dynamic_lookup may not work with chained fixups

anjohnson, 2023-01-01:
Apple's XCode 14 warns about the use of `-undefined dynamic_lookup` when linking shared and loadable libraries (bundles). I can remove that flag from `SHRLIB_LDFLAGS` and Base seems to build properly, but it is required in `LOADABLE_SHRLIB_LDFLAGS` to build the `Cap5.bundle` Perl bindings (both are set in configure/os/CONFIG.darwinCommon.darwinCommon). Other EPICS code that builds loadable libraries will probably also get this warning with XCode 14.

We are not the only software to see this, a similar [CPython issue](https://github.com/python/cpython/issues/97524) exists (which has some links to reports for other software too including Perl) and Apple have responded to that, e.g. in their [Developer Forums](https://developer.apple.com/forums/thread/719961), although they haven't said yet what the resolution is likely to be.

This issue is a reminder to monitor the CPython issue for further information and their final solution.

anjohnson, 2023-01-01:
An Apple-specific linker description covering what chained fixups are can be found in [these WWDC notes](https://www.wwdcnotes.com/notes/wwdc22/110362/).

mdavidsaver, 2024-06-19:
@anjohnson or other XCode user.  Any update?  I notice some activity on the linked python issue.

anjohnson, 2024-06-27:
Using XCode 15.0.1 with the `-undefined dynamic_lookup` option present the warning has disappeared, but the Perl bindings still work just fine. Removing that option causes the link of the `Cap5.bundle` library to fail with many undefined symbols, so it looks like we don't need to do anything to our build flags, and this issue seem to be moot.

closed 2024-06-27 by anjohnson