epics-base/epics-base #330: tsDLList is private  within this context (3.14 FTBFS)

mdavidsaver, 2023-02-07:
The 3.14 branch no longer compiles with recent GCC.

```
 In file included from ../../../../src/cas/generic/caServer.cc:21:
../../../../src/cas/generic/caServerI.h: At global scope:
../../../../src/cas/generic/caServerI.h:91:5: error: ‘class tsDLList<ioBlocked> tsDLList<ioBlocked>::tsDLList’ is private within this context
   91 |     tsDLList < casStrmClient > clientList;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../../../src/cas/generic/caServerI.h:33,
                 from ../../../../src/cas/generic/caServer.cc:21:
../../../../src/cas/generic/ioBlocked.h:43:7: note: declared private here
   43 | class ioBlockedList : private tsDLList<ioBlocked> {
      |       ^~~~~~~~~~~~~
In file included from ../../../../src/cas/generic/caServer.cc:21:
../../../../src/cas/generic/caServerI.h:92:5: error: ‘class tsDLList<ioBlocked> tsDLList<ioBlocked>::tsDLList’ is private within this context
   92 |     tsDLList < casIntfOS > intfList;
      |     ^~~~~~~~~~~~~~~~~~~~~~
In file included from ../../../../src/cas/generic/caServerI.h:33,
                 from ../../../../src/cas/generic/caServer.cc:21:
../../../../src/cas/generic/ioBlocked.h:43:7: note: declared private here
   43 | class ioBlockedList : private tsDLList<ioBlocked> {
      |       ^~~~~~~~~~~~~
```

anjohnson, 2023-02-07:
These errors are coming from code in src/cas/generic, so this is PCAS code. Presumably the stand-alone version of CAS will also fail against these GCC versions, and that is something that we're probably going to want to fix somehow, so it might be a good place to experiment. Does downgrading the C++ standard version for builds in this directory work?

mdavidsaver, 2023-02-07:
fyi. with GHA this is an issue beginning with the `ubuntu-22.04` image.  Builds with the `ubuntu-20.04` image still succeed.

ralphlange, 2023-02-07:
This is on my to-do-list (since the telecom before our last) and I intend to take care of it.

ralphlange, 2023-02-08:
Fixed by https://github.com/epics-base/epics-base/commit/612d57a9b3aa651960400f651ff93d8bdc858dc8

closed 2023-02-08 by ralphlange