epics-base/epics-base #331: `db_create_read_log()` w/o locking

mdavidsaver, 2023-02-22:
It looks to me like `db_create_read_log()` is called without locking the associated record.  So the target field value, and time+alarm meta-data are accessed without locking during a CA GET.

https://github.com/epics-base/epics-base/blob/718da5c9be96b7eccd7ce3a95f3b2d232c11ca32/modules/database/src/ioc/rsrv/camessage.c#L538-L548

mdavidsaver, 2023-02-22:
Appears to `valgrind --tool=helgrind` as:

```
...
Possible data race during write of size 2 at 0x5175D7C by thread #14
Locks held: 1, at address 0x51845D0
   at 0x48D4A28: recGblResetAlarms (recGbl.c:197)
   by 0x485E15B: monitor (calcRecord.c:401)
   by 0x485E15B: process (calcRecord.c:132)
   by 0x48B5E63: dbProcess (dbAccess.c:610)
   by 0x48CB2C3: scanList (dbScan.c:1012)
   by 0x48CB6D7: periodicTask (dbScan.c:804)
   by 0x496B7AC: start_routine (osdThread.c:441)
   by 0x483F876: mythread_wrapper (hg_intercepts.c:387)
   by 0x4F41EA6: start_thread (pthread_create.c:477)
   by 0x4CC9A2E: clone (clone.S:95)

This conflicts with a previous read of size 2 by thread #22
Locks held: 1, at addresses  (and 1 that can't be shown)
   at 0x48CE6C5: db_create_field_log (dbEvent.c:709)
   by 0x48CE6C5: db_create_read_log (dbEvent.c:764)
   by 0x48FBC0F: read_reply (camessage.c:540)
   by 0x48FD912: read_notify_action (camessage.c:745)
   by 0x48FF067: camessage (camessage.c:2546)
   by 0x48FAC5F: camsgtask (camsgtask.c:116)
   by 0x496B7AC: start_routine (osdThread.c:441)
   by 0x483F876: mythread_wrapper (hg_intercepts.c:387)
   by 0x4F41EA6: start_thread (pthread_create.c:477)
...
```

anjohnson, 2023-02-22:
Ack. The 3.15 version of `db_create_read_log()` didn't access the record at all, and the filters run by the `dbChannelRunP*Chain()` routines look to do their own locking, so unfortunately I suspect this points to 27fe3e4 as when this came in.

mdavidsaver, 2023-02-23:
How about moving the handling of the `pfl==NULL` case into `dbChannel_get_count()`?  I think this could be done in a way which is transparent to callers.