epics-base/epics-base #332: epicsTimeTest fails 7 tests between #16-#28 on 32-bit linux-arm target

anjohnson, 2023-02-22:
I'm running the binary cross-compiled from `modules/libcom/test/epicsTimeTest.cpp` from EPICS 7.0.6.1 built using gcc 4.7.2 (arm-xilinx-linux-gnueabi target).

These are the test results from my armv7l linux-arm device:
```
not ok 16 - testDiff(1:999999999 - 2:0) -> 4294967296.000000000 ~= -0.000000001 (4.29497e+09)
ok 17 - testDiff(1:0 - 4294967295:0) -> 2.000000000 ~= 2.000000000 (0)
not ok 18 - testDiff(4294967295:0 - 1:0) -> 4294967294.000000477 ~= -2.000000000 (4.29497e+09)
ok 19 - testDiff(1:999999999 - 4294967295:999999999) -> 2.000000000 ~= 2.000000000 (0)
not ok 20 - testDiff(4294967295:999999999 - 1:999999999) -> 4294967294.000000477 ~= -2.000000000 (4.29497e+09)
ok 21 - testDiff(0:999999999 - 4294967295:0) -> 1.999999999 ~= 1.999999999 (2.22045e-16)
not ok 22 - testDiff(4294967295:0 - 0:999999999) -> 4294967294.000000477 ~= -1.999999999 (4.29497e+09)
ok 23 - testDiff(2147483648:0 - 2147483647:0) -> 1.000000000 ~= 1.000000000 (0)
not ok 24 - testDiff(2147483647:0 - 2147483648:0) -> 4294967295.000000477 ~= -1.000000000 (4.29497e+09)
ok 25 - testDiff(2147483648:0 - 2147483647:999999999) -> 0.000000001 ~= 0.000000001 (0)
not ok 26 - testDiff(2147483647:999999999 - 2147483648:0) -> 4294967296.000000000 ~= -0.000000001 (4.29497e+09)
ok 27 - testDiff(2147483648:999999999 - 2147483647:0) -> 1.999999999 ~= 1.999999999 (2.22045e-16)
not ok 28 - testDiff(2147483647:0 - 2147483648:999999999) -> 4294967294.000000477 ~= -1.999999999 (4.29497e+09)
```

On an i686 linux-x86 device I get these passing results:
```
ok 16 - testDiff(1:999999999 - 2:0) -> -0.000000001 ~= -0.000000001 (0)
ok 17 - testDiff(1:0 - 4294967295:0) -> 2.000000000 ~= 2.000000000 (1.24466e-16)
ok 18 - testDiff(4294967295:0 - 1:0) -> -2.000000000 ~= -2.000000000 (-1.24466e-16)
ok 19 - testDiff(1:999999999 - 4294967295:999999999) -> 2.000000000 ~= 2.000000000 (1.24466e-16)
ok 20 - testDiff(4294967295:999999999 - 1:999999999) -> -2.000000000 ~= -2.000000000 (-1.24466e-16)
ok 21 - testDiff(0:999999999 - 4294967295:0) -> 1.999999999 ~= 1.999999999 (2.07299e-16)
ok 22 - testDiff(4294967295:0 - 0:999999999) -> -1.999999999 ~= -1.999999999 (-2.07299e-16)
ok 23 - testDiff(2147483648:0 - 2147483647:0) -> 1.000000000 ~= 1.000000000 (6.22332e-17)
ok 24 - testDiff(2147483647:0 - 2147483648:0) -> -1.000000000 ~= -1.000000000 (-6.22332e-17)
ok 25 - testDiff(2147483648:0 - 2147483647:999999999) -> 0.000000001 ~= 0.000000001 (0)
ok 26 - testDiff(2147483647:999999999 - 2147483648:0) -> -0.000000001 ~= -0.000000001 (0)
ok 27 - testDiff(2147483648:999999999 - 2147483647:0) -> 1.999999999 ~= 1.999999999 (2.07299e-16)
ok 28 - testDiff(2147483647:0 - 2147483648:999999999) -> -1.999999999 ~= -1.999999999 (-2.07299e-16)
```

It's not obvious from looking at the code whether this is an artifact of the test code or of the `epicsTimeDiffInSeconds()` API routine that it's calling. I suspect it was introduced by cb8c7998b62701a849a6fa9c299cc1613f66a627 but I don't have a build of an earlier Base version for that target to confirm that.

If someone has access to a Raspberry Pi running the 32-bit OS it would be nice to compare the output on that to see if you get the same failures.

tboegi, 2023-02-23:
Running it on my old raspi:
`uname -a
Linux tb-raspi4 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

ok 16 - testDiff(1:999999999 - 2:0) -> -0.000000001 ~= -0.000000001 (0)
ok 17 - testDiff(1:0 - 4294967295:0) -> 2.000000000 ~= 2.000000000 (0)
ok 18 - testDiff(4294967295:0 - 1:0) -> -2.000000000 ~= -2.000000000 (0)
ok 19 - testDiff(1:999999999 - 4294967295:999999999) -> 2.000000000 ~= 2.000000000 (0)
ok 20 - testDiff(4294967295:999999999 - 1:999999999) -> -2.000000000 ~= -2.000000000 (0)
ok 21 - testDiff(0:999999999 - 4294967295:0) -> 1.999999999 ~= 1.999999999 (2.22045e-16)
ok 22 - testDiff(4294967295:0 - 0:999999999) -> -1.999999999 ~= -1.999999999 (-2.22045e-16)
ok 23 - testDiff(2147483648:0 - 2147483647:0) -> 1.000000000 ~= 1.000000000 (0)
ok 24 - testDiff(2147483647:0 - 2147483648:0) -> -1.000000000 ~= -1.000000000 (0)
ok 25 - testDiff(2147483648:0 - 2147483647:999999999) -> 0.000000001 ~= 0.000000001 (0)
ok 26 - testDiff(2147483647:999999999 - 2147483648:0) -> -0.000000001 ~= -0.000000001 (0)
ok 27 - testDiff(2147483648:999999999 - 2147483647:0) -> 1.999999999 ~= 1.999999999 (2.22045e-16)
ok 28 - testDiff(2147483647:0 - 2147483648:999999999) -> -1.999999999 ~= -1.9999

`

anjohnson, 2023-02-23:
@tboegi Interesting, thanks. What version of gcc was that built with?

My binaries were cross-built with a Xilinx tool-set (on a 32-bit host, I'm currently trying to find the 64-bit tool-set):
```
libera-sdev% ./gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc --version
arm-xilinx-linux-gnueabi-gcc (Sourcery CodeBench Lite 2012.09-104) 4.7.2
```
The device running the test is
```
# uname -a
Linux libera 3.15.0-itech #19 SMP PREEMPT Mon Dec 21 18:52:00 CET 2020 armv7l GNU/Linux
```

mdavidsaver, 2023-02-23:
In addition to compiler version, please also list any compiler arguments which might effect code generation.  eg. `-march=...` and similar.

mdavidsaver, 2023-02-23:
@anjohnson

```
gcc -Q --help=target
```

... will print the default `-march` and others.  Note that this may not be what is expected based on `uname -a`.  eg. a raspbian bullseye (gcc 10.2.1) install on a pi3 defaults to `-march=armv6+fp`.

tboegi, 2023-02-23:
The raspi is running raspbian stretch

$ gcc --version
gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

`$ gcc -Q --help=target
The following options are target specific:
  -mabi=                      		aapcs-linux
  -mabort-on-noreturn         		[disabled]
  -mandroid                   		[disabled]
  -mapcs                      		[disabled]
  -mapcs-float                		[disabled]
  -mapcs-frame                		[disabled]
  -mapcs-reentrant            		[disabled]
  -mapcs-stack-check          		[disabled]
  -march=                     		armv6
  -marm                       		[enabled]
  -masm-syntax-unified        		[disabled]
  -mbig-endian                		[disabled]
  -mbionic                    		[disabled]
  -mcallee-super-interworking 		[disabled]
  -mcaller-super-interworking 		[disabled]
  -mcpu=                      		[default]
  -mfix-cortex-m3-ldrd        		[enabled]
  -mflip-thumb                		[disabled]
  -mfloat-abi=                		hard
  -mfp16-format=              		none
  -mfpu=                      		vfp
  -mglibc                     		[enabled]
  -mhard-float                		
  -mlittle-endian             		[enabled]
  -mlong-calls                		[disabled]
  -mmusl                      		[disabled]
  -mneon-for-64bits           		[disabled]
  -mnew-generic-costs         		[disabled]
  -mold-rtx-costs             		[disabled]
  -mpic-data-is-text-relative 		[enabled]
  -mpic-register=             		
  -mpoke-function-name        		[disabled]
  -mprint-tune-info           		[disabled]
  -mrestrict-it               		[enabled]
  -msched-prolog              		[enabled]
  -msingle-pic-base           		[disabled]
  -mslow-flash-data           		[disabled]
  -msoft-float                		
  -mstructure-size-boundary=  		0x20
  -mthumb                     		[disabled]
  -mthumb-interwork           		[enabled]
  -mtls-dialect=              		gnu
  -mtp=                       		auto
  -mtpcs-frame                		[disabled]
  -mtpcs-leaf-frame           		[disabled]
  -mtune=                     		[default]
  -muclibc                    		[disabled]
  -munaligned-access          		[enabled]
  -mvectorize-with-neon-double 		[disabled]
  -mvectorize-with-neon-quad  		[enabled]
  -mword-relocations          		[disabled]

  Known ARM ABIs (for use with the -mabi= option):
    aapcs aapcs-linux apcs-gnu atpcs iwmmxt

  Known ARM architectures (for use with the -march= option):
    armv2 armv2a armv3 armv3m armv4 armv4t armv5 armv5e armv5t armv5te armv6 armv6-m armv6j armv6k armv6kz armv6s-m armv6t2 armv6z armv6zk armv7 armv7-a armv7-m
    armv7-r armv7e-m armv7ve armv8-a armv8-a+crc armv8.1-a armv8.1-a+crc iwmmxt iwmmxt2 native

  Known __fp16 formats (for use with the -mfp16-format= option):
    alternative ieee none

  Known ARM FPUs (for use with the -mfpu= option):
    crypto-neon-fp-armv8 fp-armv8 fpv4-sp-d16 fpv5-d16 fpv5-sp-d16 neon neon-fp-armv8 neon-fp16 neon-vfpv4 vfp vfp3 vfpv3 vfpv3-d16 vfpv3-d16-fp16 vfpv3-fp16 vfpv3xd
    vfpv3xd-fp16 vfpv4 vfpv4-d16

  Valid arguments to -mtp=:
    auto cp15 soft

  Known floating-point ABIs (for use with the -mfloat-abi= option):
    hard soft softfp

  Known ARM CPUs (for use with the -mcpu= and -mtune= options):
    arm1020e arm1020t arm1022e arm1026ej-s arm10e arm10tdmi arm1136j-s arm1136jf-s arm1156t2-s arm1156t2f-s arm1176jz-s arm1176jzf-s arm2 arm250 arm3 arm6 arm60
    arm600 arm610 arm620 arm7 arm70 arm700 arm700i arm710 arm7100 arm710c arm710t arm720 arm720t arm740t arm7500 arm7500fe arm7d arm7di arm7dm arm7dmi arm7m arm7tdmi
    arm7tdmi-s arm8 arm810 arm9 arm920 arm920t arm922t arm926ej-s arm940t arm946e-s arm966e-s arm968e-s arm9e arm9tdmi cortex-a12 cortex-a15 cortex-a15.cortex-a7
    cortex-a17 cortex-a17.cortex-a7 cortex-a32 cortex-a35 cortex-a5 cortex-a53 cortex-a57 cortex-a57.cortex-a53 cortex-a7 cortex-a72 cortex-a72.cortex-a53 cortex-a8
    cortex-a9 cortex-m0 cortex-m0.small-multiply cortex-m0plus cortex-m0plus.small-multiply cortex-m1 cortex-m1.small-multiply cortex-m3 cortex-m4 cortex-m7
    cortex-r4 cortex-r4f cortex-r5 cortex-r7 cortex-r8 ep9312 exynos-m1 fa526 fa606te fa626 fa626te fa726te fmp626 generic-armv7-a iwmmxt iwmmxt2 marvell-pj4 mpcore
    mpcorenovfp native qdf24xx strongarm strongarm110 strongarm1100 strongarm1110 xgene1 xscale

  TLS dialect to use:
    gnu gnu2
`

anjohnson, 2023-02-23:
```
libera-sdev% ./gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-gcc -Q --help=target
The following options are target specific:
  -mabi=                      		aapcs-linux
  -mabort-on-noreturn         		[disabled]
  -mandroid                   		[disabled]
  -mapcs                      		[disabled]
  -mapcs-float                		[disabled]
  -mapcs-frame                		[disabled]
  -mapcs-reentrant            		[disabled]
  -mapcs-stack-check          		[disabled]
  -march=                     		armv7-a
  -marm                       		[enabled]
  -mbig-endian                		[disabled]
  -mbionic                    		[disabled]
  -mcallee-super-interworking 		[disabled]
  -mcaller-super-interworking 		[disabled]
  -mcirrus-fix-invalid-insns  		[disabled]
  -mcpu=                      		[default]
  -mfix-cortex-m3-ldrd        		[enabled]
  -mfloat-abi=                		softfp
  -mfp16-format=              		none
  -mfp=2                      		
  -mfp=3                      		
  -mfpe                       		[disabled]
  -mfpe=2                     		
  -mfpe=3                     		
  -mfpu=                      		neon-fp16
  -mglibc                     		[enabled]
  -mhard-float                		
  -mlittle-endian             		[enabled]
  -mlong-calls                		[disabled]
  -mpic-register=             		
  -mpoke-function-name        		[disabled]
  -msched-prolog              		[enabled]
  -msingle-pic-base           		[disabled]
  -msoft-float                		
  -mstructure-size-boundary=  		0x20
  -mthumb                     		[disabled]
  -mthumb-interwork           		[enabled]
  -mtls-dialect=              		gnu
  -mtp=                       		auto
  -mtpcs-frame                		[disabled]
  -mtpcs-leaf-frame           		[disabled]
  -mtune=                     		[default]
  -muclibc                    		[disabled]
  -munaligned-access          		[enabled]
  -mvectorize-with-neon-double 		[disabled]
  -mvectorize-with-neon-quad  		[enabled]
  -mword-relocations          		[disabled]
  -mwords-little-endian       		[disabled]

  Known ARM ABIs (for use with the -mabi= option):
    aapcs aapcs-linux apcs-gnu atpcs iwmmxt

  Known ARM architectures (for use with the -march= option):
    armv2 armv2a armv3 armv3m armv4 armv4t armv5 armv5e armv5t armv5te armv6
    armv6-m armv6j armv6k armv6s-m armv6t2 armv6z armv6zk armv7 armv7-a armv7-m
    armv7-r armv7e-m ep9312 iwmmxt iwmmxt2 native

  Known __fp16 formats (for use with the -mfp16-format= option):
    alternative ieee none

  Known ARM FPUs (for use with the -mfpu= option):
    fpa fpe2 fpe3 fpv4-sp-d16 maverick neon neon-fp16 neon-vfpv4 vfp vfp3 vfpv3
    vfpv3-d16 vfpv3-d16-fp16 vfpv3-fp16 vfpv3xd vfpv3xd-fp16 vfpv4 vfpv4-d16

  Valid arguments to -mtp=:
    auto cp15 soft

  Known floating-point ABIs (for use with the -mfloat-abi= option):
    hard soft softfp

  Known ARM CPUs (for use with the -mcpu= and -mtune= options):
    arm1020e arm1020t arm1022e arm1026ej-s arm10e arm10tdmi arm1136j-s
    arm1136jf-s arm1156t2-s arm1156t2f-s arm1176jz-s arm1176jzf-s arm2 arm250
    arm3 arm6 arm60 arm600 arm610 arm620 arm7 arm70 arm700 arm700i arm710
    arm7100 arm710c arm710t arm720 arm720t arm740t arm7500 arm7500fe arm7d
    arm7di arm7dm arm7dmi arm7m arm7tdmi arm7tdmi-s arm8 arm810 arm9 arm920
    arm920t arm922t arm926ej-s arm940t arm946e-s arm966e-s arm968e-s arm9e
    arm9tdmi cortex-a15 cortex-a5 cortex-a7 cortex-a8 cortex-a9 cortex-m0
    cortex-m1 cortex-m3 cortex-m4 cortex-r4 cortex-r4f cortex-r5 ep9312 fa526
    fa606te fa626 fa626te fa726te fmp626 generic-armv7-a iwmmxt iwmmxt2 mpcore
    mpcorenovfp native strongarm strongarm110 strongarm1100 strongarm1110 xscale

  TLS dialect to use:
    gnu gnu2
```
So the main difference to Torsten's is that my compiler defaults to `-mfloat-abi=softfp`.

The build commands are:
```
/home/libera/Xilinx/ise/EDK/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-g++  -D_GNU_SOURCE -D_DEFAULT_SOURCE            -DUNIX  -Dlinux      -O3   -Wall            -D_FILE_OFFSET_BITS=64  -I. -I../O.Common -I. -I. -I.. -I../../../../include/compiler/gcc -I../../../../include/os/Linux -I../../../../include         -I/home/libera/Xilinx/ise/EDK/gnu/arm/lin/arm-xilinx-linux-gnueabi/include -c ../epicsTimeTest.cpp
/home/libera/Xilinx/ise/EDK/gnu/arm/lin/bin/arm-xilinx-linux-gnueabi-g++ -o epicsTimeTest  -L/C2/ioc-modules/build/BASE/7.0.6.1-C2-1+LIB-003/lib/linux-arm -Wl,-rpath,/C2/ioc-modules/build/BASE/7.0.6.1-C2-1+LIB-003/lib/linux-arm            -rdynamic -L/home/libera/Xilinx/ise/EDK/gnu/arm/lin/arm-xilinx-linux-gnueabi/lib         epicsTimeTest.o    -lCom   -lpthread    -lm -lrt -ldl -lgcc
```

Any suggestions for compiler flags to try, given this?
```
# cat /proc/cpuinfo                         
processor	: 0
model name	: ARMv7 Processor rev 0 (v7l)
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

processor	: 1
model name	: ARMv7 Processor rev 0 (v7l)
Features	: swp half thumb fastmult vfp edsp neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0
```

Thanks!

mdavidsaver, 2023-02-24:
> Any suggestions for compiler flags to try, given this?

I don't think compiler flags are going to "fix" this.  I'm trying to understand what is different.  I agree the soft-float seems the most likely candidate, although I don't understand why.  You might try disassembling and/or stepping through `epicsTimeAddSeconds()` to see what happens.  `epicsTimeDiffInSeconds()` is also a candidate, although I think less likely.

mdavidsaver, 2023-02-24:
Stating the perhaps less than obvious, the failing tests are the ones where the output difference should be negative.  Some changes I'm playing around with, which do not change the outcome on amd64.

```diff
index 9d00acec2..171a2da21 100644
@@ -425,7 +425,7 @@ double epicsStdCall epicsTimeDiffInSeconds (const epicsTimeStamp *pLeft, const e
     // handle over/underflow as u32 when subtracting
     epicsInt64 nsec = epicsInt32(pLeft->secPastEpoch - pRight->secPastEpoch);
     nsec *= nSecPerSec;
-    nsec += epicsInt32(pLeft->nsec) - epicsInt32(pRight->nsec);
+    nsec += epicsInt32(pLeft->nsec - pRight->nsec);
 
     return double(nsec)*1e-9;
 }
@@ -435,9 +435,9 @@ void epicsStdCall epicsTimeAddSeconds (epicsTimeStamp *pDest, double seconds)
     epicsInt64 nsec = pDest->secPastEpoch;
     nsec *= nSecPerSec;
     nsec += epicsInt64(pDest->nsec);
-    nsec += epicsInt64(seconds*1e9 + (seconds>=0.0 ? 0.5 : -0.5));
+    nsec += epicsInt64(llround(seconds*1e9));
     pDest->secPastEpoch = nsec/nSecPerSec;
-    pDest->nsec         = (nsec>=0 ? nsec : -nsec)%nSecPerSec;
+    pDest->nsec         = (nsec%nSecPerSec)&0x7fffffff; // llabs(nsec%nSecPerSec)
 }
```

mdavidsaver, 2023-02-25:
> Optimizer bug, I took out the -O3 for compiling epicsTime.cpp and it passes.

I hesitate to simply blame GCC.  It seems to me more likely that there is UB lurking.  I would be more comfortable considering this "resolved" if a newer `arm-xilinx-linux-gnueabi-gcc` produced the expected result.

mdavidsaver, 2023-02-25:
I've remembered that I have an old version of the xilinx toolchain.  As luck would have it, this includes gcc 4.7.2.

Compiling [epicsTimeDiffInSeconds.cpp](https://github.com/epics-base/epics-base/files/10831864/epicsTimeDiffInSeconds.c.txt) with `gcc -o /tmp/out.s -S -fverbose-asm /tmp/epicsTimeDiffInSeconds.cpp  -O3` with the [raspbian](https://github.com/epics-base/epics-base/files/10831863/pi.s.txt) and [xilinx](https://github.com/epics-base/epics-base/files/10831866/xilinx.s.txt) toolchains shows one key difference.

```
	smlal	r0, r1, r3, ip	@,, tmp129, tmp138
```

vs.

```
	umlal	r4, r5, ip, r1	@, tmp161, D.6581
```

Use of the unsigned [`umlal`](https://developer.arm.com/documentation/dui0473/m/arm-and-thumb-instructions/umlal) instead of the signed [`smlal`](https://developer.arm.com/documentation/dui0473/m/arm-and-thumb-instructions/smlal) would certainly explain the difference in output.  Now... why?

mdavidsaver, 2023-02-25:
It seems like my tinkering in https://github.com/epics-base/epics-base/issues/332#issuecomment-1444127926 was in the right area, but going the wrong direction.  This change has the desired effect of emitting `smlal` with both ARM toolchains.  Which also makes my comment incorrect.

```diff
@@ -423,7 +423,7 @@ double epicsStdCall epicsTimeDiffInSeconds (const epicsTimeStamp *pLeft, const e
      */
 
     // handle over/underflow as u32 when subtracting
-    epicsInt64 nsec = epicsInt32(pLeft->secPastEpoch - pRight->secPastEpoch);
+    epicsInt64 nsec = epicsInt32(pLeft->secPastEpoch) - epicsInt32(pRight->secPastEpoch);
     nsec *= nSecPerSec;
     nsec += epicsInt32(pLeft->nsec) - epicsInt32(pRight->nsec);
```

I still don't understand why this makes a difference.  It is like the cast of the intermediate difference to `epicsInt32` is ignored.

I also notice that `nSecPerSec` is unsigned, but changing this doesn't effect which instruction is select.

mdavidsaver, 2023-02-25:
... but with this change the clang amd64 GHA jobs and linux and osx now fail a slightly different set of tests.

```
  not ok 23 - testDiff(2147483648:0 - 2147483647:0) -> -4294967295.000000477 ~= 1.000000000 (-4.29497e+09)
  not ok 24 - testDiff(2147483647:0 - 2147483648:0) -> 4294967295.000000477 ~= -1.000000000 (4.29497e+09)
  not ok 25 - testDiff(2147483648:0 - 2147483647:999999999) -> -4294967296.000000000 ~= 0.000000001 (-4.29497e+09)
  not ok 26 - testDiff(2147483647:999999999 - 2147483648:0) -> 4294967296.000000000 ~= -0.000000001 (4.29497e+09)
  not ok 27 - testDiff(2147483648:999999999 - 2147483647:0) -> -4294967294.000000477 ~= 1.999999999 (-4.29497e+09)
  not ok 28 - testDiff(2147483647:0 - 2147483648:999999999) -> 4294967294.000000477 ~= -1.999999999 (4.29497e+09)
```

anjohnson, 2023-02-26:
Since you reopened this, I will submit a related issue that I wrote up last night which points out that some of your tests calling `testDiff()` (cases 17 through 22) actually give the wrong expected answer, and as a result your implementation of `epicsTimeDiffInSeconds()` is also incorrect. I'll let you read that issue to understand it, I've included an implementation that works for me on darwin-aarch64, 32- and 64-bit Ubuntu and on my Xilinx linux-arm target. I think that code is easier to understand but it might also have hidden UB so you might want to recode it.

Thanks!

anjohnson, 2023-03-01:
This routine passes all my [modified epicsTimeTest cases](https://github.com/epics-base/epics-base/issues/333#issuecomment-1445259224) on all the arch's I've tried it on:
```c++
double epicsStdCall epicsTimeDiffInSeconds (const epicsTimeStamp *pLeft, const epicsTimeStamp *pRight)
{
    epicsInt64 delta = pLeft->secPastEpoch;
    delta -= pRight->secPastEpoch;
    delta *= nSecPerSec;
    delta += pLeft->nsec;
    delta -= pRight->nsec;

    return double(delta) / nSecPerSec;
}
```

This version fails tests 17-22 on linux-arm, linux-x86 and linux-x86_64:
```c++
double epicsStdCall epicsTimeDiffInSeconds (const epicsTimeStamp *pLeft, const epicsTimeStamp *pRight)
{
    epicsInt64 delta = pLeft->secPastEpoch;
    delta -= pRight->secPastEpoch;
    delta *= nSecPerSec;
    delta += pLeft->nsec;
    delta -= pRight->nsec;

    return double(delta) * 1e-9;
}
```

Here are the failure results on linux-arm
```
not ok 17 - testDiff(1:0 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-4.76837e-07)
not ok 18 - testDiff(4294967295:0 - 1:0) -> 4294967294.000000477 ~= 4294967294.000000000 (4.76837e-07)
not ok 19 - testDiff(1:999999999 - 4294967295:999999999) -> -4294967294.000000477 ~= -4294967294.000000000 (-4.76837e-07)
not ok 20 - testDiff(4294967295:999999999 - 1:999999999) -> 4294967294.000000477 ~= 4294967294.000000000 (4.76837e-07)
not ok 21 - testDiff(0:999999999 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-4.76837e-07)
not ok 22 - testDiff(4294967295:0 - 0:999999999) -> 4294967294.000000477 ~= 4294967295.000000000 (-1)
```

On linux-x86_64 the output is almost but not quite identical:
```
not ok 17 - testDiff(1:0 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-4.76837e-07)
not ok 18 - testDiff(4294967295:0 - 1:0) -> 4294967294.000000477 ~= 4294967294.000000000 (4.76837e-07)
not ok 19 - testDiff(1:999999999 - 4294967295:999999999) -> -4294967294.000000477 ~= -4294967294.000000000 (-4.76837e-07)
not ok 20 - testDiff(4294967295:999999999 - 1:999999999) -> 4294967294.000000477 ~= 4294967294.000000000 (4.76837e-07)
not ok 21 - testDiff(0:999999999 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-4.76837e-07)
not ok 22 - testDiff(4294967295:0 - 0:999999999) -> 4294967294.000000477 ~= 4294967294.000000000 (4.76837e-07)
```
In particular note the results from test 22 above.

On linux-x86 there are more differences:
```
not ok 17 - testDiff(1:0 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-2.67522e-07)
not ok 18 - testDiff(4294967295:0 - 1:0) -> 4294967294.000000477 ~= 4294967294.000000000 (2.67522e-07)
not ok 19 - testDiff(1:999999999 - 4294967295:999999999) -> -4294967294.000000477 ~= -4294967294.000000000 (-2.67522e-07)
not ok 20 - testDiff(4294967295:999999999 - 1:999999999) -> 4294967294.000000477 ~= 4294967294.000000000 (2.67522e-07)
not ok 21 - testDiff(0:999999999 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-2.68454e-07)
not ok 22 - testDiff(4294967295:0 - 0:999999999) -> 4294967294.000000477 ~= 4294967295.000000000 (-1)
```

I generated .tap files on the two Intel systems, and the results are slightly different than the ones above. Here's the first diff chunk between those test results:
```patch
--- O.linux-x86/epicsTimeTest.tap	2023-02-28 21:00:18.823456000 -0600
+++ O.linux-x86_64/epicsTimeTest.tap	2023-02-28 21:00:03.671244000 -0600
@@ -18,18 +18,18 @@
 ok 14 - testDiff(1:1 - 1:1) -> 0.000000000 ~= 0.000000000 (0)
 ok 15 - testDiff(2:0 - 1:999999999) -> 0.000000001 ~= 0.000000001 (0)
 ok 16 - testDiff(1:999999999 - 2:0) -> -0.000000001 ~= -0.000000001 (0)
-not ok 17 - testDiff(1:0 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-2.67522e-07)
+not ok 17 - testDiff(1:0 - 4294967295:0) -> -4294967294.000000477 ~= -4294967294.000000000 (-4.76837e-07)
-not ok 18 - testDiff(4294967295:0 - 1:0) -> 4294967294.000000477 ~= 4294967294.000000000 (2.67522e-07)
[... 27 lines cut ...]
    0.00000000100000000000000006228159145777985642
$ perl -e 'printf "%50.44f\n", 4294967295.0e-9'
    4.29496729500000018475702745490707457065582275
```

Also the `testDiff()` routine isn't taking into account the precision available given the magnitude of the time difference, it's assuming that a double can always hold 9 decimal places:
```C++
static const unsigned mSecPerSec = 1000u;
static const unsigned uSecPerSec = 1000u * mSecPerSec;
static const unsigned nSecPerSec = 1000u * uSecPerSec;
static const double precisionEPICS = 1.0 / nSecPerSec;
...

    double actual = epicsTimeDiffInSeconds(&lhs, &rhs);
    double diff = actual - expect;

    testOk(fabs(diff)<precisionEPICS,
           "testDiff(%u:%u - %u:%u) -> %.9f ~= %.9f (%g)",
           unsigned(lhs.secPastEpoch), unsigned(lhs.nsec),
           unsigned(rhs.secPastEpoch), unsigned(rhs.nsec),
           actual, expect, diff);
```
Thus we could adjust the test to ignore the lower digits when the diff is large (there's probably a C99 function that could help with that) which might be enough for the tests to pass, but it seems much easier to use the version of `epicsTimeDiffInSeconds()` that divides by 1e9 instead of multiplying by 1e-9 since that does seem to work everywhere.

I will create a PR with that, but not tonight.

mdavidsaver, 2025-02-19:
I think this issue is still relevant.