epics-base/epics-base #333: epicsTimeTest cases for testDiff() and the code in epicsTimeDiffInSeconds() are wrong

anjohnson, 2023-02-26:
I copied the `testAdd()` and `testDiff()` test cases into the 3.15 version of epicsTimeTest.cpp, and its results disagreed with the expected answers for tests 17 through 22 inclusive.

The time difference between a later timestamp and an earlier one should _always_ be positive when expressed as a double, and the 3.15 version of `epicsTimeDiffInSeconds()` handles that correctly (despite 3.15 not actually having any tests to prove it).

The version of `epicsTimeDiffInSeconds()` shipped since the EPICS 7.0.6.1 release returns incorrect time differences in some circumstances, I think when the number of seconds exceeds 2**31 but I haven't confirmed that exactly. For example see test case 18:
```
    testDiff(0xffffffff,0, 1,0, -2.0);
```
Translating into what that's saying in English, this suggests that the beginning of the last second representable as an epicsTimeStamp (0xffffffff:0) occurred 2 seconds _before_ the beginning of the first second after the EPICS epoch (1:0). That's obviously wrong, `epicsTimeDiffInSeconds()` can and should return the time difference between those two timestamps as +4294967294.0 seconds.

These are the incorrectly failing tests returned by the 3.15 version of epicsTime.cpp:
```
not ok 17 - testDiff(1:0 - 4294967295:0) -> -4294967294.000000000 ~= 2.000000000 (-4.29497e+09)
not ok 18 - testDiff(4294967295:0 - 1:0) -> 4294967294.000000000 ~= -2.000000000 (4.29497e+09)
not ok 19 - testDiff(1:999999999 - 4294967295:999999999) -> -4294967294.000000000 ~= 2.000000000 (-4.29497e+09)
not ok 20 - testDiff(4294967295:999999999 - 1:999999999) -> 4294967294.000000000 ~= -2.000000000 (4.29497e+09)
not ok 21 - testDiff(0:999999999 - 4294967295:0) -> -4294967294.000000000 ~= 1.999999999 (-4.29497e+09)
not ok 22 - testDiff(4294967295:0 - 0:999999999) -> 4294967294.000000000 ~= -1.999999999 (4.29497e+09)
```

A simple working implementation is:
```c++
double epicsStdCall epicsTimeDiffInSeconds (const epicsTimeStamp *pLeft, const epicsTimeStamp *pRight)
{
    epicsInt64 sec = pLeft->secPastEpoch;
    sec -= pRight->secPastEpoch;
    epicsInt32 nsec = pLeft->nsec - pRight->nsec;

    return sec + double(nsec) * 1e-9;
}
```
I don't claim this as the most efficient implementation possible and it might also have some subtle UB in it, but it gives the same right answers as the original 3.15 version on darwin-aarch64, Ubuntu 18.04 on both 32 and 64-bit x86, and with the Xilinx compiler on linux-arm).

anjohnson, 2023-02-26:
These are the changes I made to epicsTimeTest.cpp to correct the expected difference:
```patch
diff --git a/modules/libcom/test/epicsTimeTest.cpp b/modules/libcom/test/epicsTimeTest.cpp
index 66d13e9..6e8a2cd 100644
--- a/modules/libcom/test/epicsTimeTest.cpp
+++ b/modules/libcom/test/epicsTimeTest.cpp
@@ -181,14 +181,14 @@ MAIN(epicsTimeTest)
     testDiff(2,0, 1,999999999, 0.000000001);
     testDiff(1,999999999, 2,0, -0.000000001);
 
-    testDiff(1,0, 0xffffffff,0, 2.0);
-    testDiff(0xffffffff,0, 1,0, -2.0);
+    testDiff(1,0, 0xffffffff,0, -4294967294.0);
+    testDiff(0xffffffff,0, 1,0, 4294967294.0);
 
-    testDiff(1,999999999, 0xffffffff,999999999, 2.0);
-    testDiff(0xffffffff,999999999, 1,999999999, -2.0);
+    testDiff(1,999999999, 0xffffffff,999999999, -4294967294.0);
+    testDiff(0xffffffff,999999999, 1,999999999, 4294967294.0);
 
-    testDiff(0,999999999, 0xffffffff,0, 1.999999999);  // 0.99999.. - -1.0
-    testDiff(0xffffffff,0, 0,999999999, -1.999999999); // -1.0 - 0.999..
+    testDiff(0,999999999, 0xffffffff,0, -4294967293.999999999);
+    testDiff(0xffffffff,0, 0,999999999, 4294967294.000000001);
 
     testDiff(0x80000000,0, 0x7fffffff,0, 1.0);
     testDiff(0x7fffffff,0, 0x80000000,0, -1.0);
```

You could also consider adding some tests that round-trip calls to both `epicsTimeDiffInSeconds()` then `epicsTimeAddSeconds()` and _vice versa_ since the two are supposed to be complementary.

Any reason you didn't test for the maximum time difference, which would be `(0xffffffff:999999999) - (0:1)`? (Actually 0:0 should be okay to calculate differences with, although we do treat it as `<undefined>` in some places).

I would also like to see some `testAdd()` calls with large values for `rhs` (i.e. larger than ±2**31 seconds), you currently only check for tiny ones so I guess we don't know for sure that the new `epicsTimeAddSeconds()` implementation is right for those cases. While I'm thinking about it, shouldn't `epicsTime::operator+(double)` and friends also throw an exception on overflow?

mdavidsaver, 2023-02-26:
@anjohnson Since you are "suggesting" specific changes, how about a PR?