epics-base/epics-base #343: `std::unexpected()` deprecated

dirk-zimoch, 2023-03-07:
The function  `epicsThread :: printLastChanceExceptionMessage` in modules/libcom/src/osi/epicsThread.cpp calls `std::unexpected ()`.
https://github.com/epics-base/epics-base/blob/fa00572780a00bea2f57dbff69653b14bd792be0/modules/libcom/src/osi/epicsThread.cpp#L84

The gcc-12 compiler claims that `std::unexpected ()` is deprecated:
```
../osi/epicsThread.cpp: In member function ‘void epicsThread::printLastChanceExceptionMessage(const char*, const char*)’:
../osi/epicsThread.cpp:84:21: warning: ‘void std::unexpected()’ is deprecated [-Wdeprecated-declarations]
   84 |     std::unexpected ();
      |     ~~~~~~~~~~~~~~~~^~
In file included from ../osi/epicsThread.cpp:14:
/opt/rh/gcc-toolset-12/root/usr/include/c++/12/exception:109:8: note: declared here
  109 |   void unexpected() __attribute__ ((__noreturn__));
      |        ^~~~~~~~~~
```
When using `-std=c++23`, compilation fails:
```
../osi/epicsThread.cpp: In member function ‘void epicsThread::printLastChanceExceptionMessage(const char*, const char*)’:
../osi/epicsThread.cpp:84:10: error: ‘unexpected’ is not a member of ‘std’
   84 |     std::unexpected ();
      |          ^~~~~~~~~~
```
See also: https://en.cppreference.com/w/cpp/error/unexpected

mdavidsaver, 2023-03-08:
I guess the current default behavior would be maintained by replacing with `std::terminate()`.  Although I wonder if simply returning and allowing the thread to stop would be preferable.

mdavidsaver, 2023-05-03:
Looking at this again.  `std::unexpected()` is only called (indirectly) from `epicsThreadCallEntryPoint()` after catching an exception.  I think it might be best to simply remove the call to `std::unexpected()`, so that an unhandled exception in a worker thread will log an error, but not `std::terminate()` the process.

dirk-zimoch, 2023-05-03:
I think the comment shows the wrong (or outdated?) understanding of the function of std::unexpected:
    // This behavior matches the C++ implementation when an exception
    // isn't handled by the thread code. Users can install their own
    // application-specific unexpected handler if preferred.
The comment suggests that std::unexpected() is called when an exception is not caught. But that is not true, according to every C++ reference I could find. (It is hard to find C++98 documentation, though.) An uncaught exception directly calls std::terminate(). What calls std::unexpected() is an exception thrown by a function that had promised to not throw that type of exception, i.e. a function declared with throw(exceptiontype, …) throwing an exception not in the list. Since those “ Dynamic exception specifications” (https://en.cppreference.com/w/cpp/language/except_spec) no longer exist since C++17 (and was deprecated since C++11), the concept of “unexpected exceptions” is obsolete and thus there is no more use case for std::unexpected().
The code should always have called std::terminate(). (Or re-throw the exception from the catch(…) clause.)

mdavidsaver, 2024-06-19:
> ... simply returning and allowing the thread to stop would be preferable.

`timerQueueActive :: _printLastChanceExceptionMessage()` does this.

I think I would be fine with a PR removing the call to `std::unexpected()`.

kiwichris, 2024-07-02:
I am now seeing this with RTEMS 6 as we have updated the gcc version and that has changed the default C++ standard.

Would you like me to create a pull request to remove this call?

anjohnson, 2024-07-03:
@kiwichris If this is affecting your RTEMS-6 builds then yes, please do.

anjohnson, 2024-07-03:
@dirk-zimoch I have a copy of the ANSI/ISO C++98 standard. Here's the description from 15.5.2 of that; there are several other mentions of it elsewhere though:

**15.5.2 The `unexpected()` function \[except.unexpected\]**

1. If a function with an _exception-specification_ throws an exception that is not listed in the _exception-specification_, the function
```
    void unexpected();
```
   is called (18.6.2) immediately after completing the stack unwinding for the former function

2. The `unexpected()` function shall not return, but it can throw (or re-throw) an exception. If it throws a new exception which is allowed by the exception specification which previously was violated, then the search for another handler will continue at the call of the function whose exception specification was violated. If it throws or rethrows an exception that the _exception-specification_ does not allow then the following happens: If the _exception-specification_ does not include the class `std::bad_exception` (18.6.2.1) then the function terminate() is called, otherwise the thrown exception is replaced by an implementation-defined object of the type std::bad_exception and the search for another handler will continue at the call of the function whose _exception-specification_ was violated.

3. Thus, an _exception-specification_ guarantees that only the listed exceptions will be thrown. If the _exception-specification_ includes the type `std::bad_exception` then any exception not on the list may be replaced by `std::bad_exception` within the function `unexpected()`.

dirk-zimoch, 2024-07-03:
So it is still the same: `unexpected()` is for exceptions violating the exception specification, not for uncaught exceptions. Thus the use in this context is wrong.

PR #543 (2024-09-09): libCom: don't use std::unexpected in epicsThread.
Besides being deprecated in C++11 and removed in C++17, the intended use case for this was always wrong, since std::unexpected() is called by the C++ runtime when a function throws an exception that was not specified in its "dynamic exception specification", which is different from an exception thrown by user code which wasn't caught [1,2].

Allowing the program to proceed instead of terminating it (via std::terminate(), which is the default unexpected handler) also seems to be more correct, since the thread which threw the exception has alread [... cut]
closed 2024-09-09 by mdavidsaver