epics-base/epics-base #351: Installation artifacts are not installed correctly if they exist in a parent directory

simon-ess, 2023-03-08:
When EPICS base builds, it installs (for example) `alarm.h` (from modules/libcom/src/misc) into `$(TOP)/includes`. If the _parent_ directory to EPICS base also has an `includes/alarm.h` then this file will not be installed. This applies to any of the targets that we are trying to install. That is, if we have something akin to
```
├── epics-base
│   ├── includes
│   ├── bin
│   ├── modules
│   ├── dbd
│   ├── ...
│   └── includes
│       └── *.h
└── install
    └── alarm.h
```
then alarm.h will fail to install upon running `make inc` from the `epics-base` directory.

The reason is as follows: one of the paths we search for our files is `..`; see https://github.com/epics-base/epics-base/blob/7.0/configure/RULES_BUILD#L23 and https://github.com/epics-base/epics-base/blob/7.0/configure/CONFIG_COMMON#L158-L163. When `make` then tries to look for the target `$(INSTALL_INC)/%.h`, it first looks using any of the paths in `vpath` for this target; if it finds it then it treats the parent directory's copy as already existing and so does not build the target.

Note: this is actually only true if the file in the parent directory is newer. As such a fresh clone of EPICS base _should_ not run into this issue. However, a case has arisen where header files are copied automatically by a development environment in a way that prevents them from being installed correctly.

A relatively simple example can be done with the following setup:
```
├── build
│   ├── Makefile
│   └── src
│       └── foo.h
└── include
    └── foo.h
```
with the Makefile having the contents
```make
INC += foo.h
INSTALL_DIR = include
SRC_DIR = src
INSTALL_INC = $(addprefix $(INSTALL_DIR)/,$(INC))

vpath %.h .. $(SRC_DIR)

.PHONY: all
all: $(INSTALL_INC)

$(INSTALL_DIR):
	mkdir $@

$(INSTALL_DIR)/%: % | $(INSTALL_DIR)
	@echo Copying file '$<' to '$@'
	cp $< $@
```
Running `make all` in this should copy `foo.h` to `include`, but its existence in the parent directory blocks it due to the `vpath` rule.

mdavidsaver, 2023-05-03:
I'm moderately confused.  (situation normal for me wrt. Make)  Maybe I'm missing something about how `vpath` works.  Wouldn't `vpath ..` only act the way you describe when `$(PWD)==$(TOP)`?  What happens in an `O.*` subdirectory, where install rules are normally found?

Adding `build/O.linux-x86` to the situation given above:

```
├── build
│   ├── O.linux-x86
│   ├── Makefile
│   └── src
│       └── foo.h
└── include
    └── foo.h
```

A more representative situation would start with:

```
make -C build/O.linux-x86 -f ../Makefile
```

Then, as I understand it, `vpath ..` would search in `build/`.

simon-ess, 2023-05-09:
I think you're correct, but `$(PWD)==$(TOP)` when you first run `make` in the top directory to build EPICS base in the first place...

mdavidsaver, 2024-06-19:
@anjohnson Is this behavior "won't fix"?

anjohnson, 2024-06-24:
Probably. Until such time as @simon-ess or I can find enough time to work out if there's any way to avoid this behavior I think the Build Facility paragraph quoted above should be added back into the equivalent chapter in the epics-docs copy of the AppDevGuide. @ttkorhonen was that deleted intentionally?

ttkorhonen, 2024-06-24:
Certainly not intentionally. I am not sure if it was due to a) error in conversion from the original TeX source (conversions are by no means perfect, but this piece was not particularly exotic) or b) I started from a wrong version.
However, I created a pull request which puts back this sentence and, while I was at it, contains a full review of the chapter, to match the latest 3.16.2 release but keeping the recent changes in Github (I tried not to revert any changes by others.)
However, one major change was to re-format the big "Table of Makefile definitions" to Markdown style. It is not perfect but I think it presents the content much better now.
Reviews (of the doc PR 95) are welcome. Just be aware that it took me almost a full day to review and edit.

ttkorhonen, 2025-02-20:
The missing text, warning about the side effect, is in [the docs site text.](https://docs.epics-controls.org/en/latest/build-system/specifications.html#install-directories). 
I could still highlight the note if wanted, but unless the behavior will be changed, the documentation is correct as far as I can tell.