epics-base/epics-base #362: iocsh cannot handle escaped quotes in quotes

dirk-zimoch, 2023-03-09:
Commands like `echo "x\"y"` falsely trigger an "unbalanced quote" error.
The reason is that within the quoted string, backslash is not recognizes and thus the escaped inner quote is interpreted as the closing quote of the string.

daykin, 2023-03-09:
The solution seems trivial enough - when you are checking for 'quote=c' e.g. in iocsh.cpp:337, check if preceding character is a backslash as well. But, do we want to handle [literal backslashes, which themselves are escaped?](https://xkcd.com/1638)

dirk-zimoch, 2023-03-10:
Simply checking for `\"` is not enough. What about `echo "\\"` or `echo "\\\\\\\\\\\""`?

dirk-zimoch, 2023-03-10:
But the soulution of course goes into this direction: Properly handle the `backslash` variable in the code.

daykin, 2023-03-10:
OK, how about this:
1. During scanning, a backslash is seen - point to it with `firstBackslash`. 
2. Continue scanning (icin++), until a backslash is not seen. 
3. `backslash` is set to `(icin - firstBackslash)%2`
4. even numbers of backslashes are thus paired off as literal backslashes, and odd numbers will escape the character immediately following.

daykin, 2023-03-14:
Ah, you've beaten me to the punch 👍 
I agree that this whole thing should probably be simplified.