epics-base/epics-base #369: errPrintf mangles error messages

dirk-zimoch, 2023-03-22:
With commit 29fa0621, readability of error messages has degraded. The cause is here:
https://github.com/epics-base/epics-base/blob/5eff3803a83fbc3ea855b63841f5359c09830ade/modules/libcom/src/error/errlog.c#L503

First, the code does no longer checks if a filename is given, second the newline has been removed (and not been replaced with space). 

User code like
```
recGblRecordError(S_db_badField, precord, "SOME MESSAGE");
```
Now results in hard to read output like this:
```
Illegal field value filename="(null)" line number=0PV: PVNAME SOME MESSAGE
```
Where we had something like this before:
```
Illegal field value PV: PVNAME SOME MESSAGE
```

dirk-zimoch, 2023-03-22:
If a file name was provided, we used to get errors like this:
```
Bad Field value 
filename="../macCore.c" line number=709
<string>: SOME_MACRONAME referenced but undefined
```
While now I see messages like this:
```
filename="../dbStatic/dbLexRoutines.c" line number=270dbRead opening file FILENAMEdbLoadRecords: failed to load 'FILENAME'
```
which have no whitespace between message parts.

mdavidsaver, 2023-03-22:
The current lack of whitespace is an oversight on my part.

wrt. to a general lack of brevity.  Instead of `"filename=\"%s\" line number=%d\n"`, maybe just `%s:%d`.  I don't think quoting the path adds much.  Additionally, maybe strip off any directories, and only print the file name?  All of our `__FILE__` have at least a useless `../` prefix.

dirk-zimoch, 2023-03-23:
So we are aiming for something like:
```
dbLexRoutines.c:270 Read opening file FILENAME
```
Maybe we should check all error messages in base for readability and clarity. It should be clear to the user where the problem is. Pointing to the parser source code line number when there is an problem with a `file` command in a subs file is quite pointless. Refer to the subs file and line number instead. This may take a while to find.

ralphlange, 2023-03-23:
> > It should be clear to the user where the problem is ...
> 
> imo. the message string is for users. The file+line is for us when the message appears on tech-talk.

I agree to both:
Source code file+line for us is useful.
An error that happens while reading a user-controlled configuration file should have the name and line of the file where the error has been found. As part of the message.

mdavidsaver, 2025-02-19:
> The current lack of whitespace is an oversight on my part.

Specifically corrected by 77490d4939bcce41bc2b053053e63dd820e1329d

I did not attempt any of the potential format changes discussed above.  I think some change in this area would be useful, and could be a small project for a codeathon.