epics-base/epics-base #386: Move commandline editing (libreadline dependency) from libCom to iocShell

ralphlange, 2023-05-12:
See this [Tech-Talk thread](https://epics.anl.gov/tech-talk/2023/msg00673.php):

The readline library has been introduced to improve usability of the iocShell. The current use of readline within libCom introduces a dependency on libreadline for anything that uses libCom, like, e.g., libCa. A network library depending on a command line editing library is awkward.

Maybe the code using readline can be moved into the iocShell realm?

mdavidsaver, 2023-05-16:
I think a separate library would be the right approach if iocsh were being introduced into Base today.  Unfortunately, there is now 23 years of legacy.  At this late date I think such a incompatible change would be more trouble than it is worth.

One possible, though more technically complex way, to avoid a NEEDED dependency in libCom would be to move the readline support code (but not all of iocsh) into a "plugin" library when linking libCom.so (but not libCom.a).  Something analogous to the several NSS plugin libraries of glibc.  Such a plugin library would be opportunistically loaded the first time `iocsh(NULL)` is actually called.

On the whole though, I see the cost of such a scheme as outweighing the benefit.

anjohnson, 2023-05-16:
Most EPICS applications built by our build system should be using one of the `EPICS_BASE_HOST_LIBS` or `EPICS_BASE_IOC_LIBS` Makefile variables to get the list of Base libraries to be linked against. Those variables have been defined in Base since 3.14.something in 2003 to let us change the libraries that we build and that apps need to be linked against. Their values changed between the 3.14, 3.15 and 7.0 branches, and at one time we even used to have a separate libiocsh library. I could thus see it as being feasible to split libCom into a non-shell part and another part that provides iocsh and the readline bindings.

`EPICS_BASE_HOST_LIBS` is defined in `configure/CONFIG_CA_MODULE` and was meant to be for command-line tools, which I could squint to see it as requesting the non-shell libraries, so I'd define another variable `EPICS_BASE_SHCOM_LIBS` say for the libraries that include the shell. For backwards-compatible builds, applications that need the shell would have to use something like
```Makefile
PROD_LIBS += $(if $(EPICS_BASE_SHCOM_LIBS), $(EPICS_BASE_SHCOM_LIBS), $(EPICS_BASE_HOST_LIBS))
```
to use the new one when it's available or fall-back to the old one when not.

Is this going too far?

mdavidsaver, 2023-05-16:
I see no reason to discuss changing anything other than the epicsReadline wrappers.  As far as I am concerned, the core IOC shell will remain in libCom.