epics-base/epics-base #399: RTEMS SMP does not support interrupt disable/enable

kiwichris, 2023-07-06:
EPCIS  does not build if RTEMS is built with SMP enabled:
```
../osi/os/RTEMS/osdInterrupt.c: In function 'epicsInterruptLock':
../osi/os/RTEMS/osdInterrupt.c:30:5: error: implicit declaration of function 'rtems_interrupt_disable'; did you mean 'rtems_interrupt_raise'? [-Werror=implicit-function-declaration]
   30 |     rtems_interrupt_disable (level);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
      |     rtems_interrupt_raise
../osi/os/RTEMS/osdInterrupt.c: In function 'epicsInterruptUnlock':
../osi/os/RTEMS/osdInterrupt.c:39:5: error: implicit declaration of function 'rtems_interrupt_enable'; did you mean 'rtems_interrupt_raise'? [-Werror=implicit-function-declaration]
   39 |     rtems_interrupt_enable (level);
      |     ^~~~~~~~~~~~~~~~~~~~~~
      |     rtems_interrupt_raise
../osi/os/RTEMS/osdInterrupt.c: In function 'epicsInterruptLock':
../osi/os/RTEMS/osdInterrupt.c:30:5: warning: 'level' is used uninitialized [-Wuninitialized]
   30 |     rtems_interrupt_disable (level);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../osi/os/RTEMS/osdInterrupt.c:28:27: note: 'level' was declared here
   28 |     rtems_interrupt_level level;
      |                           ^~~~~
```
The RTEMS documentation details the specifics in the [Disabling of Interrupts](https://docs.rtems.org/branches/master/c-user/symmetric_multiprocessing_services.html#disabling-of-interrupts) section of SMP [Application Issues](https://docs.rtems.org/branches/master/c-user/symmetric_multiprocessing_services.html#application-issues).

I do not see a means of resolving this without an API change?

mdavidsaver, 2023-07-06:
fyi. the default implementation of osdInterrupt.c is a global mutex.  For RTEMS w/ SMP, I guess the equivalent would be something like a global spinlock?

Does RTEMS >=5 provide something like a spinlock?  If so, then `posix/osdSpin.c` may need replacing as well.

There are no many usages of `epicsInterruptLock()`.  And fewer still of the newer `epicsSpin*()`.  Although most are in RTEMS/vxWorks code.  eg. around read-modify-write MMIO access.

https://github.com/search?q=org%3Aepics-modules%20epicsInterruptLock&type=code

kiwichris, 2023-07-06:
@mdavidsaver thanks for this and yes I only found a single place of use in `epics-base`. in RTEMS there is:
```
  rtems_interrupt_lock_context lock_context;
  rtems_interrupt_lock_acquire( &lock, &lock_context );
  /* Critical section */
  rtems_interrupt_lock_release( &lock, &lock_context );
```
or:
```
pthread_spinlock_t lock;
int error;
error = pthread_spin_lock( &lock );
assert( error == 0 );
/* Critical section */
error = pthread_spin_unlock( &lock );
assert( error == 0 );
```
Both approaches add an argument which is not currently present and that makes changing the code in a compatible way harder.

I also do not know how such a change would effect vsWorks?

mdavidsaver, 2023-07-07:
> Both approaches add an argument ...

Please take a look at how this is handled with the default implementation, which is used on non-RTOS targets.  (Without the lazy initialization)  Would this approach work with RTEMS/SMP replacing a pthread mutex with a spin_lock?

https://github.com/epics-base/epics-base/blob/42604fc794b6f26b60db953a230b52c1118d3039/modules/libcom/src/osi/os/default/osdInterrupt.c#L27
...

https://github.com/epics-base/epics-base/blob/42604fc794b6f26b60db953a230b52c1118d3039/modules/libcom/src/osi/os/default/osdInterrupt.c#L35-L40

kiwichris, 2023-07-07:
> Would this approach work with RTEMS/SMP replacing a pthread mutex with a spin_lock?

Yes I it would. 

The only location and the error counter would be better as an atomic but I am not sure what the status of atomic is in EPICS base.

mdavidsaver, 2023-07-07:
> ... I am not sure what the status of atomic is in EPICS base.

I am not sure what "atomic" you are referring to.  Although as an internal detail of RTEMS specific code, maybe just some `#ifdef` for the SMP case?

kiwichris, 2023-07-07:
> am not sure what "atomic" you are referring to. 

Sorry, I was not clear in what I said. I mean atomic variables  such as `std::atomic_bool` in C++. RTEMS supports atomic variables on most or all architectures.